borglab/gtsam #1448: Incorrect MPE for Long Chains of Discrete Factor Graphs

pcl15423, 2023-02-07:
## Description
Hi team, thanks for the wonderful library!

I've run into an issue where I have a fairly large Discrete Factor Graph that I believe is running into an underflow type of error. I've copied a minimal toy example below that shows the same symptom. I'm not sure where to turn for documentation of python bindings or how to debug the optimization further. Thank you for the help!

## Steps to reproduce
Toy example is a hidden markov model with 3 possible states and 200 Observations. In this example, the transition matrix is mostly identity. The desired state outputs any number [0, 199] with equal probability. All other states output 200 with probability 1. I set each measurement equal to the observation index (so only [0,200] are observed and the desired state should be preferred always).

``` python
import gtsam
import numpy as np

import gtbook
from gtbook.discrete import Variables
VARIABLES = Variables()

def toy_problem(num_states, num_obs, desired_idx):
    states = list(range(num_states))
    X = VARIABLES.discrete_series("X", range(num_obs), states)
    Z = VARIABLES.discrete_series("Z", range(num_obs), range(num_obs+1))
    graph = gtsam.DiscreteFactorGraph()

    # Mostly identity transition matrix
    transitions = np.eye(num_states)
    
    # Needed otherwise mpe is always state 0?
    transitions += 0.1/(num_states)
    
    transition_cpt = []
    for i in range(0, num_states):
        transition_row = "/".join([str(x) for x in transitions[i]])
        transition_cpt.append(transition_row)
    transition_cpt = " ".join(transition_cpt)

    for i in reversed(range(1, num_obs)):
        transition_conditional = gtsam.DiscreteConditional(X[i], [X[i-1]], transition_cpt)
        graph.push_back(transition_conditional)

    # Contrived example such that the desired state gives measurements [0, num_obs) with equal probability
    #   but all other states always give measurement num_obs
    obs = np.zeros((num_states, num_obs+1))
    obs[:,-1] = 1
    obs[DESIRED_STATE,0: -1] = 1
    obs[DESIRED_STATE,-1] = 0
    obs_cpt_list = []
    for i in range(0, num_states):
        obs_row = "/".join([str(z) for z in obs[i]])
        obs_cpt_list.append(obs_row)
    obs_cpt = " ".join(obs_cpt_list)
    
    # Contrived example where each measurement is its own index
    for i in range(0, num_obs):
        obs_conditional = gtsam.DiscreteConditional(Z[i], [X[i]], obs_cpt)
        factor = obs_conditional.likelihood(i)
        graph.push_back(factor)
    
    mpe = graph.optimize()
    vals = [mpe[X[i][0]] for i in range(num_obs)]
    print("MPE Vals: ")
    print(vals)
    print("Indices where mpe does not equal desired state:")
    print(np.where(np.array(vals) != DESIRED_STATE))

    # Missing many outputs
    sum_product = graph.sumProduct()
    breakpoint()

if __name__ == "__main__":
    NUM_STATES = 3
    NUM_OBS = 200
    DESIRED_STATE = 1

    toy_problem(NUM_STATES, NUM_OBS, DESIRED_STATE)
```

Output:
```
MPE Vals: 
[1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
Indices where mpe does not equal desired state:
(array([138, 139, 140, 141, 142, 143, 144, 145, 146, 147, 148, 149, 150,
       151, 152, 153, 154, 155, 156, 157, 158, 159, 160, 161, 162, 163,
       164, 165, 166, 167, 168, 169, 170, 171, 172, 173, 174, 175, 176,
       177, 178, 179, 180, 181, 182, 183, 184, 185, 186, 187, 188, 189,
       190, 191, 192, 193, 194, 195, 196, 197, 198, 199]),)

```

## Expected behavior

Expected behavior is that the maximum probability estimate is equal to DESIRED_STATE (1) for all hidden states. This does not hold true for hidden states 138 and above. Additionally, the graph resulting from `sum_product = graph.sumProduct()` is missing many values for these incorrect states.

## Environment

I'm running on Python 3.8.13, gtsam version 4.2a9, Ubuntu 20.04

keevindoherty, 2023-02-07:
I'm not sure about the sum-product results, and I'm on an earlier version of the GTSAM max-product code, but I had similar issues. I had some success modifying the GTSAM `EliminateForMPE` function (maybe @dellaert can advise on what the latest API is for this - sorry I'm not 100% up to speed on the latest in GTSAM discrete!) as follows:

```c++
// Alternate eliminate function for MPE
static std::pair<DiscreteConditional::shared_ptr,
                 DecisionTreeFactor::shared_ptr>  //
EliminateForMPE(const DiscreteFactorGraph& factors,
                const Ordering& frontalKeys) {
  // PRODUCT: multiply all factors
  gttic(product);
  DecisionTreeFactor product;
  for (auto&& factor : factors) product = (*factor) * product;
  gttoc(product);

  // Can we divide out a scalar?
  Ordering ordering = Ordering::Colamd(DiscreteFactorGraph{product});
  DecisionTreeFactor::shared_ptr normalization = product.sum(ordering);
  product = product / (*normalization);

  // max out frontals, this is the factor on the separator
  gttic(max);
  DecisionTreeFactor::shared_ptr max = product.max(frontalKeys);
  gttoc(max);

  // Ordering keys for the conditional so that frontalKeys are really in front
  DiscreteKeys orderedKeys;
  for (auto&& key : frontalKeys)
    orderedKeys.emplace_back(key, product.cardinality(key));
  for (auto&& key : max->keys())
    orderedKeys.emplace_back(key, product.cardinality(key));

  // Make lookup with product
  gttic(lookup);
  size_t nrFrontals = frontalKeys.size();
  auto lookup =
      boost::make_shared<DiscreteLookupTable>(nrFrontals, orderedKeys, product);
  gttoc(lookup);

  return std::make_pair(
      boost::dynamic_pointer_cast<DiscreteConditional>(lookup), max);
}
```

The key line here is `product = product / (*normalization)` which should normalize the intermediate max marginals. This seemed to fix results for me (buyer beware: I haven't triple-checked the math on this yet, and wasn't planning on raising this until I was 100% sure this was kosher, but your issue made it seem like sharing this now could be useful). The alternative, I suppose, would be to do all the intermediate arithmetic in log-space, but this could come with some loss of precision.

I also don't know if this is easily modifiable via the Python discrete factor graph API, but I was able to use my custom `EliminateForMPE` by passing it in explicitly to GTSAM (i.e. without making direct modifications to `DiscreteFactorGraph.cpp`).

keevindoherty, 2023-02-07:
@dellaert Plenty of time :)

I'll try to get the above change PR'd in the next few minutes.

keevindoherty, 2023-02-08:
@pcl15423 Pull request #1449 should fix the MPE issue, but does not address any issue with sum-product. Do you have more information on the sum-product expected vs. actual results in this case? I wasn't sure what was meant by missing values here. By "missing" do you mean incorrect, or actually not present in the output? Trying to determine if this is the same computational issue or a separate issue that happens to also be related to scaling to large chains.

dellaert, 2023-02-08:
@keevindoherty I have been thinking some HMM-specific classes might be a good addition. If we provide a shared pointer to  sparse Eigen matrices we can
- save tons of memory
- support very large state/observation cardinalities
- create a sum-product example that calculates the stationary distribution, which we can check with the transition matrix Eigenvector...

dellaert, 2023-02-08:
...and implement EM for parameter estimation, e.g., here https://gtbook.github.io/gtsam-examples/intro.html

keevindoherty, 2023-02-08:
@dellaert It's not a bad idea, since I'd anticipate this would be a major use case for this tool. I'm particularly interested in the large state/observation case lately (maybe with sparse observation / transition probabilities); dealing with "large" but sparse discrete factors would be really valuable beyond the immediate application to HMMs..

dellaert, 2023-02-08:
Starting a discussion/design doc here: #1451

pcl15423, 2023-02-08:
> @pcl15423 Pull request #1449 should fix the MPE issue, but does not address any issue with sum-product. Do you have more information on the sum-product expected vs. actual results in this case? I [...]

Thanks for putting up the PR! I'm really impressed by the speed of development here.

I built again with your new changes and it does fix the mpe issue, but does not fix the sum-product like you suspected. I think it is probably a similar issue though.

When I try to inspect `sum_product`, I do not see the full discrete conditional table that I expect for later values (even after your fix). Previously, the mpe started being incorrect at state 138. Here the sum_product also seems to be incorrect at state 138. 

```
ipdb> sum_product.at(0)
Discrete Conditional
 P( X0 | X1 ):
 Choice(X1) 
 0 Choice(X0) 
 0 0 Leaf    0
 0 1 Leaf    1
 0 2 Leaf    0
 1 Choice(X0) 
 1 0 Leaf    0
 1 1 Leaf    1
 1 2 Leaf    0
 2 Choice(X0) 
 2 0 Leaf    0
 2 1 Leaf    1
 2 2 Leaf    0

ipdb> sum_product.at(137)
Discrete Conditional
 P( X137 | X138 ):
 Choice(X138) 
 0 Choice(X137) 
 0 0 Leaf    0
 0 1 Leaf    1
 0 2 Leaf    0
 1 Choice(X137) 
 1 0 Leaf    0
 1 1 Leaf    1
 1 2 Leaf    0
 2 Choice(X137) 
 2 0 Leaf    0
 2 1 Leaf    1
 2 2 Leaf    0

ipdb> sum_product.at(138)
Discrete Conditional
 P( X138 | X139 ):
 Leaf    0

ipdb> sum_product.at(199)
Discrete Conditional
 P( X199 ):
 Leaf    0
```

keevindoherty, 2023-02-08:
@pcl15423 OK so I have an update on this even faster than I anticipated. Can you "git checkout" the `hotfix/sumproduct` branch of my GTSAM fork here: https://github.com/keevindoherty/gtsam/tree/hotfix/sumproduct and try running your code with that to verify that it behaves as you'd expect.

I implemented the same change as above in the sum-product elimination function and modified the test to print the sum-product results for the keys you suggested. I got:
```
test_sumProduct_chain (test_DiscreteFactorGraph.TestDiscreteFactorGraph)
Test for numerical underflow in EliminateDiscrete on long chains. ... 
(This should have 9 potential assignments)
Discrete Conditional
 P( X0 | X1 ):
 Choice(X1) 
 0 Choice(X0) 
 0 0 Leaf    0
 0 1 Leaf    1
 0 2 Leaf    0
 1 Choice(X0) 
 1 0 Leaf    0
 1 1 Leaf    1
 1 2 Leaf    0
 2 Choice(X0) 
 2 0 Leaf    0
 2 1 Leaf    1
 2 2 Leaf    0

(This should have 9 potential assignments)
Discrete Conditional
 P( X138 | X139 ):
 Choice(X139) 
 0 Choice(X138) 
 0 0 Leaf    0
 0 1 Leaf    1
 0 2 Leaf    0
 1 Choice(X138) 
 1 0 Leaf    0
 1 1 Leaf    1
 1 2 Leaf    0
 2 Choice(X138) 
 2 0 Leaf    0
 2 1 Leaf    1
 2 2 Leaf    0
```
which sounds like it was what you were looking for?

pcl15423, 2023-02-08:
Hi yes, this does look it like resolves my issue in both the test case and in my project!

When I try to run all of the unit tests with `make check`, it says that a different unit test now fails:
```
99% tests passed, 1 tests failed out of 286

Total Test time (real) =   3.99 sec

The following tests FAILED:
         82 - testDiscreteFactorGraph (Failed)
Errors while running CTest
make[3]: *** [CMakeFiles/check.dir/build.make:76: CMakeFiles/check] Error 8
make[2]: *** [CMakeFiles/Makefile2:2932: CMakeFiles/check.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:2939: CMakeFiles/check.dir/rule] Error 2
make: *** [Makefile:229: check] Error 2
```

Fortunately, this might just require the expected values to be adjusted? The old expected values seem to be unnormalized versions of the new ones.
```
 82/286 Test  #82: testDiscreteFactorGraph .......................***Failed    0.01 sec
Not equal:
expected:
 f[ (1,2), (2,2), ]
 Choice(2) 
 0 Choice(1) 
 0 0 Leaf   10
 0 1 Leaf    6
 1 Choice(1) 
 1 0 Leaf    6
 1 1 Leaf   10
actual:
 f[ (1,2), (2,2), ]
 Choice(2) 
 0 Choice(1) 
 0 0 Leaf 0.3125
 0 1 Leaf 0.1875
 1 Choice(1) 
 1 0 Leaf 0.1875
 1 1 Leaf 0.3125
/home/phlee/gtsam/gtsam_scratch/gtsam/gtsam/discrete/tests/testDiscreteFactorGraph.cpp:122: Failure: "assert_equal(expectedFactor, *newFactor)" 
There were 1 failures
```

dellaert, 2023-02-09:
That is a reasonable assumption.

varunagrawal, 2023-02-09:
Sorry for jumping in, but yes if you divide all the values by the sum (32 in this case) you'll get the actual values exactly. This should be just a regression update.

keevindoherty, 2023-02-09:
I just added a patch for this test in the latest commit on the `hotfix/sumproduct` branch of my fork here: https://github.com/keevindoherty/gtsam/tree/hotfix/sumproduct (I simply normalized the "expected" result). On my machine all tests are now passing with the normalized product operation in `DiscreteEliminate`. Happy to kick off a new PR if that would be useful?

dellaert, 2023-02-10:
Please do! Thanks!!!

PR #1453 (2023-02-12): Add normalization to sum-product, avoiding underflow.
This quick fix should actually resolve #1448 now. Adds the same product normalization from #1449 to the `DiscreteFactorGraph::DiscreteEliminate` function. This broke an existing test in `testDiscreteFactorGraph.cpp` due to an expected result being unnormalized, where the result from `DiscreteEliminate` was normalized, so I modified the test to expect a normalized result. I also added a "visual test" to `test_DiscreteFactorGraph.py` to verify the expected results in #1448.

Couple notes for reviewer(s):
- *Highest priority:* The test I added [... cut]
closed 2023-02-12 by dellaert