borglab/gtsam #1452: Memory keeps growing when using gtsam::IncrementalFixedLagSmoother for marginalization

kokeliang, 2023-02-10:
## Description

Hello. I am using gtsam::IncrementalFixedLagSmoother for marginalization. I think the marginalization worked but memory keeps growing. 

`
gtsam::IncrementalFixedLagSmoother optimizer;
.
.
.
optimizer.update(graphFactors, graphValues, smootherTimestamps);
std::cout << "factorgraph size: << optimizer.getFactors().size() << "factorgraph non-null size:  " << optimizer.getFactors().nrFactors() << std::endl;
`

`factorgraph size` keeps growing and causes memory to grow while `factorgraph non-null size` is stable after marginalization works.

Is this a problem with gtsam or something wrong with my marginalization code? Thanks a lot for all replies.

dellaert, 2023-02-12:
It's a bit hard to say without more info/code. Also, this might be more for the Google group - if you want to join please make sure to give a reason as I reject other requests as spam.

gumpcarl, 2023-12-07:
Has this problem been solved？

JokerJohn, 2025-01-09:
@kokeliang @dellaert 
I’ve encountered the same issue. Have you found a solution yet? Currently, I’ve only seen people rebuild the factor graph when it reaches a fixed number of nodes (e.g., 500) to achieve this, but this leads to disconsistancy. Is there a more elegant solution?
[image]

keevindoherty, 2025-01-10:
I think the issue you are running into here is that the test does not set `findUnusedFactorSlots = true` in the `ISAM2Params` struct that gets passed to the smoother constructor.

I tried changing the constructor setup around line 102 of `testIncrementalFixedLagSmoother.cpp` to:
```c++
  auto params = ISAM2Params();
  params.findUnusedFactorSlots = true;
  IncrementalFixedLagSmoother smoother(12.0, params);
```

If you just make that change as I did, an exception is thrown in the line where factors are removed (somewhere around line 258 of `testIncrementalFixedLagSmoother.cpp`). Commenting out that line, the rest of the test works I think more or less as expected, and the factor graph size remains bounded:

**Before**
graph size: 1934
graph nrfactors: 31

**After**
graph size: 35
graph nrfactors: 31

It looks like there may be something fishy going on when trying to explicitly remove factors in the smoother update when `findUnusedFactorSlots = true`; I haven't dug into this, so I don't know if this is something specific to that test or a deeper issue. But, if you're not trying to remove factors, I think setting `findUnusedFactorSlots = true`should help with the unbounded graph size issue.

Hope that helps!

JokerJohn, 2025-01-10:
@keevindoherty 

Thanks for your help!

In fact, I've  already set this parameter `findUnusedFactorSlots = true`， and my settings are as follows. 
[image]

Theoretically, my sliding window should have 10 states (1 second), but the situation hasn’t changed, and the graph size continues to increase. 
[image]

I printed out the graph using xx and found that **the old factors were reset to null**, but they weren’t deleted. This could be one of the core reasons.
[image]

dellaert, 2025-01-10:
Yeah, the idea is that nullptr slots are re-used. It would be great to have a minimal repro example if there is a bug there, but maybe first try develop to make sure it is not fixed there. If you have a repro, I can look into it.

JokerJohn, 2025-01-12:
@dellaert 
Hi, Frank, thank you for your response. 
I’ve generated a complete real-world test dataset from the `corridor_day` sequence of the [FusionPortable](https://fusionportable.github.io/dataset/fusionportable/) SLAM dataset, which contains only prior factor (1) and LiDAR odometry between factors. The prior factor and their covariance come from ICP matching between the first point cloud frame and the prior map. The relative pose estimates and covariance estimates for the odometry come from the [FAST-LIO2](https://github.com/hku-mars/FAST_LIO) algorithm.
[image]
```
1645005900.499903202057 0 0 6.212382037760 0.123492019685 0.785078536573 -0.033043253282 0.046918795098 0.269745850974 0.000021367074 0.000001103400 -0.000000177223 -0.000000846409 0.000019141959 0.000017130724 0.000001103400 0.000001647049 -0.000000004437 -0.000001401443 0.000000963738 0.000009233901 -0.000000177223 -0.000000004437 0.000001520314 -0.000000802863 -0.000008134038 -0.000000117329 -0.000000846409 -0.000001401443 -0.000000802863 0.000104448212 0.000003494338 -0.000007807899 0.000019141959 0.000000963738 -0.000008134038 0.000003494338 0.000161853532 0.000015216778 0.000017130724 0.000009233901 -0.000000117329 -0.000007807899 0.000015216778 0.000160403069
```
Each row in the test data represents the following: 
```
timestamp | factor_type | key | x | y | z | roll | pitch | yaw | covariance matrix (6*6)
```
I’ve written a minimal code example, but I’m not sure if it meets the requirements. I hope this data can help us address the issue with minimal cost, assuming it’s determined to be a bug.
```
#include <fstream>
#include <iostream>
#include <string>
#include <sstream>
#include <gtsam/geometry/Pose3.h>
#include <gtsam/nonlinear/NonlinearFactorGraph.h>
#include <gtsam/nonlinear/Values.h>
#include <gtsam/nonlinear/ISAM2.h>
#include <gtsam/slam/PriorFactor.h>
#include <gtsam/slam/BetweenFactor.h>
#include <gtsam_unstable/nonlinear/IncrementalFixedLagSmoother.h>
#include <gtsam/inference/Symbol.h>
#include <gtsam/slam/dataset.h> // For writeG2o

using namespace std;
using namespace gtsam;
using symbol_shorthand::X;  // Pose3 (x,y,z,r,p,y)

// Factor types definition
enum FactorType {
    PRIOR = 0,
    BETWEEN = 1
};

// Helper function to read covariance matrix from input stream
Matrix6 readCovarianceMatrix(istringstream &iss) {
    Matrix6 cov;
    for (int r = 0; r < 6; ++r) {
        for (int c = 0; c < 6; ++c) {
            iss >> cov(r, c);
        }
    }
    return cov;
}

// Helper function to create pose from parameters
Pose3 createPose(double x, double y, double z, double roll, double pitch, double yaw) {
    return Pose3(Rot3::RzRyRx(roll, pitch, yaw), Point3(x, y, z));
}

// Helper function to save graph to g2o file
void saveG2oGraph(const NonlinearFactorGraph& graph, const Values& estimate,
                  const string& filename, int iterCount) {
    // Create filename with iteration count
    string countStr = to_string(iterCount);
    string paddedCount = string(5 - countStr.length(), '0') + countStr;
    string fullFilename = filename + "_" + paddedCount + ".g2o";
    // Save to g2o file
    writeG2o(graph, estimate, fullFilename);
    cout << "Saved graph to: " << fullFilename << endl;
}

int main() {
    // Open factor graph file
    const string path = "/home/xchu/test_ws/src/LTLoc/ltloc/src/";
    ifstream infile(path+"factor_graph_data.txt");
    if (!infile.is_open()) {
[... 95 lines cut ...]
```
The output of the code is as follows:
```
Processing line 5715
Before update - Factors: 4515, NR Factors: 10
New factors: 1, New values: 1
After update - Factors: 4516, NR Factors: 10
Processing line 5716
Before update - Factors: 4516, NR Factors: 10
New factors: 1, New values: 1
After update - Factors: 4517, NR Factors: 10
Processing line 5717
Before update - Factors: 4517, NR Factors: 10
New factors: 1, New values: 1
After update - Factors: 4517, NR Factors: 11
Processing line 5718
Before update - Factors: 4517, NR Factors: 11
New factors: 1, New values: 1
After update - Factors: 4518, NR Factors: 10
Saved graph to: /home/xchu/test_ws/src/LTLoc/ltloc/src/isam_05718.g2o
```
The final saved g2o file shows the visualization of the last sliding window, which can be used to verify that the data is real and valid.
[image]

[factor_graph_data.txt](https://github.com/user-attachments/files/18390672/factor_graph_data.txt)

dellaert, 2025-01-12:
Somehow, I can’t find this response on GitHub. But if you would like me
to look at it, it would be great to have a much smaller example. It
seems your text file alone is 3.9 MB :-) Does the problem only occur for
a very large graphs? I understand it might not be easy to create a
minimal repro example.

—
auth/ACQHGSLPKGZ2TX3NN3R255L2KK3M5AVCNFSM6AAAAABU3MNRYGVHI2DSMVQWIX3LMV43OSLTON2WKQ3PNVWWK3TUHMZDKOBVHA3TSNJVHE>.

JokerJohn, 2025-01-13:
@dellaert 
Hi Frank, thank you for your quick response!
To validate the data and the minimal test case, I reorganized the data and removed the original comments, which is why you didn’t see them. In order to test for possible discrepancies under a large number of factors, I’ve provided a real-world dataset with over 5000 factors, totaling 3.7MB. Following your advice, I’ve reduced the [**test data**](https://github.com/user-attachments/files/18393557/factor_graph_data.txt) to 200 factors.

I’ve generated a complete real-world [test data](https://github.com/user-attachments/files/18393557/factor_graph_data.txt) from the `corridor_day` sequence of the [FusionPortable](https://fusionportable.github.io/dataset/fusionportable/) SLAM dataset, containing only prior factors (1) and LiDAR odometry between factors. The prior factors and their covariance come from ICP matching between the first point cloud frame and the prior map, to fix the coordinate system. The relative pose estimates and covariance estimates for the lidar-inertial odometry are obtained from the [FAST-LIO2](https://github.com/hku-mars/FAST_LIO) algorithm.
[image]
The [test data](https://github.com/user-attachments/files/18393557/factor_graph_data.txt) is as follows:
```
1 1645005900.701431751251 0 1 0.005593021030 0.001526481838 -0.001566024571 -0.000225927310 0.000128593690 -0.003791450430 0.000000179286 -0.000000039948 0.000000006043 -0.000000004575 -0.000000022632 0.000000091709 -0.000000039948 0.000000018979 -0.000000001546 0.000000001926 0.000000006078 -0.000000017221 0.000000006043 -0.000000001546 0.000000015351 -0.000000013913 0.000000001925 0.000000001075 -0.000000004575 0.000000001926 -0.000000013913 0.000000561628 -0.000000028062 0.000000012214 -0.000000022632 0.000000006078 0.000000001925 -0.000000028062 0.000000490365 -0.000000005530 0.000000091709 -0.000000017221 0.000000001075 0.000000012214 -0.000000005530 0.000000550990
```
Each row in the [test data](https://github.com/user-attachments/files/18393557/factor_graph_data.txt) represents the following:
```
factor_type | timestamp  | key | x | y | z | roll | pitch | yaw | covariance matrix (6*6)
```
The test code is as follows:
```
#include <iostream>
#include <string>
#include <gtsam/geometry/Pose3.h>
#include <gtsam/nonlinear/NonlinearFactorGraph.h>
#include <gtsam/nonlinear/Values.h>
#include <gtsam/nonlinear/ISAM2.h>
#include <gtsam/slam/PriorFactor.h>
#include <gtsam/slam/BetweenFactor.h>
#include <gtsam_unstable/nonlinear/IncrementalFixedLagSmoother.h>
#include <gtsam/inference/Symbol.h>
#include <gtsam/slam/dataset.h> // For writeG2o

using namespace std;
using namespace gtsam;
using symbol_shorthand::X;  // Pose3 (x,y,z,r,p,y)

// Factor types definition
enum FactorType {
    PRIOR = 0,
    BETWEEN = 1
};

// Helper function to read covariance matrix from input stream
Matrix6 readCovarianceMatrix(istringstream &iss) {
    Matrix6 cov;
    for (int r = 0; r < 6; ++r) {
        for (int c = 0; c < 6; ++c) {
            iss >> cov(r, c);
        }
    }
    return cov;
}

// Helper function to create pose from parameters
Pose3 createPose(double x, double y, double z, double roll, double pitch, double yaw) {
    return Pose3(Rot3::RzRyRx(roll, pitch, yaw), Point3(x, y, z));
}

// Helper function to save graph to g2o file
void saveG2oGraph(const NonlinearFactorGraph &graph, const Values &estimate,
                  const string &filename, int iterCount) {
    // Create filename with iteration count
    string countStr = to_string(iterCount);
    string paddedCount = string(5 - countStr.length(), '0') + countStr;
    string fullFilename = filename + "_" + paddedCount + ".g2o";
    // Save to g2o file
    writeG2o(graph, estimate, fullFilename);
    cout << "\nSaved graph to: " << fullFilename << endl;
}

/**
 * Data Format
 * PRIOR factor: factor_type timestamp key pose(x y z roll pitch yaw) cov(6*6)
 * BETWEEN factor: factor_type timestamp key1 key2 pose(x y z r p y) cov(6*6)
[... 111 lines cut ...]
            cerr << "Update failed: " << e.what() << endl;
        }
    }

    // Save final graph
    saveG2oGraph(smoother.getFactors(), currentEstimate, path + "isam", lineCount);
    infile.close();

    return 0;
}
```

I tested the minimal case in the environment of **GTSAM 4.2.0** and **Ubuntu 20.04**, with the sliding window set to **1s**. I added a LiDAR odometry factor every 0.1s, so I expected the number of factors to stay around 10-11 after each sliding window update, with many nullptr factors being released. However, when I set `findUnusedFactorSlots=false`, the results were as follows:
[image]

Results with `findUnusedFactorSlots=true`:
[image]
Then I print the graph using `smoother.getFactors().print();` It correctly performed automatic marginalization, but retained a large number of nullptr factors.

[image]

I obtained the same output on another computer using **GTSAM 4.0.0** and **Unbuntu 18.04**(`findUnusedFactorSlots=true`).
[image]

I hope this real-world test data and the small test case can help us identify potential issues with the `IncrementalFixedLagSmoother`. It could be due to my limited understanding of GTSAM or a configuration error.

keevindoherty, 2025-01-13:
@JokerJohn Earlier in the thread Frank suggested trying the develop branch instead of the releases; that’s what I’d recommend as well. There we some relevant iSAM2 fixes that landed in 2023 which I’m not sure were part of the 4.2 release (e.g. https://github.com/borglab/gtsam/pull/1599 to give one example).

JokerJohn, 2025-01-14:
Thank you for the update @keevindoherty @dellaert.
I've switched to the develop branch and can confirm that the issue has been resolved - I'm now getting the same results as Frank demonstrated. The iSAM2 behavior is working as expected.
[image]

I'll prepare a clean test case with the relevant code and data to submit to the appropriate branch soon.

Thanks everyone for your help with this!