borglab/gtsam #1454: 3rdparty dependencies aren't all in gtsam/3rdparty

dkogan, 2023-02-11:
Hi. Most of the non-gtsam-written libraries live in gtsam/3rdparty, but I see at least one that's somewhere else: wrap/pybind11. Can this (and any others I missed) be moved together with the others? Ideally none of these would exist in this repo since they're not a part of gtsam. But at least putting them all together in one place would make packaging easier. Thanks

dellaert, 2023-02-12:
@ProfFan 
We pull in wrap as a git-subtree.

dkogan, 2023-02-12:
I'm not sure what that means. The tag tarball github provides has wrap/pybind11. This isn't a part of gtsam, and it's not in the 3rdparty/ directory with everything else.

I'm building the package for Debian, so I'm culling everything that doesn't have sources and everything that's a dependency. So I'm seeing stuff like this.

Thanks

dellaert, 2023-02-12:
It means you should consider it part of GTSAM, at a higher status than Eigen. We pull in a new wrap version once in a while. TL;DR: don't cull it if you want to build the python wrapper.

dkogan, 2023-02-12:
Hi. This is packaging for Debian. Everything that's a dependency comes from packages, and this qualifies. Unless of course this pybind11 is full of gtsam-specific modifications. Is it? Thanks for being responsive. I'm almost done.

varunagrawal, 2023-02-15:
`wrap/pybind11` is not a depedency for GTSAM to build so you should be able to package it without the `wrap` folder. The `wrap` folder contains code we built to automatically generate Python and Matlab bindings for GTSAM. Since it is not a build dependency for the GTSAM library, it is not part of `3rdparty`. Hope that answers your question.

dkogan, 2023-02-16:
Hi. The Debian packages ship everything, including the Python bindings. The Debian package thus uses pybind11. You didn't write wrap/pybind11 and you don't maintain it. All I'm asking is that all components that you didn't write and don't maintain live in the same clearly-identified directory so that people don't need to go looking for these. Thanks.

ProfFan, 2023-02-16:
@dkogan Yeah actually if pybind11 is available in the system packages we can conditionally use the system version. Your suggestion is great but we don't currently have the time for it. Is your problem related to the `src` packaging in Debian? I suggest adding a rule to remove the folder from the tarball when you package as a temporary solution...

dkogan, 2023-02-16:
Hi. Thanks for replying. I'm building the package successfully, and I don't NEED you to do anything. This issue is just a suggestion to reorganize your tree to move wrap/pybind11/ to 3rdparty/. This will make it clearer for anybody looking at your code what you did or did not write. You really shouldn't be shipping any of this 3rdparty/ stuff in the repo, but getting there would mean more work for you.

varunagrawal, 2023-03-10:
Closing as resolved since there is no action to be taken.

closed 2023-03-10 by varunagrawal