borglab/gtsam #1455: Build scripts don't detect boost versions properly

dkogan, 2023-02-11:
Hi. I'm on a very recent Debian/unstable system. I'm trying to build the latest tag of gtsam: 4.2a9. This happens:

```
$ mkdir build; cd build; cmake ..

....
-- Could NOT find Boost: missing: timer (found /usr/lib/x86_64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake (found suitable version "1.74.0", minimum required is "1.65"))
WARNING:  GTSAM timing instrumentation will use the older, less accurate, Boost timer library because boost older than 1.48 was found.

....
```

So some parts of the build think I have boost 1.74.0, and some others think that this is older than 1.48. Everything is stock Debian. Here's the package in question:

```
dima@shorty:~$ dpkg -l libboost-dev
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name               Version      Architecture Description
+++-==================-============-============-=======================================================
ii  libboost-dev:amd64 1.74.0.3     amd64        Boost C++ Libraries development files (default version)
```

Thanks

dellaert, 2023-02-12:
From the issue description I have no idea what exactly is happening or how to fix :-) Happy to help if you are able to provide more information or (better yet) tell us what to fix...

dkogan, 2023-02-12:
I generally avoid cmake (and boost for that matter) like the plague, so I was hoping you'd do the heavy lifting here.

I have the timer package:

```
$ dpkg -l libboost-timer-dev
...
ii  libboost-timer-dev:amd64 1.74.0.3     amd64        C++ wall clock and CPU process timers (default version)
```

The cmake thing that finds it:

```
find_package(Boost ${BOOST_FIND_MINIMUM_VERSION} COMPONENTS ${BOOST_FIND_MINIMUM_COMPONENTS})
```

does not set the global variable

```
Boost_TIMER_LIBRARY
```

The "1.48" business is hardcoded in the gtsam cmake stuff. I have no idea what `find_package` does or how to debug it.

dellaert, 2023-02-12:
I'll track this down. Note that in the bleeding edge (develop) you can now turn off boost altogether - at the cost of having no python wrapper. I'm updating the README file to make that more clear.

dellaert, 2023-02-12:
Try pulling the branch in #1462 and let me know if this resolves your issues? Now require 1.67

dkogan, 2023-02-12:
Thank you. This did the trick. The packaging is almost done. The tree is here: https://salsa.debian.org/science-team/gtsam

I added this commit to the patch stack for this issue: https://github.com/borglab/gtsam/commit/a8f5dcad53846c20ab1c268ea977dfcc8a2d5fb6

closed 2023-02-12 by dkogan