borglab/gtsam #1457: Build isn't clear about when gtsam_unstable should be built

dkogan, 2023-02-11:
Hi. I'm building gtsam for upload to Debian. Thus I'm using the tag tarballs provided by github: https://github.com/borglab/gtsam/tags

These include everything including gtsam_unstable. The build config says that gtsam_unstable should only exist if we're using a git checkout:

https://github.com/borglab/gtsam/blob/b71ed8f91d836475935753343cf9d89f9318a709/cmake/HandleGeneralOptions.cmake#L4

But this isn't true here: I'm using a release tarball, and gtsam_unstable/ exists. I'm guessing that I shouldn't be building and shipping gtsam_unstable. Yes? And can we please fix the comment?

Thanks!

dkogan, 2023-02-12:
A follow-up: apparently the examples are all built unconditionally with -lgtsam_unstable, so building without that library makes the build fail. This happens:

```
$ mkdir obj-x86_64-linux-gnu && cd obj-x86_64-linux-gnu && cmake \
  -DCMAKE_INSTALL_PREFIX=/usr \
  -DCMAKE_BUILD_TYPE=None \
  -DCMAKE_INSTALL_SYSCONFDIR=/etc \
  -DCMAKE_INSTALL_LOCALSTATEDIR=/var \
  -DCMAKE_EXPORT_NO_PACKAGE_REGISTRY=ON \
  -DCMAKE_FIND_USE_PACKAGE_REGISTRY=OFF \
  -DCMAKE_FIND_PACKAGE_NO_PACKAGE_REGISTRY=ON \
  -DFETCHCONTENT_FULLY_DISCONNECTED=ON \
  -DCMAKE_INSTALL_RUNSTATEDIR=/run "-GUnix Makefiles" \
  -DCMAKE_VERBOSE_MAKEFILE=ON \
  -DCMAKE_INSTALL_LIBDIR=lib/x86_64-linux-gnu \
  -DGTSAM_BUILD_UNSTABLE=OFF \
  -DGTSAM_UNSTABLE_BUILD_PYTHON=OFF \
  -DGTSAM_UNSTABLE_INSTALL_MATLAB_TOOLBOX=OFF \
  -DBUILD_SHARED_LIBS=ON \
  -DGTSAM_WITH_TBB=ON \
  -DGTSAM_WITH_EIGEN_MKL=OFF \
  -DGTSAM_WITH_EIGEN_MKL_OPENMP=OFF \
  -DGTSAM_BUILD_PYTHON=OFF \
  -DGTSAM_INSTALL_MATLAB_TOOLBOX=OFF \
  -DGTSAM_ALLOW_DEPRECATED_SINCE_V42=ON \
  -DGTSAM_BUILD_WITH_CCACHE=ON \
  -DGTSAM_USE_SYSTEM_EIGEN=ON \
  -DGTSAM_USE_SYSTEM_METIS=ON \
  .. && make

....

cd /home/dima/debianstuff/gtsam/obj-x86_64-linux-gnu/examples && /usr/bin/cmake -E cmake_link_script CMakeFiles/CameraResectioning.dir/link.txt --verbose=1
ccache /usr/lib/ccache/c++ -rdynamic CMakeFiles/CameraResectioning.dir/CameraResectioning.cpp.o -o CameraResectioning  -Wl,-rpath,/home/dima/debianstuff/gtsam/obj-x86_64-linux-gnu/gtsam ../gtsam/libgtsam.so.4.2.0 -lgtsam_unstable /usr/lib/x86_64-linux-gnu/libboost_program_options.so.1.74.0 /usr/lib/x86_64-linux-gnu/libboost_serialization.so.1.74.0 /usr/lib/x86_64-linux-gnu/libboost_system.so.1.74.0 /usr/lib/x86_64-linux-gnu/libboost_filesystem.so.1.74.0 /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.74.0 /usr/lib/x86_64-linux-gnu/libboost_atomic.so.1.74.0 /usr/lib/x86_64-linux-gnu/libboost_date_time.so.1.74.0 /usr/lib/x86_64-linux-gnu/libboost_regex.so.1.74.0 /usr/lib/x86_64-linux-gnu/libboost_timer.so.1.74.0 /usr/lib/x86_64-linux-gnu/libboost_chrono.so.1.74.0 /usr/lib/x86_64-linux-gnu/libtbb.so /usr/lib/x86_64-linux-gnu/libtbbmalloc.so -lmetis 
/bin/ld: cannot find -lgtsam_unstable: No such file or directory
collect2: error: ld returned 1 exit status
examples/CMakeFiles/CameraResectioning.dir/build.make:113: *** [examples/CameraResectioning] error 1
```

This is the 4.2a9 tag, with some patches applied to use the system dependencies: https://salsa.debian.org/science-team/gtsam/-/tree/master/debian/patches

dkogan, 2023-02-12:
Confirmed. This fails with the unmodified 4.2a9 tarball:

```
mkdir build && cd build && cmake -DGTSAM_BUILD_UNSTABLE=OFF .. && make
```

Eventually you get:

```
[ 74%] Linking CXX executable HMMExample
....
/bin/ld: cannot find -lgtsam_unstable: No such file or directory
```

dellaert, 2023-02-12:
Ah, interesting. I guess some examples need it and so this flag was added thoughtlessly. Let me see what the fix is...

dellaert, 2023-02-12:
I tried removing gtsam_unstable from examples/CMakeLists.txt and that seems to just work. If you conform that works for you I'll patch the impending 4.2. Thanks !!!

dkogan, 2023-02-12:
I tried to simply remove gtsam_unstable from the example builds, like you suggested, and it appears to work. Thanks!

closed 2023-02-12 by dkogan