borglab/gtsam #1463: The build system installs the CppUnitLite library, even though it's a build-time thing only

dkogan, 2023-02-12:
Hi. I found another build quirk. Currently the CMake stuff distributes libCppUnitLite.a and /usr/include/CppUnitLite/. These are just to make the unit tests work, right? If so, we should build them, build the tests, but NOT install them. Am I understanding them properly? If we DO need to install them, they should be namespaced: gtsam-CppUnitLite or something like that.

Thanks!

varunagrawal, 2023-04-26:
We install CppUnitLite since then we can use the unit testing framework in downstream code bases as well.

dkogan, 2023-04-27:
Then can you please split this into a separate library? If it's useful for other non-gtsam codebases, you shouldn't ask them to install gtsam just to get the testing framework.

Thanks