borglab/gtsam #1471: cmake build errors while building gtsam in docker container -

shreyasaha1997, 2023-02-17:
## Description

I am unable to cmake build gtsam in my docker container and below is the following error I get - 

## Dockerfile Code - 

```
FROM nvidia/cuda:11.6.0-cudnn8-devel-ubuntu20.04

ENV DEBIAN_FRONTEND noninteractive

RUN apt-get update && apt-get install -y \
	git \
	python3 \
	python3-pip \
    && rm -rf /var/lib/apt/lists/*

RUN pip install torch==1.12.1+cu116 torchvision==0.13.1+cu116 --extra-index-url https://download.pytorch.org/whl/cu116

WORKDIR /NeRF-SLAM

COPY NeRF-SLAM/requirements.txt requirements.txt
RUN pip install -r requirements.txt
RUN pip install opencv-python open3d

RUN pip install --no-cache-dir torch-scatter -f https://pytorch-geometric.com/whl/torch-1.12.0+cu116.html

RUN apt-get update && apt-get install ffmpeg libsm6 libxext6  -y
RUN pip install git+https://github.com/princeton-vl/lietorch.git
RUN pip install cmake --upgrade
RUN apt-get update && apt-get -y install libboost-all-dev
RUN apt-get update && apt-get -y install xorg-dev
RUN apt-get update && apt-get -y install libglew-dev
COPY . .

RUN cmake NeRF-SLAM/thirdparty/gtsam -DGTSAM_BUILD_PYTHON=1 -DGTSAM_WITH_TBB=OFF -DGTSAM_WITH_EIGEN_MKL=OFF -DGTSAM_WITH_EIGEN_MKL_OPENMP=OFF -B build_gtsam 

RUN cmake --build build_gtsam

RUN cd build_gtsam
RUN make python-install

RUN cmake NeRF-SLAM/thirdparty/instant-ngp -B build_ngp
RUN cmake --build build_ngp 

RUN python3 NeRF-SLAM/setup.py install
CMD [ "python3", "NeRF-SLAM/examples/slam_demo.py", "--dataset_dir=./datasets/Replica/office0", "--dataset_name=nerf", "--buffer=100", "--slam", "--img_stride=2", "--fusion=nerf", "--multi_gpu", "--gui"]
```

## Error Stack Trace

trimmed the logs here due to word limit. The error starts for the following file - 

[ 97%] Building CXX object python/CMakeFiles/gtsam_py.dir/linear.cpp.o
In file included from /usr/include/x86_64-linux-gnu/c++/9/bits/c++allocator.h:33,
                 from /usr/include/c++/9/bits/allocator.h:46,
                 from /usr/include/c++/9/memory:63,
                 from /usr/include/boost/config/no_tr1/memory.hpp:21,
                 from /usr/include/boost/smart_ptr/shared_ptr.hpp:23,
                 from /usr/include/boost/shared_ptr.hpp:17,
                 from /NeRF-SLAM/build_gtsam/python/linear.cpp:13:
/usr/include/c++/9/ext/new_allocator.h: In instantiation of ‘class __gnu_cxx::new_allocator<const Eigen::Matrix<double, -1, -1>&>’:
/usr/include/c++/9/bits/allocator.h:111:11:   required from ‘class std::allocator<const Eigen::Matrix<double, -1, -1>&>’
/usr/include/c++/9/bits/stl_vector.h:84:21:   required from ‘struct std::_Vector_base<const Eigen::Matrix<double, -1, -1>&, std::allocator<const Eigen::Matrix<double, -1, -1>&> >’
/usr/include/c++/9/bits/stl_vector.h:386:11:   required from ‘class std::vector<const Eigen::Matrix<double, -1, -1>&>’
/usr/include/c++/9/type_traits:883:12:   required from ‘struct std::is_constructible<gtsam::HessianFactor, const std::vector<long unsigned int, std::allocator<long unsigned int> >&, const std::vector<const Eigen::Matrix<double, -1, -1, 0, -1, -1>&, std::allocator<const Eigen::Matrix<double, -1, -1, 0, -1, -1>&> >&, const std::vector<const Eigen::Matrix<double, -1, 1, 0, -1, 1>&, std::allocator<const Eigen::Matrix<double, -1, 1, 0, -1, 1>&> >&, double>’
/NeRF-SLAM/NeRF-SLAM/thirdparty/gtsam/wrap/pybind11/include/pybind11/detail/init.h:231:47:   required by substitution of ‘template<class Class, class ... Extra, typename std::enable_if<(Class::has_alias && (! std::is_constructible<typename Class::type, const std::vector<long unsigned int, std::allocator<long unsigned int> >&, const std::vector<const Eigen::Matrix<double, -1, -1, 0, -1, -1>&, std::allocator<const Eigen::Matrix<double, -1, -1, 0, -1, -1>&> >&, const std::vector<const Eigen::Matrix<double, -1, 1, 0, -1, 1>&, std::allocator<const Eigen::Matrix<double, -1, 1, 0, -1, 1>&> >&, double>::value)), int>::type <anonymous> > static void pybind11::detail::initimpl::constructor<const std::vector<long unsigned int, std::allocator<long unsigned int> >&, const std::vector<const Eigen::Matrix<double, -1, -1, 0, -1, -1>&, std::allocator<const Eigen::Matrix<double, -1, -1, 0, -1, -1>&> >&, const std::vector<const Eigen::Matrix<double, -1, 1, 0, -1, 1>&, std::allocator<const Eigen::Matrix<double, -1, 1, 0, -1, 1>&> >&, double>::execute<Class, Extra ..., <enumerator> >(Class&, const Extra& ...) [with Class = pybind11::class_<gtsam::HessianFactor, gtsam::GaussianFactor, boost::shared_ptr<gtsam::HessianFactor> >; Extra = {pybind11::arg, pybind11::arg, pybind11::arg, pybind11::arg}; typename std::enable_if<(Class::has_alias && (! std::is_constructible<typename Class::type, const std::vector<long unsigned int, std::allocator<long unsigned int> >&, const std::vector<const Eigen::Matrix<double, -1, -1, 0, -1, -1>&, std::allocator<const Eigen::Matrix<double, -1, -1, 0, -1, -1>&> >&, const std::vector<const Eigen::Matrix<double, -1, 1, 0, -1, 1>&, std::allocator<const Eigen::Matrix<double, -1, 1, 0, -1, 1>&> >&, double>::value)), int>::type <anonymous> = <missing>]’
/NeRF-SLAM/NeRF-SLAM/thirdparty/gtsam/wrap/pybind11/include/pybind11/pybind11.h:1596:9:   required from ‘pybind11::class_<type_, options>& pybind11::class_<type_, options>::def(const pybind11::detail::initimpl::constructor<Args ...>&, const Extra& ...) [with Args = {const std::vector<long unsigned int, std::allocator<long unsigned int> >&, const std::vector<const Eigen::Matrix<double, -1, -1, 0, -1, -1>&, std::allocator<const Eigen::Matrix<double, -1, -1, 0, -1, -1>&> >&, const std::vector<const Eigen::Matrix<double, -1, 1, 0, -1, 1>&, std::allocator<const Eigen::Matrix<double, -1, 1, 0, -1, 1>&> >&, double}; Extra = {pybind11::arg, pybind11::arg, pybind11::arg, pybind11::arg}; type_ = gtsam::HessianFactor; options = {gtsam::GaussianFactor, boost::shared_ptr<gtsam::HessianFactor>}]’
/NeRF-SLAM/build_gtsam/python/linear.cpp:400:200:   required from here
/usr/include/c++/9/ext/new_allocator.h:63:26: error: forming pointer to reference type ‘const Eigen::Matrix<double, -1, -1>&’
[... 75 lines cut ...]
/NeRF-SLAM/build_gtsam/python/linear.cpp:400:200:   required from here
/usr/include/c++/9/bits/stl_vector.h:471:20: error: no members matching ‘std::vector<const Eigen::Matrix<double, -1, -1>&>::_Base {aka std::_Vector_base<const Eigen::Matrix<double, -1, -1>&, std::allocator<const Eigen::Matrix<double, -1, -1>&> >}::_M_allocate’ in ‘std::vector<const Eigen::Matrix<double, -1, -1>&>::_Base’ {aka ‘struct std::_Vector_base<const Eigen::Matrix<double, -1, -1>&, std::allocator<const Eigen::Matrix<double, -1, -1>&> >’}
  471 |       using _Base::_M_allocate;
      |                    ^~~~~~~~~~~
/usr/include/c++/9/bits/stl_vector.h:472:20: error: no members matching ‘std::vector<const Eigen::Matrix<double, -1, -1>&>::_Base {aka std::_Vector_base<const Eigen::Matrix<double, -1, -1>&, std::allocator<const Eigen::Matrix<double, -1, -1>&> >}::_M_deallocate’ in ‘std::vector<const Eigen::Matrix<double, -1, -1>&>::_Base’ {aka ‘struct std::_Vector_base<const Eigen::Matrix<double, -1, -1>&, std::allocator<const Eigen::Matrix<double, -1, -1>&> >’}
  472 |       using _Base::_M_deallocate;
      |                    ^~~~~~~~~~~~~
/usr/include/c++/9/bits/stl_vector.h:1165:7: error: forming pointer to reference type ‘const Eigen::Matrix<double, -1, -1>&’
 1165 |       data() _GLIBCXX_NOEXCEPT
      |       ^~~~
/usr/include/c++/9/bits/stl_vector.h:1169:7: error: forming pointer to reference type ‘const Eigen::Matrix<double, -1, -1>&’
 1169 |       data() const _GLIBCXX_NOEXCEPT
      |       ^~~~
/usr/include/c++/9/bits/stl_vector.h:1200:7: error: ‘void std::vector<_Tp, _Alloc>::push_back(std::vector<_Tp, _Alloc>::value_type&&) [with _Tp = const Eigen::Matrix<double, -1, -1>&; _Alloc = std::allocator<const Eigen::Matrix<double, -1, -1>&>; std::vector<_Tp, _Alloc>::value_type = const Eigen::Matrix<double, -1, -1>&]’ cannot be overloaded with ‘void std::vector<_Tp, _Alloc>::push_back(const value_type&) [with _Tp = const Eigen::Matrix<double, -1, -1>&; _Alloc = std::allocator<const Eigen::Matrix<double, -1, -1>&>; std::vector<_Tp, _Alloc>::value_type = const Eigen::Matrix<double, -1, -1>&]’
 1200 |       push_back(value_type&& __x)
      |       ^~~~~~~~~
/usr/include/c++/9/bits/stl_vector.h:1184:7: note: previous declaration ‘void std::vector<_Tp, _Alloc>::push_back(const value_type&) [with _Tp = const Eigen::Matrix<double, -1, -1>&; _Alloc = std::allocator<const Eigen::Matrix<double, -1, -1>&>; std::vector<_Tp, _Alloc>::value_type = const Eigen::Matrix<double, -1, -1>&]’
 1184 |       push_back(const value_type& __x)
      |       ^~~~~~~~~
make[2]: *** [python/CMakeFiles/gtsam_py.dir/build.make:251: python/CMakeFiles/gtsam_py.dir/linear.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:32314: python/CMakeFiles/gtsam_py.dir/all] Error 2
make: *** [Makefile:166: all] Error 2
The command '/bin/sh -c cmake --build build_gtsam' returned a non-zero code: 2

Any help would be greatly appreciated!

closed 2023-02-19 by shreyasaha1997

Terry10086, 2023-05-29:
I am facing the same issue, how did you solve it?

josyulavt, 2024-08-02:
+1 any solution for this?