borglab/gtsam #1476: Enabling MATLAB Toolbox upon Compilation is Failing

dlolson512, 2023-02-20:
## Description

With reference to branch release/4.2a8, enabling the MATLAB toolbox installation is failing. After making and jumping into the build directory, running the command:

```
cmake .. -DGTSAM_INSTALL_MATLAB_TOOLBOX=ON
```

based on [this documentation](https://github.com/borglab/gtsam/blob/develop/matlab/README.md) results in the following error:

[image]

It appears I am not the only one having this issue, given that I've seen it asked in the GTSAM Google group ([for example](https://groups.google.com/g/gtsam-users/c/nNlgjOY-lhg/m/Whi4a8gSCAAJ)) multiple times without any helpful responses.

Can someone provide a clear answer on how to enable to MATLAB toolbox? If so, can we update the documentation to help out the other MATLAB users? Thanks!

## Steps to reproduce

1. Clone Repo
2. "git checkout release/4.2a8"
3. Jump into "gtsam" directory, then:

```
mkdir build
cd build
cmake .. -DGTSAM_INSTALL_MATLAB_TOOLBOX=ON
```
And that is where the fail occurs. 

**NOTE**: A basic build with no flags seems to work just fine. The issue is enabling the MATLAB toolbox. 

## Environment

System Configuration:
- OS: Ubuntu 22.04 LTS
- MATLAB: R2022b
- CMake: Ver 3.25
- MEX Compiler: g++ v11.3

## Additional information

If this issue is solved, I'd be happy to contribute to the new README.md under the matlab/ folder to help others avoid the same pitfalls.

mattking-smith, 2023-03-13:
Where do you have MATLAB installed? Standard location?

varunagrawal, 2023-04-26:
@dlolson512 reminder. Can you please let us know where have you installed Matlab? It seems like this is a configuration problem on your machine where Cmake can't find Matlab.

varunagrawal, 2023-05-01:
Closing this due to lack of a response. I recommend asking this question on the Google Group since this is more of a tech support question than an issue with the Matlab toolbox.

closed 2023-05-01 by varunagrawal