borglab/gtsam #1477: Matlab wrapper, Values::update with Point2 fails

la-sve, 2023-02-24:
## Description
A gtsam Point2 type in Matlab can not be updated within a Values class instance. There is some conflict with dynamic and static matrix types.

## Steps to reproduce

Use GTSAM 4.2a9 with Matlab Toolbox compiled, start Matlab and add the proper paths for using the GTSAM Matlab wrapper.

### Example 1:
`a=gtsam.Values(); a.insert(0,[1;3]);a.update(0,[3;3]);`

**Results in:**
```
Error using gtsam_wrapper
Exception from gtsam:
Attempting to retrieve value with key "0", type stored in Values is
gtsam::GenericValue<Eigen::Matrix<double, -1, 1, 0, -1, 1> > but requested type was
gtsam::GenericValue<Eigen::Matrix<double, 2, 1, 0, 2, 1> >

Error in gtsam.Values/update (line 1395)
        gtsam_wrapper(2567, this, varargin{:});
```

### Example 2:
`a=gtsam.Values(); a.insert(0,gtsam.Point2(1,3));a.update(0,gtsam.Point2(3,3));`

**Results in:**
```
Error using gtsam_wrapper
Exception from gtsam:
Attempting to retrieve value with key "0", type stored in Values is
gtsam::GenericValue<Eigen::Matrix<double, -1, 1, 0, -1, 1> > but requested type was
gtsam::GenericValue<Eigen::Matrix<double, 2, 1, 0, 2, 1> >

Error in gtsam.Values/update (line 1395)
        gtsam_wrapper(2567, this, varargin{:});
```

## Expected behavior

The value at index 0 should be updated without error.

## Environment

- Ubuntu 20.04
- GTSAM 4.2a9
- Matlab R2022a

ProfFan, 2023-03-02:
I think this is related to a similar issue in Python, we should already have checks in the C++ side that does conversion, at least for Point3...

mattking-smith, 2023-03-13:
@ProfFan What is the difference between `a.insert()` versus `a.insert_or_assign()`? 

To me on the `MATLAB` side if I call the code with `a.insert()` land try an update with `a.update()` like @la-sve I see the same errors he is reporting. 

But if I call: 
```
a.insert_or_assign(0,[2;3])
``` 
then go for a values update with 
```
a.update(0,[1;3])
```
the `GTSAM` gods are happy and no error is thrown. What gives?

ProfFan, 2023-03-13:
Still a type overload problem

dellaert, 2023-06-04:
I don't use MATLAB anymore - an issue perhaps for @mattking-smith with @varunagrawal to nail down :-)

varunagrawal, 2023-10-10:
So I figured this out. The wrapper has the overloaded `Values::insert` methods ordered as:

```cpp
void insert(size_t j, Matrix matrix);
void insert(size_t j, Vector vector);
void insert(size_t j, const gtsam::Point2& point2);
```

but the overloaded `Values::update` methods are ordered as:

```cpp
void insert(size_t j, const gtsam::Point2& point2);
void insert(size_t j, Matrix matrix);
void insert(size_t j, Vector vector);
```

When calling those methods, for `insert` the version with Vector is the first one matched, which is why the type of the stored object is `Eigen::Matrix<double, -1, 1, 0, -1, 1>`, but when calling `update`, the `Point2` version is the first match and ends up being called.

The fix is to just re-order the `insert` method so that the Point2 version is before the vector version.

varunagrawal, 2023-10-10:
@la-sve check it out. This is the fix working on my local machine (Mac M1 no less)

[image]

varunagrawal, 2023-10-18:
Fixed by #1654

PR #1654 (2023-10-12): Wrapper Fixes
This PR fixes #1477 with a simple re-arrangement of the order of the `Values::insert` method.

Also using this PR to piggyback a fix for #1648.
closed 2023-10-18 by varunagrawal