borglab/gtsam #1478: Matlab wrapper cannot be complied

bowei20222023, 2023-02-27:
My computer is Macbook Pro 16inch with M1 pro. 
I am trying to install with Matlab wrapper by using the following command:
cmake  -DGTSAM_INSTALL_MATLAB_TOOLBOX=ON  ..

Error shown:
-- GTSAM Version: 4.3a0
-- GTSAM is a shared library due to BUILD_SHARED_LIBS is ON
-- GTSAM_POSE3_EXPMAP=ON, enabling GTSAM_ROT3_EXPMAP as well
-- Found Eigen version: 3.4.0
-- checking for thread-local storage - found
-- Could NOT find MKL (missing: MKL_INCLUDE_DIR MKL_LIBRARIES) 
-- Could NOT find OpenMP_C (missing: OpenMP_C_FLAGS OpenMP_C_LIB_NAMES) 
-- Could NOT find OpenMP_CXX (missing: OpenMP_CXX_FLAGS OpenMP_CXX_LIB_NAMES) 
-- Could NOT find OpenMP (missing: OpenMP_C_FOUND OpenMP_CXX_FOUND) 
-- Could NOT find TBB (missing: TBB_INCLUDE_DIRS TBB_LIBRARIES tbb tbbmalloc) (Required is at least version "4.4")
-- Building 3rdparty
-- Could NOT find GeographicLib (missing: GeographicLib_LIBRARY_DIRS GeographicLib_LIBRARIES GeographicLib_INCLUDE_DIRS) 
-- Building base
-- Building basis
-- Building geometry
-- Building inference
-- Building symbolic
-- Building discrete
-- Building hybrid
-- Building linear
-- Building nonlinear
-- Building sam
-- Building sfm
-- Building slam
-- Building navigation
-- GTSAM Version: 4.3.0
-- Install prefix: /usr/local
-- Building GTSAM - as a SHARED library
-- Building base_unstable
-- Building geometry_unstable
-- Building linear_unstable
-- Building discrete_unstable
-- Building dynamics_unstable
-- Building nonlinear_unstable
-- Building slam_unstable
-- Building partition_unstable
-- GTSAM_UNSTABLE Version: 4.3.0
-- Install prefix: /usr/local
-- gtwrap Package config : /usr/local/lib/cmake/gtwrap
-- gtwrap version        : 1.0
-- gtwrap CMake path     : /usr/local/lib/cmake/gtwrap
-- gtwrap library path   : /usr/local/lib/gtwrap
-- gtwrap binary path    : /usr/local/bin/gtwrap
-- gtwrap header path    : /usr/local/include/gtwrap
CMake Error at /opt/homebrew/Cellar/cmake/3.25.2/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Matlab (missing: Matlab_MEX_LIBRARY Matlab_MX_LIBRARY
  MEX_COMPILER) (found version "9.13")
Call Stack (most recent call first):
  /opt/homebrew/Cellar/cmake/3.25.2/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
  /opt/homebrew/Cellar/cmake/3.25.2/share/cmake/Modules/FindMatlab.cmake:2038 (find_package_handle_standard_args)
  wrap/cmake/MatlabWrap.cmake:10 (find_package)
  wrap/cmake/MatlabWrap.cmake:67 (find_and_configure_matlab)
  matlab/CMakeLists.txt:85 (matlab_wrap)

Can you help me out? Thanks in advance

bowei20222023, 2023-03-01:
No, it is a standard place under Applications folder. 
It can be found using
ls /Applications/MATLAB_R2022b.app/bin

bowei20222023, 2023-03-01:
This is the results of ls

% ls /Applications/MATLAB_R2022b.app/bin
activate_matlab.sh	lcdata_utf8.xml		mcc
controllingProduct.json	ldd			mex
crash_analyzer.cfg	m3iregistry		mexext
deactivate_matlab.sh	maci64			mw_mpiexec
deploytool		matlab			mw_smpd
icutzdata		matlab-glselector.sh	mwpython
lcdata.xml		matlab_jenv		util
lcdata.xsd		mbuild			worker

mikesheffler, 2023-03-02:
I'm not sure what's going on here, but a couple things that might be worth thinking about:

1) R2022b is fairly new. Is your `CMake` install up to date? I'm looking at the line

> CMake Error at /opt/homebrew/Cellar/cmake/3.25.2/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:230

and wondering if FindPackage (or whatever it's dispatching) knows how to look for your version of MATLAB.

2) Maybe MATLAB can't match up the (MATLAB) library resources for your compiler? On Windows, there's a path similar to

`C:\Program Files\MATLAB\R2022a\bin\win64\mexopts`

that contains .xml files for all of the compilers that version of MATLAB knows how to work with, Sometimes when a newer compiler is used with an older version of MATLAB, that older version of MATLAB doesn't know how to coordinate with the newer compiler (imagine MATLAB 2018a and Visual Studio 2022 -- that version of Visual Studio didn't exist when that version of MATLAB shipped).

Anyway, I don't have any great ideas, but your MATLAB might indeed be installed in a 'normal' place and the build system doesn't think it can find it because an issue with CMake, MATLAB, or the compiler knowing about one another.

Good luck.

bowei20222023, 2023-03-02:
I updated Cmake and the error now is 

CMake Error at matlab/CMakeLists.txt:46 (message):
  GTSAM_INSTALL_MATLAB_TOOLBOX and BUILD_SHARED_LIBS=OFF.The MATLAB wrapper
  cannot be compiled with a static GTSAM library because mex modules are
  themselves shared libraries.If you want a self-contained mex module, enable
  GTSAM_MEX_BUILD_STATIC_MODULE instead of BUILD_SHARED_LIBS=OFF.

bowei20222023, 2023-03-03:
I tried to install an early version of Matlab 2018b. 
But the issues exist:

They either give me the error like

CMake Error at /opt/homebrew/Cellar/cmake/3.25.2/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
Could NOT find Matlab (missing: Matlab_MEX_LIBRARY Matlab_MX_LIBRARY
MEX_COMPILER) (found version "9.13")
Call Stack (most recent call first):
/opt/homebrew/Cellar/cmake/3.25.2/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
/opt/homebrew/Cellar/cmake/3.25.2/share/cmake/Modules/FindMatlab.cmake:2038 (find_package_handle_standard_args)
wrap/cmake/MatlabWrap.cmake:10 (find_package)
wrap/cmake/MatlabWrap.cmake:67 (find_and_configure_matlab)
matlab/CMakeLists.txt:85 (matlab_wrap)

or 
CMake Error at matlab/CMakeLists.txt:46 (message):
GTSAM_INSTALL_MATLAB_TOOLBOX and BUILD_SHARED_LIBS=OFF.The MATLAB wrapper
cannot be compiled with a static GTSAM library because mex modules are
themselves shared libraries.If you want a self-contained mex module, enable
GTSAM_MEX_BUILD_STATIC_MODULE instead of BUILD_SHARED_LIBS=OFF.

mikesheffler, 2023-03-03:
This error:

```cmake
CMake Error at matlab/CMakeLists.txt:46 (message):
GTSAM_INSTALL_MATLAB_TOOLBOX and BUILD_SHARED_LIBS=OFF.The MATLAB wrapper
cannot be compiled with a static GTSAM library because mex modules are
themselves shared libraries.If you want a self-contained mex module, enable
GTSAM_MEX_BUILD_STATIC_MODULE instead of BUILD_SHARED_LIBS=OFF.
```

is from GTSAM (you can tell that from the relative path `matlab/CMakeLists.txt:46`).

The wording could probably be clearer, but what it is trying to tell you is that compiling _GTSAM_ as a static library (something with a `.a` suffix that includes symbols for many of the libraries on which _it_ relies) is not compatible with building the MATLAB toolbox.

If you are just trying to use GTSAM via the MATLAB toolbox, you'll need to compile _GTSAM_ as a shared library (one with a `.dylib` suffix), and to do that you're going to need `-DBUILD_SHARED_LIBS=ON` as one of your cmake arguments (I thought that was the default option, but I'm not 100% sure).

You _may need other options_, but the cmake flags might look (in part) something like this:

```bash
$ cmake -DBUILD_SHARED_LIBS=ON -DGTSAM_INSTALL_MATLAB_TOOLBOX=ON ..
```

You should _not_ need to specify a value for `-DGTSAM_MEX_BUILD_STATIC_MODULE`, and if you were to supply one, it should be `OFF`.

Try building things that way first. If it works, and you really _do_ need static libraries (either GTSAM or something MATLAB-related), you can work up to that problem later.

mattking-smith, 2023-03-13:
Is this a MATLAB 2022 issue somehow? 

A similar MATLAB toolbox enable problem looks like it has popped up on a Unix OS on issue #1476

varunagrawal, 2023-03-13:
This is definitely a Mac issue. I unfortunately don't have cycles to debug but it's definitely a path related issue.

varunagrawal, 2023-03-17:
I have an update on this. For Matlab 2022, you need to use CMake 2.26 and above. It comes with improved support for finding the necessary mex binaries especially on Apple silicon.
As for the the shared library issue, @mikesheffler has done a wonderful job answering your question. @bowei20222023 can you please try these out and let us know if it works for you?

bowei20222023, 2023-03-17:
Thanks, it works for " cmake -DBUILD_SHARED_LIBS=ON -DGTSAM_INSTALL_MATLAB_TOOLBOX=ON .."
 Just one minor correction, it is CMake 2.36, which need "brew install cmake" to install (not port, as port only support 3.24). 

But right now, it shows the following error, ld: symbol(s) not found for architecture x86_64
Is there any quick fix? or I should start a new thread for this topic?

[  7%] Linking CXX shared library libgtsam.dylib
ld: warning: ignoring file /opt/homebrew/lib/libboost_serialization-mt.dylib, building for macOS-x86_64 but attempting to link with file built for macOS-arm64
ld: warning: ignoring file /opt/homebrew/lib/libboost_system-mt.dylib, building for macOS-x86_64 but attempting to link with file built for macOS-arm64
ld: warning: ignoring file /opt/homebrew/lib/libboost_thread-mt.dylib, building for macOS-x86_64 but attempting to link with file built for macOS-arm64
ld: warning: ignoring file /opt/homebrew/lib/libboost_filesystem-mt.dylib, building for macOS-x86_64 but attempting to link with file built for macOS-arm64
ld: warning: ignoring file /opt/homebrew/lib/libboost_date_time-mt.dylib, building for macOS-x86_64 but attempting to link with file built for macOS-arm64
ld: warning: ignoring file /opt/homebrew/lib/libboost_timer-mt.dylib, building for macOS-x86_64 but attempting to link with file built for macOS-arm64
ld: warning: ignoring file /opt/homebrew/lib/libboost_regex-mt.dylib, building for macOS-x86_64 but attempting to link with file built for macOS-arm64
ld: warning: ignoring file /opt/homebrew/lib/libboost_chrono-mt.dylib, building for macOS-x86_64 but attempting to link with file built for macOS-arm64
ld: warning: ignoring file /opt/homebrew/lib/libboost_atomic-mt.dylib, building for macOS-x86_64 but attempting to link with file built for macOS-arm64
Undefined symbols for architecture x86_64:
  "boost::serialization::typeid_system::extended_type_info_typeid_0::type_register(std::type_info const&)", referenced from:
      boost::serialization::singleton<boost::serialization::extended_type_info_typeid<gtsam::Subgraph> >::get_instance() in SubgraphBuilder.cpp.o
      boost::serialization::singleton<boost::serialization::extended_type_info_typeid<std::__1::vector<gtsam::Subgraph::Edge, std::__1::allocator<gtsam::Subgraph::Edge> > > >::get_instance() in SubgraphBuilder.cpp.o
      boost::serialization::singleton<boost::serialization::extended_type_info_typeid<gtsam::Subgraph::Edge> >::get_instance() in SubgraphBuilder.cpp.o
  "boost::serialization::typeid_system::extended_type_info_typeid_0::type_unregister()", referenced from:
      boost::serialization::detail::singleton_wrapper<boost::serialization::extended_type_info_typeid<gtsam::Subgraph> >::~singleton_wrapper() in SubgraphBuilder.cpp.o
      boost::serialization::detail::singleton_wrapper<boost::serialization::extended_type_info_typeid<gtsam::Subgraph> >::~singleton_wrapper() in SubgraphBuilder.cpp.o
      boost::serialization::extended_type_info_typeid<gtsam::Subgraph>::~extended_type_info_typeid() in SubgraphBuilder.cpp.o
      boost::serialization::extended_type_info_typeid<gtsam::Subgraph>::~extended_type_info_typeid() in SubgraphBuilder.cpp.o
      boost::serialization::detail::singleton_wrapper<boost::serialization::extended_type_info_typeid<std::__1::vector<gtsam::Subgraph::Edge, std::__1::allocator<gtsam::Subgraph::Edge> > > >::~singleton_wrapper() in SubgraphBuilder.cpp.o
      boost::serialization::detail::singleton_wrapper<boost::serialization::extended_type_info_typeid<std::__1::vector<gtsam::Subgraph::Edge, std::__1::allocator<gtsam::Subgraph::Edge> > > >::~singleton_wrapper() in SubgraphBuilder.cpp.o
      boost::serialization::extended_type_info_typeid<std::__1::vector<gtsam::Subgraph::Edge, std::__1::allocator<gtsam::Subgraph::Edge> > >::~extended_type_info_typeid() in SubgraphBuilder.cpp.o
      ...
  "boost::serialization::typeid_system::extended_type_info_typeid_0::extended_type_info_typeid_0(char const*)", referenced from:
      boost::serialization::singleton<boost::serialization::extended_type_info_typeid<gtsam::Subgraph> >::get_instance() in SubgraphBuilder.cpp.o
      boost::serialization::singleton<boost::serialization::extended_type_info_typeid<std::__1::vector<gtsam::Subgraph::Edge, std::__1::allocator<gtsam::Subgraph::Edge> > > >::get_instance() in SubgraphBuilder.cpp.o
      boost::serialization::singleton<boost::serialization::extended_type_info_typeid<gtsam::Subgraph::Edge> >::get_instance() in SubgraphBuilder.cpp.o
  "boost::serialization::typeid_system::extended_type_info_typeid_0::~extended_type_info_typeid_0()", referenced from:
      boost::serialization::singleton<boost::serialization::extended_type_info_typeid<gtsam::Subgraph> >::get_instance() in SubgraphBuilder.cpp.o
      boost::serialization::singleton<boost::serialization::extended_type_info_typeid<std::__1::vector<gtsam::Subgraph::Edge, std::__1::allocator<gtsam::Subgraph::Edge> > > >::get_instance() in SubgraphBuilder.cpp.o
      boost::serialization::singleton<boost::serialization::extended_type_info_typeid<gtsam::Subgraph::Edge> >::get_instance() in SubgraphBuilder.cpp.o
      boost::serialization::detail::singleton_wrapper<boost::serialization::extended_type_info_typeid<gtsam::Subgraph> >::~singleton_wrapper() in SubgraphBuilder.cpp.o
      boost::serialization::detail::singleton_wrapper<boost::serialization::extended_type_info_typeid<gtsam::Subgraph> >::~singleton_wrapper() in SubgraphBuilder.cpp.o
[... cut ...]
      typeinfo for boost::archive::detail::oserializer<boost::archive::text_oarchive, std::__1::vector<gtsam::Subgraph::Edge, std::__1::allocator<gtsam::Subgraph::Edge> > > in SubgraphBuilder.cpp.o
      typeinfo for boost::archive::detail::oserializer<boost::archive::text_oarchive, gtsam::Subgraph> in SubgraphBuilder.cpp.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[3]: *** [gtsam/libgtsam.4.3.0.dylib] Error 1
make[2]: *** [gtsam/CMakeFiles/gtsam.dir/all] Error 2
make[1]: *** [CMakeFiles/check.dir/rule] Error 2
make: *** [check] Error 2

mikesheffler, 2023-03-17:
> But another error happened to the very last step when doing "make install " ... please see the following? ... Should I open another thread?

Not yet -- I think this still makes sense as part of the issue we have been discussing.

This part:

```make
[100%] Linking CXX shared module ../wrap/gtsam_mex/gtsam_wrapper.mexmaci64
ld: warning: ignoring file /Applications/MATLAB_R2022b.app/bin/maci64/libmex.dylib, building for macOS-arm64 but attempting to link with file built for macOS-x86_64
ld: warning: ignoring file /Applications/MATLAB_R2022b.app/bin/maci64/libmx.dylib, building for macOS-arm64 but attempting to link with file built for macOS-x86_64
ld: warning: ignoring file /Applications/MATLAB_R2022b.app/bin/maci64/libmat.dylib, building for macOS-arm64 but attempting to link with file built for macOS-x86_64
Undefined symbols for architecture arm64:
"_mexAtExit", referenced from:
gtsamKeyList_collectorInsertAndMakeBase_0(int, mxArray_tag**, int, mxArray_tag const**) in gtsam_wrapper.cpp.o
```
...

is telling you that there is a linker error (`Undefined symbols ...`). The object file it is putting together when the error occurs is `gtsam_wrapper.cpp.o`. Specifically, the function `gtsamKeyList_collectorInsertAndMakeBase_0` calls the function `_mexAtExist`, and the linker does not have symbols for that function (`_mexAtExist`). In particular, ld does not have symbols in the `arm64` architecture for function `_mexAtExist`.

Before that error is emitted, we can see a _warning_ from the linker:

```make
ld: warning: ignoring file /Applications/MATLAB_R2022b.app/bin/maci64/libmex.dylib, building for macOS-arm64 but attempting to link with file built for macOS-x86_64

```
that it (`ld`) has ignored (among others) the file `libmex.dylib` because `libmex.dylib` has symbols for the `macOS-x86_64` architecture (that's the architecture for Intel-based Macs), but `ld` is currently linking for a shared module (`gtsam_wrapper.mexmaci64`) that has architecture `macOS-arm64` (the Apple M1 Silicon-based Macs).

It looks like there are a few problems that are repeats of the same underlying issue. Even though it's just a warning, it's almost definitely what is _wrong_, but I'm not 100% sure what is _right_.

1) If you have an M1 Mac and are building GTSAM 'natively' as an M1 shared library, then CMake probably has to be convinced to look for the M1-based MATLAB shared libraries that are used in mex functions.
    *  If you _don't_ have an M1-based version of MATLAB, I don't think this will work.
    * If you _do_ have an M1-based version of MATLAB, it is _probably_ the case that the issue is with CMake, or GTSAM's CMake files, or the GTSAM wrapper's CMake files
2) If you have an Intel Mac, you (I _think_) need to build GTSAM 'natively' as an x86_x64 shared library.

Hopefully, that helps.

varunagrawal, 2023-03-17:
Yeah this has to do with lack of support on our end for Apple chips. We need to make some CMake updates to get this to work.

varunagrawal, 2023-06-05:
Adding an update: the Cmake fix for this is straightforward. Unfortunately an M1 supported version of Matlab is required to successfully link against and the version from Mathworks is still in Beta. :(

I'll make the PR later today, but that's the extent of how much we can control. The only thing to do is wait for the good people at Mathworks to make an arm64 compatible release.

PR #1536 (2023-06-09): CMake Updates
1. Support for Apple silicon.
2. Added flag `GTSAM_ENABLE_MEMORY_SANITIZER` to allow for running `libsanitizer`.

Fixes #1478 & #1499.

Tagging @berndpfrommer so he can pitch in too!
closed 2023-06-09 by varunagrawal