borglab/gtsam #1489: Values::insert() doesn't work for some expressions

arutkowski, 2023-03-07:
The following code will only compile successfully if the statement with key 2 is commented out. Otherwise, the compiler generates a rather verbose chain of error messages. The lines with keys 3, 4, and 5 are suitable workarounds, but it seems natural that one should be allowed to pass an expression as an argument to `Values::insert` (as can be done with `norm2`, for example).

Perhaps this should be a consideration for #1404?

Tested with Ubuntu 18.04 and gcc 9.4. 

```
#include <gtsam/geometry/Point2.h>
#include <gtsam/nonlinear/Values.h>

int main() {
  gtsam::Values initialEstimate;

  gtsam::Point2 truth(1.0,0.0);
  gtsam::Point2 noise(0.123,0.111);
  gtsam::Point2 est(truth + noise);
    
  initialEstimate.insert(1,truth);  // OK
  initialEstimate.insert(2,truth + noise);  // error: incomplete type used in nested name specifier
  initialEstimate.insert(3,est);  // OK
  initialEstimate.insert(4,static_cast<const gtsam::Point2&>(truth + noise));  // OK 
  initialEstimate.insert(5,(truth + noise).eval());  // OK 
                        
  double a = gtsam::norm2(truth + noise);  // OK
    
  return 0;
}
```

ProfFan, 2023-03-07:
This is due to `truth+noise` being an Eigen Expression and thus the template substitution will fail. It's more like a problem with Eigen, similar to the `auto` problem

arutkowski, 2023-03-07:
@ProfFan I agree, I understand that `truth+noise` doesn't actually return a `Point2` (which I understand is just an `Eigen::Vector2d`), and I got the idea of using `eval()` from reading the guidance on `auto` in Eigen. I'm just wondering if it would be possible (without being too impractical) to make `Values.insert` work for such expressions. I think all (or at least most) Eigen functions work on such expressions.

arutkowski, 2023-05-02:
Hi @varunagrawal, thanks for the response. I have been thinking that perhaps it would be as simple as providing a template specialization for `Values::add` when the argument is a `CwiseBinaryOp`, then it could just apply the `.eval()` function and call the more general `Values::add` function. Do you think this would work? I haven't tried it myself yet primarily because the template rules in C++ are somewhat sensitive, and because it seemed there was some interest in redesigning the `Values` class. However, I may be willing to proceed if you think this idea could work (although I suspect this is complicated a bit by the use of traits as you describe above). Otherwise, I agree that this issue can be closed.

From a user perspective, I think it can be a little intimidating to get a verbose compiler error for something that has a fairly simple solution, but perhaps a more transparent solution would be a little more user-friendly. The current solution requires knowing that `Point2` is really just an `Eigen::Vector`, and then also knowing how Eigen does lazy evaluation, which may be a bit of a leap for an inexperienced user. Perhaps just having this issue archived here will be helpful to others since I am unaware of similar guidance elsewhere (aside from digging into how GTSAM and Eigen work under the hood).

varunagrawal, 2023-05-08:
Definitely agree on the issue being archived here (in a resolved state since we've come up with a solution) for inexperienced users. It's a shame that Eigen's error messages are verbose and a bit tricky to understand.

I am not sure what the momentum on the redesign of `Values` is. I know there was discussion on merging `Values` and `DiscreteValues` but that was abandoned due to differences in fundamental operations supported. I would recommend going forward with your suggested fix and leaving the redesign issue for someone to tackle in the future.

arutkowski, 2023-05-27:
@varunagrawal I think I have a pretty decent "fix" (or enhancement?) for this issue. Adding the following two functions to the `gtsam::Values` class seems to work:

```
template <typename UnaryOp, typename ValueType>
void insert(gtsam::Key j, const Eigen::CwiseUnaryOp<UnaryOp, const ValueType>& val) {
  Values::insert(j,val.eval());
}

template <typename BinaryOp, typename ValueType1, typename ValueType2>
void insert(gtsam::Key j, const Eigen::CwiseBinaryOp<BinaryOp, const ValueType1, const ValueType2>& val) {
  Values::insert(j,val.eval());
}
```

The binary operator function is used for things like `values.insert(a + b)`, `values.insert(a - b)`, and even expressions formed from a chain of operations such as `values.insert(a + b + c)`, where `a`, `b`, and `c` can be `gtsam::Point2`, `gtsam::Point3`, or even higher dimensional vectors such as `gtsam::Vector4`.

The unary operation function is used for things like `values.insert(-a)` or `values.insert(2*a)`.

There are also nullary and ternary operators in Eigen, but I don't know if they would have any practical use in this context.

I haven't actually added this to `gtsam::Values` yet. Instead, I made a separate class that inherits from `gtsam::Values` and added the above functions (hence the `Values::` in front of `insert` within the functions, which would probably be unnecessary).

If you think this is an acceptable solution, I can go ahead and add it to a branch of gtsam and make a PR. I can't think of any other functions in gtsam that might benefit from this functionality, but if there are any, perhaps those could also be modified in a similar manner.

varunagrawal, 2023-05-27:
This looks great! Can you please add the examples you've shown as unit tests? That PR would be incredible.

dellaert, 2023-06-04:
Closing as a solution was found. @arutkowski please feel free to submit a PR if you like and refer to this issue.

closed 2023-06-04 by dellaert