borglab/gtsam #1490: ISAM2 throws segfault in external libraries

varunagrawal, 2023-03-10:
## Description

With the recent `develop` of GTSAM, if the class definition in an external project has an ISAM2 member object, then the class throws a segmentation fault on construction.

For example, if we add ISAM2 to the class `Statics` in `gtdynamics` like below,
```cpp
class Statics : public Kinematics {
 protected:
  const StaticsParameters p_;  // overrides Base::p_
  gtsam::ISAM2 isam_;

 public:
```

running a test in `testStatics.run` like below will throw a segfault
```cpp
TEST(Statics, Construction) {
  Statics s1;
}
```

I see similar issues in my own downstream code so I am guessing it won't be long before others also encounter this problem.
My guess is that classes in GTSAM other than ISAM2 may also cause this because of the RCA detailed below.

## Root Cause Analysis

I am on MacOS Monterrey (M1 chip) with AppleClang version 14.0.0.14000029, and I haven't tested this on other machines so this might be an Apple specific issue, but it is an issue that needs to be addressed (since @dellaert prefers Apple silicon).

On inspection, it looks like the change that is causing this issue is 9a50233426dd5cc0e10ff513d66c1c1b7c3a6009 where the `GTSAM_CONCEPT_ASSERT` macro was updated to fix `extra-semi-stmt` warnings.
If I revert only that single line https://github.com/borglab/gtsam/commit/9a50233426dd5cc0e10ff513d66c1c1b7c3a6009#diff-3e02cab63180b3fbf323a1f46f34c773a3a8eed23d050e379035dc7c0f3b2781R22, I no longer see the segfault behavior so it is indeed the issue.

I am wondering if there is another way to fix the warning without causing the segfault behavior?

varunagrawal, 2023-03-10:
@dellaert  @ankur-rc I added the `-Wextra-semi` flag to CMake and I am not seeing any warnings due to the `GTSAM_CONCEPT_ASSERT`. Is this warning something that is Verdant specific?

ProfFan, 2023-03-10:
I think Varun is right, `void*(concept)` does not make any sense - it just declares a useless function pointer in the class and waste 8 bytes or more of memory.

varunagrawal, 2023-03-10:
It also explains why the issue I am seeing is a [Pointer Authentication Failure](https://suelan.github.io/2021/03/05/20210305-pointer-authentication-failed). Seems like that function pointer is setting some bit which causes the clang runtime to think something is wrong.

dellaert, 2023-03-10:
This was a quick and possibly misguided attempt to not break the boost concept asserts. If you find a way to do it more cleanly I am very open. But I would prefer if the semi-colons stay as they are in any new solution. I don’t quite understand why it is verdant specific, we use clang as well, but with much stricter warnings.

closed 2023-03-13 by varunagrawal