borglab/gtsam #1492: when git checkout wrap-export and  make chest  94% tests passed, 13 tests failed out of 230

dbdxnuliba, 2023-03-10:
git checkout wrap-export
mkdir build && cd build
cmake -DGTSAM_INSTALL_CYTHON_TOOLBOX:=ON ..
make check # optional, run unit tests
94% tests passed, 13 tests failed out of 230

Total Test time (real) =   3.39 sec

The following tests FAILED:
	 21 - testSerializationBase (Subprocess aborted)
	 27 - testBearingRange (Subprocess aborted)
	 55 - testUnit3 (Subprocess aborted)
	 60 - testSerializationSymbolic (Subprocess aborted)
	 94 - testSerializationLinear (Subprocess aborted)
	101 - testSerializationNonlinear (Subprocess aborted)
	104 - testBearingFactor (Subprocess aborted)
	105 - testBearingRangeFactor (Subprocess aborted)
	106 - testRangeFactor (Subprocess aborted)
	125 - testSmartFactorBase (Subprocess aborted)
	126 - testSmartProjectionCameraFactor (Subprocess aborted)
	127 - testSmartProjectionPoseFactor (Subprocess aborted)
	135 - testImuFactorSerialization (Subprocess aborted)

ProfFan, 2023-03-10:
Why do you want to use this branch?

dellaert, 2023-06-04:
No activity

closed 2023-06-04 by dellaert