borglab/gtsam #1494: Gtsam will throw an exception when optimizing a factor graph with a large number of nodes

CogmanCVN, 2023-03-20:
When gtsam optimizes a factor graph with a large number of nodes, it throws an exception and the program exits directly (for example, there are 6933 nodes in the factor graph). Is there a solution to this problem?

CogmanCVN, 2023-03-20:
The specific exception thrown is in this line of code： isam->update(gtSAMgraph, initialEstimate);

ScottMcMichael, 2023-05-23:
I encountered this earlier and ended up implementing a fix on a fork:  

https://github.com/borglab/gtsam/issues/1434
https://github.com/ScottMcMichael/gtsam/tree/fix_large_graph_serialization_stack_overflow

ProfFan, 2023-05-24:
I think we already fixed the serialization part, in #1441 ?

ScottMcMichael, 2023-05-24:
I haven't gone back and tested anything since I put together my own fix, but just looking at that PR it doesn't seem like the same thing.

dellaert, 2023-06-04:
Most likely fixed by #1441

closed 2023-06-04 by dellaert