borglab/gtsam #1495: Error when installing on Ubuntu 18.04

mpesso1, 2023-03-20:
## Description

Getting an error during install. I was able to successfully install on my mac a few weeks ago but now getting bug when trying to install on Ubuntu 18.04.

## Steps to reproduce

Here is a list of the commands I used that lead up to error:

1. Clone repo

`git clone https://github.com/borglab/gtsam.git`

2. add build file and run cmake

`mkdir build && cd build`
`cmake -DGTSAM_USE_SYSTEM_EIGEN=ON ..`

3. install 

`sudo make install`

4. Get the following error

`
mason@mason-Nitro-AN517-52:~/Code/gtsam/gtsam/build$ sudo make install
[ 16%] Built target metis-gtsam
[ 18%] Built target CppUnitLite
[ 62%] Built target gtsam
[ 62%] Linking CXX executable UGM_chain
../gtsam/libgtsam.so.4.3.0: undefined reference to `std::experimental::filesystem::v1::__cxx11::path::_M_split_cmpts()'
../gtsam/libgtsam.so.4.3.0: undefined reference to `std::experimental::filesystem::v1::status(std::experimental::filesystem::v1::__cxx11::path const&)'
../gtsam/libgtsam.so.4.3.0: undefined reference to `std::experimental::filesystem::v1::__cxx11::path::_M_find_extension() const'
../gtsam/libgtsam.so.4.3.0: undefined reference to `std::experimental::filesystem::v1::__cxx11::path::parent_path() const'
collect2: error: ld returned 1 exit status
examples/CMakeFiles/UGM_chain.dir/build.make:108: recipe for target 'examples/UGM_chain' failed
make[2]: *** [examples/UGM_chain] Error 1
CMakeFiles/Makefile2:25942: recipe for target 'examples/CMakeFiles/UGM_chain.dir/all' failed
make[1]: *** [examples/CMakeFiles/UGM_chain.dir/all] Error 2
Makefile:162: recipe for target 'all' failed
make: *** [all] Error 2

## Environment

As previously noted my os is Ubuntu 18.04 

## Additional information

It is worth mentioning that I am also passing the argument to cmake that it should use the systems installed version of eigen

mpesso1, 2023-03-20:
Just found solution:
added 
`set(CMAKE_CXX_STANDARD_LIBRARIES "-lstdc++fs")`
to the top of CMakeLists.txt file

closed 2023-03-20 by mpesso1

KnightOfTheMoonlight, 2023-03-31:
@mpesso1 Thanks. This works for me too.

edwinstudy, 2023-04-10:
> Just found solution: added `set(CMAKE_CXX_STANDARD_LIBRARIES "-lstdc++fs")` to the top of CMakeLists.txt file

It works for me ! Thanks

hhhhhhhh213, 2023-11-24:
> set(CMAKE_CXX_STANDARD_LIBRARIES "-lstdc++fs")
Thanks. This works for me too.