borglab/gtsam #1497: Normalizing attitude(rotation matrix) and SO3

synnevaag97, 2023-03-23:
Hi,

I simply wonder if the attitude represented by the rotation matrix is normalized regularly either in prediction(preintegration) of the IMU or during the optimization/correction step. I have searched in the source code and can only find that Rot3 has a function to normalize, but cannot see that it is used. I ask this because none of the examples include normalization explicitly. 

If this self-consistency check or normalization is not executed will it not fall out of SO3 as R*R' is not longer the identity over time? Consequently reducing the accuracy of the attitude.

ProfFan, 2023-03-23:
You are right that it is not normalized. We used to have a thread of debating whether to do it or not, but FWIW, unless you are using iSAM etc where the rotations undergo thousands of updates this is not a big issue.

dellaert, 2023-04-06:
It can become an issue in that case, as you point out. I would love to know in what context you encounter it and we can help come up with a solution that does not negatively impact other cases.

dellaert, 2023-04-06:
Sorry for the delay - was busy at Verdant :-)

dellaert, 2023-06-04:
We'll adopt the same policy going forward as the de-factor policy we have now: normalization can be costly to do at every update, so in long-term running algorithms where this might become an issue, (periodic) normalization will be left at the discretion of the user.

closed 2023-06-04 by dellaert