borglab/gtsam #1498: error: no matching function for call to ‘gtsam::PreintegratedImuMeasurements

federista, 2023-03-23:
## error: no matching function for call to ‘gtsam::PreintegratedImuMeasurements

## Steps to reproduce

1. clone the repository into your workspace git clone https://gitlab.kitware.com/keu-computervision/slam.git src --recursive

2. mkdir build && cd build

3. cmake ../src -GNinja -DCMAKE_BUILD_TYPE=Release
4. cmake --build . -j

`
I have followed this [solution ](https://github.com/MIT-SPARK/Kimera-VIO-ROS/issues/12)
but i couldnt get over it. When i try to manually add the cmake flag `-DGTSAM_USE_SYSTEM_EIGEN=ON` 
the cmake gives me the following output
/CMake Warning:
  Manually-specified variables were not used by the project:

    GTSAM_USE_SYSTEM_EIGEN/

I am just looking for a solution to get over this. Your guidance will be highly appreciated.

ProfFan, 2023-03-23:
This is an issue with the downstream library you are using. They are probably not using the latest GTSAM version.

LimHyungTae, 2024-08-16:
Yep, I also experience this error, and error was solved once I downgrade the version of gtsam (from `develop` to `4.2a8`)

ProfFan, 2024-08-17:
Interesting, @LimHyungTae Please try clone a fresh copy of GTSAM, run `cmake -DGTSAM_USE_SYSTEM_EIGEN=ON` in `build` and paste the results?

LimHyungTae, 2024-08-17:
Yep, there is a deprecated error:

```
/home/shapelim/kimera_vio_ws/src/Kimera-VIO/src/initial/CrossCorrTimeAligner.cpp: In member function ‘size_t VIO::CrossCorrTimeAligner::addNewImuDataFrameRate(const std::vector<long int>&, const ImuStampS&, const ImuAccGyrS&)’:
/home/shapelim/kimera_vio_ws/src/Kimera-VIO/src/initial/CrossCorrTimeAligner.cpp:106:79: error: no matching function for call to ‘gtsam::PreintegratedRotation::integrateMeasurement(const Type, const ConstantReturnType, double&)’
  106 |         imu_acc_gyrs.block<3, 1>(3, i), Eigen::Vector3d::Zero(), imu_period_s_);
      |                                                                               ^
In file included from /home/shapelim/kimera_vio_ws/src/Kimera-VIO/include/kimera-vio/initial/CrossCorrTimeAligner.h:16,
                 from /home/shapelim/kimera_vio_ws/src/Kimera-VIO/src/initial/CrossCorrTimeAligner.cpp:15:
/home/shapelim/kimera_vio_ws/devel/include/gtsam/navigation/PreintegratedRotation.h:221:25: note: candidate: ‘void gtsam::PreintegratedRotation::integrateMeasurement(const Vector3&, const Vector3&, double, gtsam::OptionalJacobian<3, 3>, gtsam::OptionalJacobian<3, 3>)’
  221 |   void GTSAM_DEPRECATED integrateMeasurement(
      |                         ^~~~~~~~~~~~~~~~~~~~
/home/shapelim/kimera_vio_ws/devel/include/gtsam/navigation/PreintegratedRotation.h:221:25: note:   candidate expects 5 arguments, 3 provided
/home/shapelim/kimera_vio_ws/src/Kimera-VIO/src/initial/CrossCorrTimeAligner.cpp:116:16: error: no matching function for call to ‘gtsam::PreintegratedRotation::integrateMeasurement(const Type, const ConstantReturnType, const double&)’
  116 |         last_dt);
      |                ^
In file included from /home/shapelim/kimera_vio_ws/src/Kimera-VIO/include/kimera-vio/initial/CrossCorrTimeAligner.h:16,
                 from /home/shapelim/kimera_vio_ws/src/Kimera-VIO/src/initial/CrossCorrTimeAligner.cpp:15:
/home/shapelim/kimera_vio_ws/devel/include/gtsam/navigation/PreintegratedRotation.h:221:25: note: candidate: ‘void gtsam::PreintegratedRotation::integrateMeasurement(const Vector3&, const Vector3&, double, gtsam::OptionalJacobian<3, 3>, gtsam::OptionalJacobian<3, 3>)’
  221 |   void GTSAM_DEPRECATED integrateMeasurement(
      |                         ^~~~~~~~~~~~~~~~~~~~
/home/shapelim/kimera_vio_ws/devel/include/gtsam/navigation/PreintegratedRotation.h:221:25: note:   candidate expects 5 arguments, 3 provided
make[2]: *** [CMakeFiles/kimera_vio.dir/build.make:790: CMakeFiles/kimera_vio.dir/src/initial/CrossCorrTimeAligner.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:799: CMakeFiles/kimera_vio.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
```

ProfFan, 2024-08-18:
You have a different problem, `chimera_vio` depends on 4.2 not `develop`.

closed 2024-08-18 by ProfFan

ArghyaChatterjee, 2024-11-22:
@LimHyungTae , were you able to solve the problem after upgrading to 4.2.a8 ? I am using 4.0.3 with eigen 3.3.7. I am also trying to compile kimera_vio_ros. 

With only catkin build, I get:
```
Errors     << kimera_vio:make /home/arghya/kimera_vio_ws/logs/kimera_vio/build.make.000.log          
In file included from /home/arghya/kimera_vio_ws/src/Kimera-VIO/include/kimera-vio/dataprovider/DataProviderInterface-definitions.h:22,
                 from /home/arghya/kimera_vio_ws/src/Kimera-VIO/src/dataprovider/DataProviderInterface-definitions.cpp:14:
/home/arghya/kimera_vio_ws/devel/include/gtsam/base/Vector.h:75:52: error: static assertion failed: Error: GTSAM was built against a different version of Eigen
   75 |     GTSAM_EIGEN_VERSION_WORLD==EIGEN_WORLD_VERSION &&
      |                                                    ^
In file included from /home/arghya/kimera_vio_ws/devel/include/gtsam/base/Matrix.h:28,
                 from /home/arghya/kimera_vio_ws/devel/include/gtsam/base/Manifold.h:22,
                 from /home/arghya/kimera_vio_ws/devel/include/gtsam/geometry/BearingRange.h:21,
                 from /home/arghya/kimera_vio_ws/devel/include/gtsam/geometry/Pose3.h:22,
                 from /home/arghya/kimera_vio_ws/src/Kimera-VIO/include/kimera-vio/frontend/OdometryParams.h:17,
                 from /home/arghya/kimera_vio_ws/src/Kimera-VIO/src/frontend/OdometryParams.cpp:15:
/home/arghya/kimera_vio_ws/devel/include/gtsam/base/Vector.h:75:52: error: static assertion failed: Error: GTSAM was built against a different version of Eigen
   75 |     GTSAM_EIGEN_VERSION_WORLD==EIGEN_WORLD_VERSION &&
      |                                                    ^
```

With catkin build -DGTSAM_USE_SYSTEM_EIGEN=ON
```
Errors     << kimera_vio:make /home/arghya/kimera_vio_ws/logs/kimera_vio/build.make.002.log          
/home/arghya/kimera_vio_ws/src/Kimera-VIO/src/initial/CrossCorrTimeAligner.cpp: In member function ‘size_t VIO::CrossCorrTimeAligner::addNewImuDataFrameRate(const std::vector<long int>&, const ImuStampS&, const ImuAccGyrS&)’:
/home/arghya/kimera_vio_ws/src/Kimera-VIO/src/initial/CrossCorrTimeAligner.cpp:106:79: error: no matching function for call to ‘gtsam::PreintegratedRotation::integrateMeasurement(const Type, const ConstantReturnType, double&)’
  106 |         imu_acc_gyrs.block<3, 1>(3, i), Eigen::Vector3d::Zero(), imu_period_s_);
      |                                                                               ^
In file included from /home/arghya/kimera_vio_ws/src/Kimera-VIO/include/kimera-vio/initial/CrossCorrTimeAligner.h:16,
                 from /home/arghya/kimera_vio_ws/src/Kimera-VIO/src/initial/CrossCorrTimeAligner.cpp:15:
/home/arghya/kimera_vio_ws/devel/include/gtsam/navigation/PreintegratedRotation.h:221:25: note: candidate: ‘void gtsam::PreintegratedRotation::integrateMeasurement(const Vector3&, const Vector3&, double, gtsam::OptionalJacobian<3, 3>, gtsam::OptionalJacobian<3, 3>)’
  221 |   void GTSAM_DEPRECATED integrateMeasurement(
      |                         ^~~~~~~~~~~~~~~~~~~~
/home/arghya/kimera_vio_ws/devel/include/gtsam/navigation/PreintegratedRotation.h:221:25: note:   candidate expects 5 arguments, 3 provided
/home/arghya/kimera_vio_ws/src/Kimera-VIO/src/initial/CrossCorrTimeAligner.cpp:116:16: error: no matching function for call to ‘gtsam::PreintegratedRotation::integrateMeasurement(const Type, const ConstantReturnType, const double&)’
  116 |         last_dt);
```

I don't see any escape from here. Also, when I upgrade to gtsam 4.2 like this:
```
sudo apt remove libgtsam-dev
sudo add-apt-repository ppa:borglab/gtsam-release-4.2
sudo apt update
sudo apt install libgtsam-dev libgtsam-unstable-dev
```
I see the same error coming in after catkin build:
```
I see the same error coming again:
```
Errors     << kimera_vio:make /home/arghya/kimera_vio_ws/logs/kimera_vio/build.make.000.log          
In file included from /home/arghya/kimera_vio_ws/devel/include/gtsam/base/Matrix.h:28,
                 from /home/arghya/kimera_vio_ws/devel/include/gtsam/base/Manifold.h:22,
                 from /home/arghya/kimera_vio_ws/devel/include/gtsam/base/Lie.h:26,
                 from /home/arghya/kimera_vio_ws/devel/include/gtsam/base/VectorSpace.h:11,
                 from /home/arghya/kimera_vio_ws/devel/include/gtsam/geometry/Point2.h:20,
                 from /home/arghya/kimera_vio_ws/devel/include/gtsam/geometry/Cal3.h:24,
                 from /home/arghya/kimera_vio_ws/devel/include/gtsam/geometry/Cal3DS2_Base.h:22,
                 from /home/arghya/kimera_vio_ws/devel/include/gtsam/geometry/Cal3DS2.h:23,
                 from /home/arghya/kimera_vio_ws/src/Kimera-VIO/include/kimera-vio/frontend/CameraParams.h:17,
                 from /home/arghya/kimera_vio_ws/src/Kimera-VIO/src/frontend/CameraParams.cpp:16:
/home/arghya/kimera_vio_ws/devel/include/gtsam/base/Vector.h:75:52: error: static assertion failed: Error: GTSAM was built against a different version of Eigen
   75 |     GTSAM_EIGEN_VERSION_WORLD==EIGEN_WORLD_VERSION &&
      |                                                    ^
In file included from /home/arghya/kimera_vio_ws/src/Kimera-VIO/include/kimera-vio/dataprovider/DataProviderInterface-definitions.h:22,
                 from /home/arghya/kimera_vio_ws/src/Kimera-VIO/src/dataprovider/DataProviderInterface-definitions.cpp:14:
/home/arghya/kimera_vio_ws/devel/include/gtsam/base/Vector.h:75:52: error: static assertion failed: Error: GTSAM was built against a different version of Eigen
   75 |     GTSAM_EIGEN_VERSION_WORLD==EIGEN_WORLD_VERSION &&
```

What's the escape from this one ?

LimHyungTae, 2024-11-22:
Hey @ArghyaChatterjee , did you put `-DGTSAM_USE_SYSTEM_EIGEN=ON` when you built the repository?

LimHyungTae, 2024-11-22:
If I recall correctly, either the options below might also resolve the issue:

```
catkin config -a --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo -DGTSAM_TANGENT_PREINTEGRATION=OFF -DGTSAM_BUILD_WITH_MARCH_NATIVE=OFF -DOPENGV_BUILD_WITH_MARCH_NATIVE=OFF
``

ArghyaChatterjee, 2024-11-22:
Hey @LimHyungTae, Which version of gtsam and eigen were you using to build this ?

ArghyaChatterjee, 2024-11-22:
I think in the `.rosinstall`, the develop branch of gtsam is cloned and built from source and it's not using the system installed version.

```
- git:
    local-name: gtsam
    uri: git@github.com:borglab/gtsam.git
    version: develop
```

Now, when I do this as you suggested:
```
catkin config -a --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo -DGTSAM_TANGENT_PREINTEGRATION=OFF -DGTSAM_BUILD_WITH_MARCH_NATIVE=OFF -DOPENGV_BUILD_WITH_MARCH_NATIVE=OFF
```
I get the following error:
```
error: static assertion failed: Error: GTSAM was built against a different version of Eigen
   75 |     GTSAM_EIGEN_VERSION_WORLD==EIGEN_WORLD_VERSION &&
```
Now, when I changed to this:
```
catkin config -a --cmake-args -DGTSAM_USE_SYSTEM_EIGEN=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo -DGTSAM_TANGENT_PREINTEGRATION=OFF -DGTSAM_BUILD_WITH_MARCH_NATIVE=OFF -DOPENGV_BUILD_WITH_MARCH_NATIVE=OFF
```
The gtsam used the system eigen but returned the same old error:
```
error: no matching function for call to ‘gtsam::PreintegratedRotation::integrateMeasurement(const Type, const ConstantReturnType, double&)’
  106 |         imu_acc_gyrs.block<3, 1>(3, i), Eigen::Vector3d::Zero(), imu_period_s_);
```
The eigen version that I am using is 3.3.7.

ArghyaChatterjee, 2024-12-11:
I think the problem lies with the version of gtsam being cloned in the Kimera-VIO-ROS repo since it's cloning from the develop branch of this repo and building gtsam from source inside that workspace. 

Inside the `kimera_vio_ros_https.rosinstall`, the correct version should be:
```
- git:
    local-name: gtsam
    uri: https://github.com/borglab/gtsam.git
    version: 4.1
```
Or for `kimera_vio_ros_ssh.rosinstall`

```
- git:
    local-name: gtsam
    uri: git@github.com:borglab/gtsam.git
    version: 4.1
```
Change in that particular class has happened in gtsam version 4.2. So, any version below that is fine. Also the correct flags for the cmake args is the following:
```
catkin config -a --cmake-args -DGTSAM_USE_SYSTEM_EIGEN=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo -DGTSAM_TANGENT_PREINTEGRATION=OFF -DGTSAM_BUILD_WITH_MARCH_NATIVE=OFF -DOPENGV_BUILD_WITH_MARCH_NATIVE=OFF
```  

Hopefully it will help other people in future as well. Thanks.

dellaert, 2024-12-12:
What change exactly broke this in 4.2? Is there something you think should be fixed in the wrapper?

ArghyaChatterjee, 2024-12-23:
Sorry for the late reply. I think it's better if Kimera-VIO package incorporate this change in their code.

The Kimera-VIO code is calling:
```
integrateMeasurement(acc, gyro, deltaT);
```
Which is deprecated in GTSAM 4.2 according to `integrateMeasurement` member function of `PreintegratedRotation` class in [PreintegratedRotation.h](https://github.com/borglab/gtsam/blob/develop/gtsam/navigation/PreintegratedRotation.h#L221) file.

The compiler expects in GTSAM 4.2:
```
integrateMeasurement(acc, gyro, deltaT, optionalJacobian1, optionalJacobian2);
```
Or this:
```
integrateGyroMeasurement(gyro, Eigen::Vector3d::Zero(), deltaT);
```

In `CrossCorrTimeAligner.cpp` of Kimera-VIO package, the calls to `integrateMeasurement` needs to be changed to include the additional parameters required by GTSAM 4.2:
```
rot_pim.integrateMeasurement(imu_acc_gyrs.block<3, 1>(3, i), Eigen::Vector3d::Zero(), imu_period_s_, boost::none, boost::none);
```
Or it can be written as:
```
rot_pim.integrateGyroMeasurement(imu_acc_gyrs.block<3, 1>(3, i), Eigen::Vector3d::Zero(), imu_period_s_);
```
As suggested by the deprecation message.

Similarly, for other calls to integrateMeasurement in the same file or elsewhere in Kimera-VIO, placeholders needed to be provided (e.g., boost::none) for the optional Jacobian arguments or one can directly call `integrateGyroMeasurement` function as replacement of the `integrateMeasurement` function.

These are the 2 places that need these changes in Kimera-VIO package. 
1. [Line 105](https://github.com/MIT-SPARK/Kimera-VIO/blob/master/src/initial/CrossCorrTimeAligner.cpp#L105)
2. [Line 113](https://github.com/MIT-SPARK/Kimera-VIO/blob/master/src/initial/CrossCorrTimeAligner.cpp#L113)