borglab/gtsam #1499: add cmake option to build with memory sanitizer library

berndpfrommer, 2023-03-23:
## Feature
Building with libsanitzer has helped me greatly in debugging memory issues. The way to build gtsam with it is described here:
https://gist.github.com/jlblancoc/44be9d4d466f0a973b1f3808a8e56782
Could this be made a cmake build option in GTSAM?
## Motivation
When users see the option they know it's easily possible to compile with sanitizer support and don't have to first find the instructions above.
@jlblancoc if you think this makes sense, then please leave this feature request open as a reminder to add this feature when you get around to it.

ProfFan, 2023-04-04:
I think you can build with ASAN easily if you use Clang - just add `-fsanitize=address,undefined` to the CXX arguments when you run CMake

berndpfrommer, 2023-04-05:
I agree that it can easily be accomplished (the link above gives the details). The rationale for adding an option is that people don't have to first search to find out how to do it (see Motivation above).

ProfFan, 2023-04-20:
I think the "recommended" way is using the CMake presets, but not sure

dellaert, 2023-06-04:
Happy to look at a PR - if none forthcoming we should probably close the issue.

closed 2023-06-15 by varunagrawal