borglab/gtsam #1503: Provide a clang-format file for better consistency

zubingtan, 2023-03-27:
I have found that there are several different coding styles in gtsam's repo at present. Can we provide a clang-format file to ensure the consistency of the code format? Such as 

.clang-format
```
BasedOnStyle: Google

BinPackArguments: false
BinPackParameters: false
ColumnLimit: 100
DerivePointerAlignment: false
IncludeBlocks: Preserve
PointerAlignment: Left
```

ProfFan, 2023-04-04:
That would be a pretty good opportunity for a PR :)

zubingtan, 2023-04-13:
Hi @dellaert . We have just merged the '.clang-format 'configuration file in https://github.com/borglab/gtsam/pull/1506, which will allow all developers to format code using the `clang-format` tool. In your opinion, will developers be forced to format their code in future PR reviews, or will a so-called readability review be required?

dellaert, 2023-04-13:
Every review to some extent reviews readability. Guideline for formatting is: only format the edits you make using “format selection” to cut down on review time and repo churn. If formatting entire files, make that a separate PR, perhaps to prep another PR. In general, though, it tends to obfuscate git-blame so it’s not necessarily desired.

dellaert, 2023-04-13:
Thanks for contributing!! I’ll close the issue.

closed 2023-04-13 by dellaert