borglab/gtsam #1513: "make install" does not install the Python bindings

dkogan, 2023-04-14:
Hi. I'm building the package for Debian. I'm doing the cmake with

```
		-DGTSAM_BUILD_PYTHON=ON				\
		-DGTSAM_INSTALL_MATLAB_TOOLBOX=OFF		\
```

The `make` should the build the Python bindings and the `make install` should install them such that I can 'python3' and 'import gtsam`. Currently this does not happen. The `make install` currently DOES install stuff in `/usr/lib/gtwrap`, which isn't the Python bindings (I think it's infrastructure to make them?) These aren't needed for the end user, so they shouldn't be installed.

Thanks

ProfFan, 2023-04-20:
Python bindings are tied to your Python environment so we do not by default install them. You can install by `make python-install`. The GTWRAP install is something we are trying to get rid of, but none of the team members have time for that...

dkogan, 2023-04-21:
I'm making the package for the distro. It anybody actually maintaining and using the python bindings? If it's an afterthought that nobody is testing, then I'd rather not even build it.

Thanks

varunagrawal, 2023-04-26:
We have lots of folks using both the Matlab and the Python bindings so they are actively maintained. As @ProfFan mentioned, the bindings are installed via a separate command to make installation cleaner and give users more control.

Can you please explain your use case a bit more? Can you run `make python-install` when packaging?

dkogan, 2023-04-27:
My use case is standard: build the library, install it, and use it. In `cmake` builds you usually you do that with

1. `cmake -Dstuff`
2. `make`
3. `make install`

But here apparently I also need to `make python-install` and I also need to delete the `gtwrap` stuff. It's different from how every single other library does it. But if you feel like this is correct, I'll do the extra steps. It's your library.

Thanks.

closed 2023-04-27 by varunagrawal