borglab/gtsam #1519: matlab/gtsam_examples/IMUKittiExampleGPS" fails to run!

lifeiyu01, 2023-04-24:
Hello Sir,
Most of the example files can be run, but the file "IMUKittiExampleGPS.m" can't be run correctly. The error messages are here:
IMUKittiExampleGPS
Example of application of ISAM2 for GPS-aided navigation on the KITTI VISION BENCHMARK SUITE (http://www.computervisiononline.com/dataset/kitti-vision-benchmark-suite)
-- Reading sensor metadata
-- Reading sensor data from file
-- Starting main loop: inference is performed at each time step, but we plot trajectory every 10 steps
错误使用(Error to use) gtsam_wrapper
Exception from gtsam:
Attempting to retrieve value with key "v3", type stored in Values is class gtsam::GenericValue<class
Eigen::Matrix<double,-1,1,0,-1,1> > but requested type was class Eigen::Matrix<double,3,1,0,3,1>
出错(ERROR) gtsam.ISAM2/update (第 237 行 line 237)
varargout{1} = gtsam_wrapper(1765, this, varargin{:});
出错(ERROR) IMUKittiExampleGPS (第 124 行 line 124)
isam.update(newFactors, newValues);
Could you help me to find what's wrong with it? Thank you very much.

mattking-smith, 2023-04-25:
Hi @lifeiyu01,

Just today I pulled and verified the [`develop` branch](git@github.com:borglab/gtsam.git) with the MATLAB toolbox.  Perhaps you could try doing this for me? 

Also make sure to set  `GTSAM_ALLOW_DEPRECATED_SINCE_V = OFF` during the build process.

dellaert, 2023-06-04:
Closing as verified to work and no further activity.

closed 2023-06-04 by dellaert