borglab/gtsam #1532: The full posterior inference gives different results when adapting the sample code from the tutorial from 2D to 3D

astumpf, 2023-05-31:
## Description

I've adapted the example from the [tutorial](https://gtsam.org/tutorials/intro.html) into the 3D world (x,y,z,roll,pitch,yaw). The resulting full posterior inference surprised me, details see below. I'm wondering if I either did a mistake during the conversion or something goes wrong here. I appreciate any feedback :-).

## Steps to reproduce

Run this code:

```
#include <gtsam/geometry/Pose2.h>
#include <gtsam/nonlinear/Marginals.h>

// Create an empty nonlinear factor graph
gtsam::NonlinearFactorGraph graph;

// Add a Gaussian prior on pose x_1
gtsam::Pose3 priorMean(gtsam::Pose2(0.0, 0.0, 0.0));

gtsam::noiseModel::Diagonal::shared_ptr priorNoise = gtsam::noiseModel::Diagonal::Sigmas((gtsam::Vector(6) << 0.3, 0.3, 0.3, 0.1, 0.1, 0.1).finished());

graph.add(gtsam::PriorFactor<gtsam::Pose3>(1, priorMean, priorNoise));

// Add two odometry factors
gtsam::Pose3 odometry(gtsam::Pose2(2.0, 0.0, 0.0));
gtsam::noiseModel::Diagonal::shared_ptr odometryNoise = gtsam::noiseModel::Diagonal::Sigmas((gtsam::Vector(6) << 0.2, 0.2, 0.2, 0.1, 0.1, 0.1).finished());
graph.add(gtsam::BetweenFactor<gtsam::Pose3>(1, 2, odometry, odometryNoise));
graph.add(gtsam::BetweenFactor<gtsam::Pose3>(2, 3, odometry, odometryNoise));

gtsam::Values initial;
initial.insert(1, gtsam::Pose3(gtsam::Pose2(0.5, 0.0, 0.2)));
initial.insert(2, gtsam::Pose3(gtsam::Pose2(2.3, 0.1, -0.2)));
initial.insert(3, gtsam::Pose3(gtsam::Pose2(4.1, 0.1, 0.1)));

// optimize using Levenberg-Marquardt optimizationw
gtsam::LevenbergMarquardtParams params = gtsam::LevenbergMarquardtParams();
gtsam::Values result = gtsam::LevenbergMarquardtOptimizer(graph, initial, params).optimize();

gtsam::Marginals marginals(graph, result);

priorMean.print("Prior mean:\n");
priorNoise->print("Prior noise model:\n");
odometry.print("Odometry:\n");
std::cout.precision(2);
std::cout << "x1 covariance:\n" << marginals.marginalCovariance(1) << std::endl;
std::cout << "x2 covariance:\n" << marginals.marginalCovariance(2) << std::endl;
std::cout << "x3 covariance:\n" << marginals.marginalCovariance(3) << std::endl;
```

## Expected behavior

I would expect the covariance along the y-axis should behave similar as in the 2D example.

2D result from the tutorial code:
```
x1 covariance:
       0.09     1.1e-47     5.7e-33
    1.1e-47        0.09     1.9e-17
    5.7e-33     1.9e-17        0.01
x2 covariance:
       0.13     4.7e-18     2.4e-18
    4.7e-18        0.17        0.02
    2.4e-18        0.02        0.02
x3 covariance:
       0.17     2.7e-17     8.4e-18
    2.7e-17        0.37        0.06
    8.4e-18        0.06        0.03
```

In the 3D case, the variance for the translation is equal along all axis:
```
x1 covariance:
    0.09 -3.9e-30        0        0        0  3.4e-32
-3.9e-30     0.09        0        0        0 -1.5e-60
       0        0     0.09  4.5e-32  5.1e-17        0
       0        0  4.5e-32     0.01 -2.5e-33        0
       0        0  5.1e-17 -2.5e-33     0.01        0
 3.4e-32 -1.5e-60        0        0        0     0.01
x2 covariance:
    0.13   -1e-31        0        0        0  2.2e-16
-7.1e-32     0.13        0        0        0    -0.18
       0        0     0.13 -2.2e-16     0.18        0
       0        0 -2.2e-16     0.02 -4.4e-16        0
       0        0     0.18 -4.4e-16     0.38        0
 2.2e-16    -0.18        0        0        0     0.38
x3 covariance:
    0.17 -2.7e-31        0        0        0    3e-16
-2.3e-31     0.17        0        0        0    -0.44
       0        0     0.17   -3e-16     0.44        0
       0        0   -3e-16     0.03 -1.2e-15        0
       0        0     0.44 -1.2e-15      1.6        0
   3e-16    -0.44        0        0        0      1.6

```

## Environment

Ubuntu 18.04, C++

dellaert, 2023-06-04:
@astumpf   The noise models should be 3DOf (not 6) and moreover, the order in which rotation and translation are specified is reversed: it is x, y, theta for Pose2. Try this and please re-open if it still misbehaves.

closed 2023-06-04 by dellaert

astumpf, 2023-06-05:
Thanks for the reply, but I am confused by your answer. For clarification, I would like to get a running example with 6 DoF. My understanding is that the results should be similar despite the problem dimension.

In the 2D (3 DoF) case, the covariance sequence for x and y is different and is for y: 0.09 -> 0.17 -> 0.37. However, if we simply rewrite the tutorial code to 6 DoF, we get for all translations (x, y, z): 0.09 -> 0.13 -> 0.17.

I wonder why the result is so different, since the optimised trajectory is the same.

astumpf, 2023-08-24:
Just reopen it, appreciate any input on that :-).

astumpf, 2023-08-24:
PS: I can't reopen issues as I'm not an official collaborator.

ProfFan, 2023-08-25:
I would suggest you add a unit test that demonstrates your problem, for example against a Monte Carlo sampling posterior, and then we can know what you are seeing.