borglab/gtsam #1535: Cannot find gtsam.symbol due to case sensitivity.

rblilja, 2023-06-05:
## Description

Some of the MATLAB examples using gtsam.symbol cannot find 'symbol'. It seems to be due to a case sensitivity. The folder /usr/local/gtsam_toolbox/+gtsam/ contains a file Symbol.m instead of the expected symbol.m.

Changing the name from Symbol.m to symbol.m and reinstalling the toolbox in MATLAB seems to redeem the issue. This is off course an ugly workaround, but I'm not sure how to fix this in the build system. Thus I have no pull request.

As well. I do not know if this causes issues in other environments than my setup (see below)

## Steps to reproduce

Clone the develop branch. Build and install. Install the MATLAB toolbox. 

Run gtsamExamples from MATLAB command line, for an instance the Visual iSAM example.

## Expected behavior

Expected to find gtsam.symbol

## Environment

macOS Monterey (12.5)
MATLAB 2020a

## Additional information

None.

varunagrawal, 2023-06-05:
Thanks for reporting this @rblilja. Would it make more sense to change the examples? I'm not sure what Matlab's recommended naming scheme is.

rblilja, 2023-06-06:
Looking in the /usr/local/gtsam_toolbox/+gtsam folder almost all files are named using camel case convention. Renaming the calls in the examples make sense as 1) it won't be breaking for any user already calling gtsam.Symbol and 2) naming convention is kept coherent.

I can attend to this issue in the coming days.

Thank you for maintaining and developing GTSAM. I recently learned about GTSAM and it is like rediscovering sliced bread :)

varunagrawal, 2023-06-07:
@rblilja something to note that while `gtsam.Symbol` is a module, `gtsam.symbol` is a function to create a symbol for use in the factor graph.

rblilja, 2023-06-13:
@varunagrawal indeed, it shall be symbol.m then as it is a function and not a class. I have tried to decipher how Symbol.m is created. It seems to originate from a wrapper script. Looking in gtsam/inference/inference.i a class `Symbol` is declared on row 28. Rather interesting is that on row 42 a function `size_t symbol(char chr, size_t index);` is declared as well. I have noted that the function name is 'symbol' (note the lower case s) in the Symbol.m file, indicating a naming conflict. My filesystem is APFS, which is not case sensitive by default.

Either this is a side affect of a case insensitive filesystem, or something that needs to be fixed. I'm inclined to the latter as it would enhance portability.

I made a crude test and indeed it is a naming conflict, this is the content of Symbol.m before the `symbol` function is wrapped: [Symbol.m.txt](https://github.com/borglab/gtsam/files/11739581/Symbol.m.txt)

And this is Symbol.m after wrapping `symbol`: [Symbol.m.txt](https://github.com/borglab/gtsam/files/11739575/Symbol.m.txt)

Any ideas how to address this appropriately - without affecting the cpp implementation - is welcome.

varunagrawal, 2023-10-02:
@rblilja I finally encountered this issue and understand what causes it. I sincerely apologize for not prioritizing this issue sooner!!

Context: Matlab finally released a version for the M1 machines so I've been working on providing support for Apple Silicon. Before this I was using the Matlab wrapper on Linux and never encountered this issue.

Turns out, Linux uses a case sensitive filesystem while Windows and macos do not. This means that when the wrapper tries to write the files Symbol.m and symbol.m, they both get written to the same file, overriding the previous content. 😞

Unfortunately there is no easy fix for this since the recommended answers are to change the file names. I have an idea though: append the content. I'll try this out and report back to you soon.

varunagrawal, 2023-10-02:
So appending doesn't work due to Matlab's syntax rules. It's unfortunate that Matlab doesn't allow for more flexibility.

The fix I am currently implementing is to ignore the `gtsam::Symbol` class. Thus the `symbol` method will work without issues.

PR #1649 (2023-10-17): Matlab Wrapper Update
Update to the Matlab wrapper to support Apple Silicon.

Also, fixes #1535 by ignoring the `gtsam::Symbol` class, as well as a more permanent fix for #1614 so that adding `std::` in front of `pair` does not cause a parser error.
closed 2023-10-17 by varunagrawal