borglab/gtsam #1539: pybind11/eigen.h: No such file or directory

bhomaidan1990, 2023-06-08:
## Description
I'm getting an error with pybind:
```
/home//gtsam/build/python/gtsam.cpp:12:10: fatal error: pybind11/eigen.h: No such file or directory
   12 | #include <pybind11/eigen.h>
```
but pybind is installed:
`/home/user/.local/lib/python3.8/site-packages/pybind11/include/pybind11/eigen.h`

## Steps to reproduce

```
$ git clone https://github.com/borglab/gtsam.git
$ cd gtsam
$ mkdir build && cd build
$ cmake .. -DGTSAM_BUILD_PYTHON=1 -DGTSAM_PYTHON_VERSION=3.8.10
```

<details>
<summary>Click to expand the output</summary>

```
-- The CXX compiler identification is GNU 9.4.0
-- The C compiler identification is GNU 9.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- GTSAM Version: 4.3a0
-- Performing Test COMPILER_HAS_WSUGGEST_OVERRIDE
-- Performing Test COMPILER_HAS_WSUGGEST_OVERRIDE - Success
-- Performing Test COMPILER_HAS_WMISSING_OVERRIDE
-- Performing Test COMPILER_HAS_WMISSING_OVERRIDE - Failed
-- GTSAM is a shared library due to GTSAM_FORCE_SHARED_LIB
-- GTSAM_POSE3_EXPMAP=ON, enabling GTSAM_ROT3_EXPMAP as well
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.65") found components: serialization system filesystem thread program_options date_time timer chrono regex 
-- Found Eigen version: 3.4.0
-- Looking for execinfo.h
-- Looking for execinfo.h - found
-- Looking for getline
-- Looking for getline - found
-- checking for thread-local storage - found
-- Could NOT find MKL (missing: MKL_INCLUDE_DIR MKL_LIBRARIES) 
-- Found OpenMP_C: -fopenmp (found version "4.5") 
-- Found OpenMP_CXX: -fopenmp (found version "4.5") 
-- Found OpenMP: TRUE (found version "4.5")  
-- Found TBB: /usr/include (found suitable version "2020.1", minimum required is "4.4") found components: tbb tbbmalloc 
-- Building 3rdparty
-- Could NOT find GeographicLib (missing: GeographicLib_LIBRARY_DIRS GeographicLib_LIBRARIES GeographicLib_INCLUDE_DIRS) 
-- Building base
-- Building basis
-- Building geometry
-- Building inference
-- Building symbolic
-- Building discrete
-- Building hybrid
-- Building linear
-- Building nonlinear
-- Building sam
-- Building sfm
-- Building slam
-- Building navigation
-- GTSAM Version: 4.3.0
-- Install prefix: /usr/local
-- Building GTSAM - as a SHARED library
-- Building base_unstable
-- Building geometry_unstable
-- Building linear_unstable
-- Building discrete_unstable
-- Building dynamics_unstable
-- Building nonlinear_unstable
-- Building slam_unstable
-- Building partition_unstable
-- GTSAM_UNSTABLE Version: 4.3.0
-- Install prefix: /usr/local
-- Found Python: /usr/local/bin/python (found suitable exact version "3.8.10") found components: Interpreter Development Development.Module Development.Embed 
-- gtwrap Package config : /usr/local/lib/cmake/gtwrap
-- gtwrap version        : 1.0
-- gtwrap CMake path     : /usr/local/lib/cmake/gtwrap
-- gtwrap library path   : /usr/local/lib/gtwrap
-- gtwrap binary path    : /usr/local/bin/gtwrap
-- gtwrap header path    : /usr/local/include/gtwrap
-- Checking Python Version
-- Setting Python version for wrapper
-- Performing Test HAS_FLTO
-- Performing Test HAS_FLTO - Success
-- Wrote /home/belal/Downloads/gtsam/build/GTSAMConfig.cmake
-- Wrote /home/belal/Downloads/gtsam/build/GTSAM_UNSTABLEConfig.cmake
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
-- ===============================================================
-- ================  Configuration Options  ======================
--  CMAKE_CXX_COMPILER_ID type                       : GNU
--  CMAKE_CXX_COMPILER_VERSION                       : 9.4.0
--  CMake version                                    : 3.26.0
--  CMake generator                                  : Unix Makefiles
--  CMake build tool                                 : /usr/bin/make
-- Build flags                                               
--  Build Tests                                      : Enabled
--  Build examples with 'make all'                   : Enabled
--  Build timing scripts with 'make all'             : Disabled
--  Build shared GTSAM libraries                     : Enabled
--  Put build type in library name                   : Enabled
--  Build libgtsam_unstable                          : Enabled
--  Build GTSAM unstable Python                      : Enabled
--  Build MATLAB Toolbox for unstable                : Disabled
--  Build for native architecture                    : Disabled
--  Build type                                       : Release
--  C compilation flags                              :  -O3 -DNDEBUG
--  C++ compilation flags                            :  -O3 -DNDEBUG
--  Enable Boost serialization                       : ON
--  GTSAM_COMPILE_FEATURES_PUBLIC                    : cxx_std_17
--  GTSAM_COMPILE_OPTIONS_PUBLIC                     : 
--  GTSAM_COMPILE_DEFINITIONS_PUBLIC                 : 
--  GTSAM_COMPILE_OPTIONS_PUBLIC_RELEASE             : 
--  GTSAM_COMPILE_DEFINITIONS_PUBLIC_RELEASE         : 
--  Use System Eigen                                 : OFF (Using version: 3.4.0)
--  Use System Metis                                 : OFF
--  Using Boost version                              : 1.71.0
--  Use Intel TBB                                    : Yes (Version: 2020.1)
--  Eigen will use MKL                               : MKL not found
--  Eigen will use MKL and OpenMP                    : OpenMP found but GTSAM_WITH_EIGEN_MKL is disabled
--  Default allocator                                : TBB
--  Cheirality exceptions enabled                    : YES
--  Build with ccache                                : No
-- Packaging flags
--  CPack Source Generator                           : TGZ
--  CPack Generator                                  : TGZ
-- GTSAM flags                                               
--  Quaternions as default Rot3                      : Disabled
--  Runtime consistency checking                     : Disabled
--  Rot3 retract is full ExpMap                      : Enabled
--  Pose3 retract is full ExpMap                     : Enabled
--  Allow features deprecated in GTSAM 4.3           : Enabled
--  Metis-based Nested Dissection                    : Enabled
--  Use tangent-space preintegration                 : Enabled
-- MATLAB toolbox flags
--  Install MATLAB toolbox                           : Disabled
-- Python toolbox flags                                      
--  Build Python module with pybind                  : Enabled
--  Python version                                   : 3.8.10
-- ===============================================================
-- Configuring done (3.1s)
-- Generating done (2.7s)
-- Build files have been written to: /home/belal/Downloads/gtsam/build
```
</details>

## Expected behavior

To compile `GTSAM` without Errors

## Environment
Ubuntu 20.04
Python 3.8.10
cmake  3.26.0

varunagrawal, 2023-06-09:
This is strange. The wrapper should be using the `eigen.h` under `wrap/pybind11/include/pybind11` and not the one in the `site-packages` folder. Can you try removing the latter?

varunagrawal, 2023-06-16:
Closing due to lack of a response. Please re-open if this is still an issue.

closed 2023-06-16 by varunagrawal

bhomaidan1990, 2023-06-19:
> This is strange. The wrapper should be using the `eigen.h` under `wrap/pybind11/include/pybind11` and not the one in the `site-packages` folder. Can you try removing the latter?

Sorry for late reply, I have uninstalled `pybind11` from python packages using pip, however still the problem unresolved!

starball5, 2023-06-19:
Related on Stack Overflow: [pybind11/eigen.h: No such file or directory](https://stackoverflow.com/q/76503953/11107541)