borglab/gtsam #1544: PreintegrationParams parameter not exposed in Python wrapper

Dead-Hand, 2023-06-14:
## Description

In the Python wrapper for GTSAM, in the PreintegrationParams class there exists a method getBodyPSensor(). However, any attempt to call that method results in a TypeError. 

Full error is:
````
----> 1 pim_params.getBodyPSensor()

TypeError: Unable to convert function return value to a Python type! The signature was
	(self: gtsam.gtsam.PreintegratedRotationParams) -> boost::optional<gtsam::Pose3>
````

## Steps to reproduce

```
import gtsam
pim_params = gtsam.PreintegrationParams.MakeSharedU()
defaultBodyPSensor = pim_params.getBodyPSensor()
```

## Expected behavior

A gtsam.Pose3 object returned containing the body_P_sensor transform that PreintegrationParams was initialized with (in this case, the default/identity) consisting of a translation and rotation. 

## Environment

OS: Ubuntu 20.04.6 LTS
Python: 3.8.10
GTSAM: 4.1.1

varunagrawal, 2023-06-15:
Thanks for the issue @Dead-Hand. This is definitely related to the Python wrapper and involves the need for the `pybind11/stl.h` header.

@dellaert @ProfFan this would be a good time to revisit this now that we're on C++17.
https://pybind11.readthedocs.io/en/stable/advanced/cast/stl.html

varunagrawal, 2023-06-16:
I can confirm that #1546 fixes this.

closed 2023-06-16 by varunagrawal