borglab/gtsam #1547: Custom factor without Jacobian

shounak873, 2023-06-18:
Hi, I want to create a custom factor but I am not sure how to get the analytic Jacobian. Can I just create the factor with the error formula z - f(x) and let gtsam calculate the Jacobian numerically?

yetongumich, 2023-06-20:
You create a custom function that computes `z-f(x)`. Then, in the `evaluateError` function of your customized function, you can call the `numericalDerivative` function to compute the Jacobian numerically.

shounak873, 2023-07-10:
Hi @yetongumich,  I am implementing a square root factor which basically looks like `sqrt(g(y - f(x)))`. `numericalDerivative`  should work fine for this?

closed 2023-07-11 by shounak873

goodgoodliving, 2023-10-27:
> You create a custom function that computes `z-f(x)`. Then, in the `evaluateError` function of your customized factor, you can call the `numericalDerivative` function to compute the Jacobian numerically. Examples in `testNumericalDerivative.cpp` will help you get familiar with the function.

Hello, may I ask if it's possible to automatically solve for the Jacobian using GTSAM if I'm implementing a custom factor in Python, where I only know the error function and am not familiar with how to perform the Jacobian calculation? Are there relevant interfaces for this in GTSAM?