borglab/gtsam #1551: Support bazel in addition to cmake

kgreenek, 2023-06-21:
## Feature

I've been using gtsam successfully for a while now with bazel using these rules:
https://github.com/aharmat/rules_gtsam

## Motivation

[Bazel](bazel.build) is a popular build system used by many robotics companies and tech companies more widely (Uber, SpaceX, Cruise, etc.). Supporting it officially would avoid the need for a separate repo for the bazel rules.

## Pitch

Add BUILD.bazel files along-side the CMakeLists.txt files that define rules for compiling the code.

## Alternatives

Keep the bazel rules as a separate unofficial repo.

## Additional context

I would be happy to contribute this for linux support! I would need some help with updating CI. I also wouldn't be able to test the mac or Windows builds if you wanted to support that too with bazel. I think just supporting bazel with linux to start would still be useful though.

varunagrawal, 2023-07-31:
@kgreenek if you can get the PR started, I can help you out.

OznOg, 2024-11-09:
IMHO, having two separate build system for the same project is painful for contributors as it forces to test both compilations.