borglab/gtsam #1585: /usr/local/include/gtsam/base/OptionalJacobian.h:126:43: error: function definition does not declare parameters

WZG3661, 2023-07-20:
## Description
I use gtsam in my code, when I compile my code, and it reports an error:
Errors     << ov_fusion:make /home/lymow/Work/open_vins/logs/ov_fusion/build.make.002.log                                
In file included from /usr/local/include/gtsam/base/Matrix.h:27,
                 from /usr/local/include/gtsam/base/VerticalBlockMatrix.h:20,
                 from /usr/local/include/gtsam/nonlinear/internal/JacobianMap.h:23,
                 from /usr/local/include/gtsam/nonlinear/Expression.h:22,
                 from /usr/local/include/gtsam/nonlinear/ExpressionFactor.h:25,
                 from /usr/local/include/gtsam/nonlinear/ExpressionFactorGraph.h:21,
                 from /home/Work/open_vins/src/open_vins/ov_fusion/src/loose_fusion.cpp:14:
/usr/local/include/gtsam/base/OptionalJacobian.h:126:43: error: function definition does not declare parameters
  126 |   OptionalJacobian(std::nullopt_t /*none*/) :
      |                                           ^
compilation terminated due to -Wfatal-errors.
make[2]: *** [CMakeFiles/ov_fusion_lib.dir/build.make:63: CMakeFiles/ov_fusion_lib.dir/src/loose_fusion.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:234: CMakeFiles/ov_fusion_lib.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
cd /home/lymow/Work/open_vins/build/ov_fusion; catkin build --get-env ov_fusion | catkin env -si  /usr/bin/make --jobserver-auth=3,4; cd -

WZG3661, 2023-07-20:
Fixed, change C++ version

closed 2023-07-20 by WZG3661

lptl, 2024-01-31:
May I ask which C++ version you are using to solve the errror?