borglab/gtsam #1588: Are you kidding me?

WZG3661, 2023-07-22:
```C++
#include <iostream>

#include <Eigen/Eigen>
#include <gtsam/geometry/Pose3.h>

int main() {
  Eigen::Quaterniond quat;
  quat.w() = 1.0;
  quat.x() = 2.0;
  quat.y() = 3.0;
  quat.z() = 4.0;
  std::cout << "Eigen:" << std::endl;
  std::cout << "w " << quat.w() << std::endl;
  std::cout << "x " << quat.x() << std::endl;
  std::cout << "y " << quat.y() << std::endl;
  std::cout << "z " << quat.z() << std::endl;

  gtsam::Pose3 pose(gtsam::Rot3(quat), gtsam::Vector3(0, 0, 0));
  std::cout << "GTSAM:" << std::endl;
  std::cout << "w " << pose.rotation().quaternion().w() << std::endl;
  std::cout << "x " << pose.rotation().quaternion().x() << std::endl;
  std::cout << "y " << pose.rotation().quaternion().y() << std::endl;
  std::cout << "z " << pose.rotation().quaternion().z() << std::endl;

  return 0;
}
```
the output is:
```
Eigen:
w 1
x 2
y 3
z 4
GTSAM:
w 4
x 1
y 2
z 3
```

dellaert, 2023-07-22:
Hi anonymous,
I’ll look into whether this is a real bug - in which case it’s embarrassing, for sure.
But I’d request to be respectful in reporting it and even help us pinpoint the bug.
Best
Frank Dellaert

dellaert, 2023-07-22:
This was discussed at length in https://github.com/borglab/gtsam/pull/1219
Since version 4.2a8 the method `.quaternion()` is deprecated (and even removed in develop, pre 4.3), in favor of `toQuaternion`. In develop this test succeeds:
```
TEST(Rot3, ConvertQuaternion) {
  Eigen::Quaterniond eigenQuaternion;
  eigenQuaternion.w() = 1.0;
  eigenQuaternion.x() = 2.0;
  eigenQuaternion.y() = 3.0;
  eigenQuaternion.z() = 4.0;
  EXPECT_DOUBLES_EQUAL(1, eigenQuaternion.w(), 1e-9);
  EXPECT_DOUBLES_EQUAL(2, eigenQuaternion.x(), 1e-9);
  EXPECT_DOUBLES_EQUAL(3, eigenQuaternion.y(), 1e-9);
  EXPECT_DOUBLES_EQUAL(4, eigenQuaternion.z(), 1e-9);

  Rot3 R(eigenQuaternion);
  EXPECT_DOUBLES_EQUAL(1, R.toQuaternion().w(), 1e-9);
  EXPECT_DOUBLES_EQUAL(2, R.toQuaternion().x(), 1e-9);
  EXPECT_DOUBLES_EQUAL(3, R.toQuaternion().y(), 1e-9);
  EXPECT_DOUBLES_EQUAL(4, R.toQuaternion().z(), 1e-9);
}
```

I'll add a PR with it.

WZG3661, 2023-07-22:
Sorry for my inappropriate wording. Thanks for your answer!

closed 2023-07-22 by dellaert