borglab/gtsam #1593: RuntimeError: Shonan::run did not converge for given pMax

anilesec, 2023-07-26:
Hello,

I am new to this whole domain of graph-optimziation.
However, I understand from high level that given relative poses (edges) between nodes (vertices). I should be able to obtain absolute rotations for each node using Shanon Rotation Averaging.

With this understanding, I created a toy-problem with 3 nodes and 3 edges (attached inp_graph.py file) and then I run shonan algorithm using the sample script here (https://github.com/dellaert/ShonanAveraging)

    python ShonanAveragingCLI.py -i inp_graph.txt
 

I get the error below:

Running Shonan averaging for SO(3) on  inp/toy.txt
Traceback (most recent call last):
  File "gtsam/ShonanAveragingCLI.py", line 126, in <module>
    run(parser.parse_args())
  File "gtsam/ShonanAveragingCLI.py", line 97, in run
    rotations, _ = shonan.run(initial, 3, 10)
RuntimeError: Shonan::run did not converge for given pMax

I don't understand how to fix this. Could you please help? I think my measurement matrix is not correct. Currently, i don't know how to set that. Therefore, I just set to constant value 100
[image]

Thank you in advance!

dellaert, 2023-07-27:
Rather than creating a file, could you try to use the code example in in test_ShonanAvaerging.py? Here is a snippet:
```
        measurements = []
        unit3 = gtsam.noiseModel.Unit.Create(3)
        m01 = BinaryMeasurementRot3(0, 1, Rot3.Yaw(math.radians(90)), unit3)
        m12 = BinaryMeasurementRot3(1, 2, Rot3.Yaw(math.radians(90)), unit3)
        measurements.append(m01)
        measurements.append(m12)
        obj = ShonanAveraging3(measurements)
        self.assertIsInstance(obj, ShonanAveraging3)
        initial = obj.initializeRandomly()
        _, cost = obj.run(initial, min_p=3, max_p=5)
        self.assertAlmostEqual(cost, 0)
```

anilesec, 2023-07-27:
Thanks for the response! I have tried this. But I am getting the following error
[image]

I think it could be mismatch in gtsam version (I installed gtsam using pip as "pip install gtsam" )

dellaert, 2023-07-27:
Yeah. Try uninstalling and `pip install gtbook`

anilesec, 2023-07-27:
I created a new conda env (python 3.6.9) and installed gtbook (pip install gtbook).

I still get the same error:

[image]

dellaert, 2023-07-30:
Dang,. Sorry this is happening to you. I will investigate but might need the 4.2 release.

anilesec, 2023-07-30:
My bad! May I ask when the 4.2 release is planned? Otherwise, if I can somehow figure out how to set the measurements value in the input(g2o) file, perhaps I can run  "ShonanAveragingCLI.py" script. Any help to set the 21 measurements values for a custom dataset would be a great help for my research. 

Thank you very much for your precious time and effort!

dellaert, 2023-07-30:
Actually, it is related to recent changes in the wrapper. There was an old way to do this, and I tried it in [this notebook](https://colab.research.google.com/drive/1x3CuWIa7IawuOG8t4EIt0DDfo4_iXMLc?usp=sharing), and it works. Here is the code snippet:

```python
%pip -q install gtbook

import math
import gtsam
from gtsam import BinaryMeasurementRot3, Rot3, ShonanAveraging3, BinaryMeasurementsRot3

measurements = BinaryMeasurementsRot3()
unit3 = gtsam.noiseModel.Unit.Create(3)
m01 = BinaryMeasurementRot3(0, 1, Rot3.Yaw(math.radians(90)), unit3)
m12 = BinaryMeasurementRot3(1, 2, Rot3.Yaw(math.radians(90)), unit3)
measurements.append(m01)
measurements.append(m12)
obj = ShonanAveraging3(measurements)
assert isinstance(obj, ShonanAveraging3)
initial = obj.initializeRandomly()
result, cost = obj.run(initial, min_p=3, max_p=5)
assert cost < 1e-4
print(result)
```

I will close this issue but try to make sure the "list" way works in 4.2.

closed 2023-07-30 by dellaert

anilesec, 2023-07-30:
Great! It works now :) Thanks a lot!

anilesec, 2023-09-13:
> ```python
> obj = ShonanAveraging3(measurements)
> ```

This snippet does not seem to work in 4.2 release.

Also how to select the noise model for the real-world dataset?