borglab/gtsam #1596: BoundingConstraint1 with double parameter

shounak873, 2023-07-28:
In my factor graph I have  scalar parameter that is being estimated. This scalar has a lower bound. I wanted to implement this constraint using BoundingConstraint1 struct. I replaced the existing template version `BoundingConstraint.h` with this version below. But I am getting segmentation fault when running with this factor. Is this implementation correct ? Thank you.

```
#pragma once

#include <gtsam/base/Lie.h>
#include <gtsam/nonlinear/NonlinearFactor.h>

namespace gtsam {

/**
 * Unary inequality constraint forcing a scalar to be
 * greater/less than a fixed threshold.  The function
 * will need to have its value function implemented to return
 * a scalar for comparison.
 * @addtogroup SLAM
 */
// template<class VALUE>
struct BoundingConstraint1: public NoiseModelFactor1<double> {
  // typedef VALUE X;
  typedef NoiseModelFactor1<double> Base;
  typedef boost::shared_ptr<BoundingConstraint1> shared_ptr;

  double threshold_;
  bool isGreaterThan_; /// flag for greater/less than

  BoundingConstraint1(Key key, double threshold,
      bool isGreaterThan, double mu = 1000.0) :
        Base(noiseModel::Constrained::All(1, mu), key),
        threshold_(threshold), isGreaterThan_(isGreaterThan) {
  }

  virtual ~BoundingConstraint1() {}

  inline double threshold() const { return threshold_; }
  inline bool isGreaterThan() const { return isGreaterThan_; }

  /**
   * function producing a scalar value to compare to the threshold
   * Must have optional argument for derivative with 1xN matrix, where
   * N = X::dim()
   */
  // virtual double value(const X& x, boost::optional<Matrix&> H =
  //     boost::none) const = 0;

  /** active when constraint *NOT* met */
  bool active(const Values& c) const {
    // note: still active at equality to avoid zigzagging
    double x = c.at<double>(this->key());
    return (isGreaterThan_) ? x <= threshold_ : x >= threshold_;
  }

  Vector evaluateError(const double& x, boost::optional<Matrix&> H =
      boost::none) const {
    // Matrix D;
    double error = x - threshold_;
    Matrix D = Matrix::Zero(1,1);
    D(0,0) = 1.0;
    if (H) {
      if (isGreaterThan_) *H = D;
      else *H = -1.0 * D;
    }

    if (isGreaterThan_)
      return (Vector(1) << error).finished();
    else
      return -1.0 * (Vector(1) << error).finished();
  }

private:

  /** Serialization function */
  friend class boost::serialization::access;
  template<class ARCHIVE>
  void serialize(ARCHIVE & ar, const unsigned int /*version*/) {
    ar & boost::serialization::make_nvp("NoiseModelFactor1",
        boost::serialization::base_object<Base>(*this));
    ar & BOOST_SERIALIZATION_NVP(threshold_);
    ar & BOOST_SERIALIZATION_NVP(isGreaterThan_);
  }
};
```
In my main code the factor is added like this 

```
boost::shared_ptr<BoundingConstraint1> boundingFactor(new BoundingConstraint1(A(alphaKey), -10.0, true));
graph->add(boundingFactor);
```

dellaert, 2023-07-30:
I don't know. Advice is to debug with lldb or gdb. If you find a bug in GTSAM, please escalate :-)

bangyii, 2023-09-15:
I'm facing the same issue, ISAM2.update() crashes here when there are inactive factors: https://github.com/borglab/gtsam/blob/develop/gtsam/symbolic/SymbolicFactor-inst.h#L45

Likely due to https://github.com/borglab/gtsam/blob/develop/gtsam/nonlinear/NonlinearFactor.cpp#L155 returning a null pointer when factor is inactive, and then passing it along until EliminateSymbolic

I'm unsure what the desired or correct behaviour is, so I have no suggestions on a fix.

Zkjoker, 2024-03-29:
Have you solved this problem?