borglab/gtsam #1601: Is the current latest tag a pre-release?

dkogan, 2023-08-05:
Hi. I'm making a package for Debian. The current latest tag is "4.2a9". Is this a preview for an upcoming 4.2 release? Should I push 4.2a9 into the distro, or should I wait until 4.2 is finalized? Is this happening in the near future?

Thanks

ProfFan, 2023-08-10:
@dellaert Should we release 4.2 after #1599 is merged?

dellaert, 2023-08-10:
I'd love to release 4.2 but #1599 is to develop, not the release branch.
Also, I do need to make sure all chapters of roboticsbook.org render correctly before we cut it. I was planning on doing that this weekend or even tonight.

dkogan, 2023-08-10:
Hi.

I just uploaded 4.2a9 to Debian. When 4.2 is released, I'll update the
package. Closing this "bug"

dkogan, 2023-08-10:
The package tree is here: https://salsa.debian.org/science-team/gtsam

The rules file to build it are here: https://salsa.debian.org/science-team/gtsam/-/blob/master/debian/rules
A number of the logic things here are needed to clean up the build

The patches are here: https://salsa.debian.org/science-team/gtsam/-/tree/master/debian/patches
Some of these are meant to clean up the build, and others are needed to build against the system dependencies.

I'll make a separate bug report about these later.

closed 2023-08-10 by dkogan