borglab/gtsam #1602: Matlab example result in an error

etamarlu, 2023-08-10:
## Description
Running the "IMUKittiExampleGPS" example result in the following error:

```
Error using gtsam_wrapper
Exception from gtsam:
Attempting to retrieve value with key "v3", type stored in Values is class gtsam::GenericValue<class Eigen::Matrix<double,-1,1,0,-1,1>
> but requested type was class Eigen::Matrix<double,3,1,0,3,1>

Error in gtsam.ISAM2/update (line 237)
        varargout{1} = gtsam_wrapper(1765, this, varargin{:});

Error in IMUKittiExampleGPS (line 124)
      isam.update(newFactors, newValues);
```
 

## Steps to reproduce

1.  download toolbox from https://github.com/jlblancoc/factor-graphs-course
2. run the "./gtsam_examples/IMUKittiExampleGPS" example

## Environment
Matlab 2023a

## Additional information

ProfFan, 2023-08-27:
You need to retrieve the value with a dynamic matrix type if what you have is a dynamic matrix, not a fixed sized version

etamarlu, 2023-08-27:
Thank you for your response! 

I don't fully understand how to retrieve the value with a dynamic matrix, is it a gtsam thing or a Matlab thing? I've worked with Matlab  many times but haven't encountered it so far, do you have any reference to Matlab's documentation? 

Moreover, the same example works in Linux

Thank you, 
Etamar

etamarlu, 2023-08-27:
> You need to retrieve the value with a dynamic matrix type if what you have is a dynamic matrix, not a fixed sized version

Thank you for your response!

I don't fully understand how to retrieve the value with a dynamic matrix, is it a gtsam thing or a Matlab thing? I've worked with Matlab many times but haven't encountered it so far, do you have any reference to Matlab's documentation?

Moreover, the same example works in Linux

Thank you,
Etamar

ProfFan, 2023-08-28:
Windows is a way less tested platform than Linux, and the problem is none of the current maintainers use Windows... So you are on your own. This is a GTSAM (or C++) thing because the type is resolved by the GTWRAP glue code.

closed 2023-08-31 by etamarlu