borglab/gtsam #1604: Error in test scrpipt  "TranslationAveragingExample.py"

anilesec, 2023-08-10:
I run this [file](https://github.com/borglab/gtsam/blob/e36590aa456f968ae5314765eacadadb5d267d87/python/gtsam/examples/TranslationAveragingExample.py
) to test TranslationAveraging, I get the attached error.

[image]

varunagrawal, 2023-08-14:
Hi @anilesec. Are you using the develop branch of GTSAM? And are you compiling and installing the GTSAM python bindings, or installing the library from pip?

We can help you if you provide more information. I cannot reproduce this issue on my machine.

anilesec, 2023-08-18:
Hi @varunagrawal 

Thanks for the response!
I have installed "pip install gtbook" and then run the TranslationAveraging.py

varunagrawal, 2023-08-28:
Any particular reason you're using the `gtbook` python package? You should be able to get the example running by using `pip install gtsam==4.2a9`.

varunagrawal, 2023-09-04:
@anilesec we've released 4.2 officially so this should be resolved. If you're still having issues, please reopen.

closed 2023-09-04 by varunagrawal