borglab/gtsam #1606: KeySet class in python is missing methods

dwiel, 2023-08-18:
## Feature

The `KeySet` class in python should have more methods available like `__iter__`, `__contains__`, `remove`, etc.

## Motivation

I'd like to be able to do basic things like:

```
for key in graph.keys():
    if key not in initial_estimate.keys():
        print(f"WARNING: missing key {key} in initial_estimate")
```

ProfFan, 2023-08-27:
Yeah this would be a great extension. We may need some special code to do this though, as `KeySet` is a bit more than a standard `std::set`. CC @varunagrawal

varunagrawal, 2023-10-04:
I think the easiest way to achieve this is to replace `gtsam::KeySet` with `std::set<gtsam::Key>` (aka the superclass) in all the .i files?

varunagrawal, 2024-01-14:
A better way to do this is to integrate dunder methods (e.g. `__iter__`) into our wrapper library `gtwrap`. @dwiel can you compile a list of all the special methods needed?

varunagrawal, 2024-01-16:
Accidentally closed this. This issue will be resolved when we merge #1712.

PR #1712 (2024-01-17): Update Wrapper
Update the wrapper with new functionality supporting a new double-underscore (dunder) syntax. This makes wrapping classes that should exhibit some behavior in Python easier.

E.g. there is an ask in #1606 for `KeySet` to allow iteration and other container based operations. By adding `__iter__` and `__contains__` dunder methods to the `gtsam.i` file, we now support those operations.
This means we can now do this:

```python
keys = graph1.keys()
for key in keys:
    if key in graph2.keys():
        print("key is in both graphs")
```

Consequently, fixes #1606.
closed 2024-01-17 by varunagrawal