borglab/gtsam #1610: how to build gtsam 4.0.3 with vs 2019

Willforcv, 2023-08-26:
when i building the release version, something strange happened~  The progress of the release version compilation process has not changed for a long time, but it is not stuck.   The progress of the debug version compilation process is ok~

Willforcv, 2023-08-26:
`
6>            AP2=boost::mpl::na,
6>            AP3=boost::mpl::na
6>        ] (compiling source file D:\software\gtsam\gtsam\slam\InitializePose3.cpp)
6>D:\software\gtsam\gtsam/inference/MetisIndex.h(53): message : see reference to class template instantiation 'boost::bimaps::bimap<gtsam::Key,int32_t,boost::mpl::na,boost::mpl::na,boost::mpl::na>' being compiled (compiling source file D:\software\gtsam\gtsam\slam\InitializePose3.cpp)
6>C:\local\boost_1_74_0\boost/concept_check.hpp(355,12): warning C4834: discarding return value of function with 'nodiscard' attribute (compiling source file D:\software\gtsam\gtsam\slam\lago.cpp)
6>C:\local\boost_1_74_0\boost/concept_check.hpp(354): message : while compiling class template member function 'void boost::BinaryFunction<Func,Return,First,Second>::test(boost::false_type)'
6>        with
6>        [
6>            Func=std::less<int>,
6>            Return=bool,
6>            First=int,
6>            Second=int
6>        ] (compiling source file D:\software\gtsam\gtsam\slam\lago.cpp)
6>C:\local\boost_1_74_0\boost/concept_check.hpp(351): message : see reference to function template instantiation 'void boost::BinaryFunction<Func,Return,First,Second>::test(boost::false_type)' being compiled
6>        with
6>        [
6>            Func=std::less<int>,
6>            Return=bool,
6>            First=int,
6>            Second=int
6>        ] (compiling source file D:\software\gtsam\gtsam\slam\lago.cpp)
6>C:\local\boost_1_74_0\boost/concept_check.hpp(349): message : see reference to class template instantiation 'boost::BinaryFunction<Func,Return,First,Second>' being compiled
6>        with
6>        [
6>            Func=std::less<int>,
6>            Return=bool,
6>            First=int,
6>            Second=int
6>        ] (compiling source file D:\software\gtsam\gtsam\slam\lago.cpp)
6>C:\local\boost_1_74_0\boost/concept/detail/has_constraints.hpp(42): message : see reference to class template instantiation 'boost::BinaryFunctionConcept<std::less<int>,bool,int,int>' being compiled (compiling source file D:\software\gtsam\gtsam\slam\lago.cpp)
6>C:\local\boost_1_74_0\boost/concept/detail/has_constraints.hpp(32): message : see reference to function template instantiation 'boost::concepts::detail::yes boost::concepts::detail::has_constraints_(Model *,boost::concepts::detail::wrap_constraints<Model,&Model::constraints> *)' being compiled (compiling source file D:\software\gtsam\gtsam\slam\lago.cpp)
6>C:\local\boost_1_74_0\boost/concept/detail/msvc.hpp(57): message : see reference to class template instantiation 'boost::concepts::not_satisfied<Model>' being compiled
6>        with
6>        [
6>            Model=boost::BinaryFunctionConcept<std::less<int>,bool,int,int>
6>        ] (compiling source file D:\software\gtsam\gtsam\slam\lago.cpp)
6>C:\local\boost_1_74_0\boost/bimap/set_of.hpp(132): message : see reference to class template instantiation 'boost::concepts::require<boost::BinaryFunctionConcept<std::less<int>,bool,int,int>>' being compiled (compiling source file D:\software\gtsam\gtsam\slam\lago.cpp)
6>C:\local\boost_1_74_0\boost/bimap/detail/manage_bimap_key.hpp(69): message : see reference to class template instantiation 'boost::bimaps::set_of<Type,std::less<int>>::lazy_concept_checked' being compiled
6>        with
6>        [
6>            Type=int32_t
6>        ] (compiling source file D:\software\gtsam\gtsam\slam\lago.cpp)
6>C:\local\boost_1_74_0\boost/bimap/detail/bimap_core.hpp(97): message : see reference to class template instantiation 'boost::bimaps::detail::manage_bimap_key<RightSetType>' being compiled
6>        with
6>        [
6>            RightSetType=int32_t
6>        ] (compiling source file D:\software\gtsam\gtsam\slam\lago.cpp)
6>C:\local\boost_1_74_0\boost/bimap/bimap.hpp(134): message : see reference to class template instantiation 'boost::bimaps::detail::bimap_core<KeyTypeA,KeyTypeB,AP1,AP2,AP3>' being compiled
6>        with
6>        [
6>            KeyTypeA=gtsam::Key,
6>            KeyTypeB=int32_t,
6>            AP1=boost::mpl::na,
6>            AP2=boost::mpl::na,
6>            AP3=boost::mpl::na
6>        ] (compiling source file D:\software\gtsam\gtsam\slam\lago.cpp)
6>D:\software\gtsam\gtsam/inference/MetisIndex.h(53): message : see reference to class template instantiation 'boost::bimaps::bimap<gtsam::Key,int32_t,boost::mpl::na,boost::mpl::na,boost::mpl::na>' being compiled (compiling source file D:\software\gtsam\gtsam\slam\lago.cpp)
6>ImuFactor.cpp
6>C:\local\boost_1_74_0\boost/bind.hpp(41): message : The practice of declaring the Bind placeholders (_1, _2, ...) in the global namespace is deprecated. Please use <boost/bind/bind.hpp> + using namespace boost::placeholders, or define BOOST_BIND_GLOBAL_PLACEHOLDERS to retain the current behavior.
6>ManifoldPreintegration.cpp
6>NavState.cpp
6>PreintegratedRotation.cpp
6>PreintegrationBase.cpp
6>PreintegrationParams.cpp
6>ScenarioRunner.cpp
6>C:\local\boost_1_74_0\boost/bind.hpp(41): message : The practice of declaring the Bind placeholders (_1, _2, ...) in the global namespace is deprecated. Please use <boost/bind/bind.hpp> + using namespace boost::placeholders, or define BOOST_BIND_GLOBAL_PLACEHOLDERS to retain the current behavior.
6>C:\local\boost_1_74_0\boost/bind.hpp(41): message : The practice of declaring the Bind placeholders (_1, _2, ...) in the global namespace is deprecated. Please use <boost/bind/bind.hpp> + using namespace boost::placeholders, or define BOOST_BIND_GLO [... cut ...]
6>            KeyTypeB=int32_t,
6>            AP1=boost::mpl::na,
6>            AP2=boost::mpl::na,
6>            AP3=boost::mpl::na
6>        ] (compiling source file D:\software\gtsam\gtsam\navigation\TangentPreintegration.cpp)
6>D:\software\gtsam\gtsam/inference/MetisIndex.h(53): message : see reference to class template instantiation 'boost::bimaps::bimap<gtsam::Key,int32_t,boost::mpl::na,boost::mpl::na,boost::mpl::na>' being compiled (compiling source file D:\software\gtsam\gtsam\navigation\TangentPreintegration.cpp)`

Willforcv, 2023-08-26:
[image]

varunagrawal, 2023-09-04:
Hi @Willforcv. This is a known issue with VS 2019 where it takes up to 2 hours for library to compile. We have some workarounds in our CI which you may want to take a look at.

cuckoo321, 2024-03-13:
I tried to build 4.0.3 with vs2019 and got the same problem as above. I found that geometry/unit3.cpp can't generate unit3.obj file. When I commented the two sentences bold, it generated.  What's the reason?
```
Unit3 Unit3::retract(const Vector2& v, OptionalJacobian<2,2> H) const {
  // Compute the 3D xi_hat vector
  const Vector3 xi_hat = basis() * v;
  const double theta = xi_hat.norm();
  const double c = std::cos(theta);

  // Treat case of very small v differently.
  Matrix23 H_from_point;
  if (theta < std::numeric_limits<double>::epsilon()) {
    const Unit3 exp_p_xi_hat = Unit3::FromPoint3(c * p_ + xi_hat,
                                                 H? &H_from_point : nullptr);
    if (H) { // Jacobian
     ;**// *H = H_from_point *  (-p_ * xi_hat.transpose() + Matrix33::Identity()) * basis();**
    }
    return exp_p_xi_hat;
  }

  const double st = std::sin(theta) / theta;
  const Unit3 exp_p_xi_hat = Unit3::FromPoint3(c * p_ + xi_hat * st,
                                               H? &H_from_point : nullptr);
  if (H) { // Jacobian
 ; _**//  *H = H_from_point *        (p_ * -st * xi_hat.transpose() + st * Matrix33::Identity() +  xi_hat * ((c - st) / std::pow(theta, 2)) * i_hat.transpose()) * basis();**_
  }
  return exp_p_xi_hat;
}
```

varunagrawal, 2024-09-17:
I believe we've fixed this issue which is why the CI no longer takes 2 hours to run. If still an issue, please reopen.

closed 2024-09-17 by varunagrawal