borglab/gtsam #1612: PCL let GTSAM crash.

lyliss, 2023-08-29:
When gtsam and pcl are linked together in the cpp file, invoking isam2.update(graph, initial) results in a program crash with the error message "interrupted by signal 11: SIGSEGV"

lyliss, 2023-09-03:
now I try it on other pc, error message changed to double free or corruption (out), if mask pcl code, the cpp is runing normally

closed 2023-09-08 by lyliss

chxyfishge, 2024-07-29:
I have the same question，and I soved it. Let me explain my situation, maybe it would be helpful. 
Ubuntu 20.04 provided **PCL 1.10.0 in /usr/add and/usr/lib**, but I compiled another **PCL 1.10.0 and installed it in/usr/local/**. The conflict between these two PCLs resulted in 
`double free or corruption (out),`