borglab/gtsam #1614: MATLAB wrapper failed to build [Ubuntu]

JTylerBoylan, 2023-08-29:
## Description

I am trying to build the GTSAM MATLAB wrapper on Ubuntu 22.04.  I've installed all the required dependencies, but am getting a build error when generating `../wrap/gtsam/gtsam_wrapper.cpp`.

I was following this README: https://github.com/borglab/gtsam/blob/develop/matlab/README.md

## Steps to reproduce

After cloning and installing dependencies (CMake, Boost, TBB), I ran these commands:
```
mkdir build
cd build
cmake .. -DGTSAM_INSTALL_MATLAB_TOOLBOX=ON
make install
```

I got this CMake output:
```
-- GTSAM Version: 4.3a0
-- GTSAM is a shared library due to GTSAM_FORCE_SHARED_LIB
-- GTSAM_POSE3_EXPMAP=ON, enabling GTSAM_ROT3_EXPMAP as well
-- Found Eigen version: 3.4.0
-- checking for thread-local storage - found
-- Could NOT find MKL (missing: MKL_INCLUDE_DIR MKL_LIBRARIES) 
-- Building 3rdparty
-- Could NOT find GeographicLib (missing: GeographicLib_LIBRARY_DIRS GeographicLib_LIBRARIES GeographicLib_INCLUDE_DIRS) 
-- Building base
-- Building basis
-- Building geometry
-- Building inference
-- Building symbolic
-- Building discrete
-- Building hybrid
-- Building linear
-- Building nonlinear
-- Building sam
-- Building sfm
-- Building slam
-- Building navigation
-- GTSAM Version: 4.3.0
-- Install prefix: /usr/local
-- Building GTSAM - as a SHARED library
-- Building base_unstable
-- Building geometry_unstable
-- Building linear_unstable
-- Building discrete_unstable
-- Building dynamics_unstable
-- Building nonlinear_unstable
-- Building slam_unstable
-- Building partition_unstable
-- GTSAM_UNSTABLE Version: 4.3.0
-- Install prefix: /usr/local
-- gtwrap Package config : /usr/local/lib/cmake/gtwrap
-- gtwrap version        : 1.0
-- gtwrap CMake path     : /usr/local/lib/cmake/gtwrap
-- gtwrap library path   : /usr/local/lib/gtwrap
-- gtwrap binary path    : /usr/local/bin/gtwrap
-- gtwrap header path    : /usr/local/include/gtwrap
-- Building wrap module gtsam
-- Installing Matlab Toolbox to /usr/local/gtsam_toolbox
-- Wrote /home/jtyler/Documents/GTSAM/build/GTSAMConfig.cmake
-- Wrote /home/jtyler/Documents/GTSAM/build/GTSAM_UNSTABLEConfig.cmake
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
-- ===============================================================
-- ================  Configuration Options  ======================
--  CMAKE_CXX_COMPILER_ID type                       : GNU
--  CMAKE_CXX_COMPILER_VERSION                       : 11.4.0
--  CMake version                                    : 3.22.1
--  CMake generator                                  : Unix Makefiles
--  CMake build tool                                 : /usr/bin/gmake
[... 385 lines cut ...]
                 from /home/jtyler/Documents/GTSAM/gtsam/global_includes.h:22,
                 from /home/jtyler/Documents/GTSAM/gtsam/base/Vector.h:28,
                 from /home/jtyler/Documents/GTSAM/wrap/matlab.h:25,
                 from /home/jtyler/Documents/GTSAM/build/wrap/gtsam/gtsam_wrapper.cpp:1:
/usr/include/c++/11/bits/stl_pair.h:315:17: note: candidate: ‘std::pair<_T1, _T2>::pair(std::pair<_T1, _T2>&&) [with _T1 = gtsam::DiscreteConditional; _T2 = gtsam::DecisionTreeFactor]’
  315 |       constexpr pair(pair&&) = default;         ///< Move constructor
      |                 ^~~~
/usr/include/c++/11/bits/stl_pair.h:315:22: note:   no known conversion for argument 1 from ‘std::pair<std::shared_ptr<gtsam::DiscreteConditional>, std::shared_ptr<gtsam::DecisionTreeFactor> >’ to ‘std::pair<gtsam::DiscreteConditional, gtsam::DecisionTreeFactor>&&’
  315 |       constexpr pair(pair&&) = default;         ///< Move constructor
      |                      ^~~~~~
/usr/include/c++/11/bits/stl_pair.h:314:17: note: candidate: ‘std::pair<_T1, _T2>::pair(const std::pair<_T1, _T2>&) [with _T1 = gtsam::DiscreteConditional; _T2 = gtsam::DecisionTreeFactor]’
  314 |       constexpr pair(const pair&) = default;    ///< Copy constructor
      |                 ^~~~
/usr/include/c++/11/bits/stl_pair.h:314:22: note:   no known conversion for argument 1 from ‘std::pair<std::shared_ptr<gtsam::DiscreteConditional>, std::shared_ptr<gtsam::DecisionTreeFactor> >’ to ‘const std::pair<gtsam::DiscreteConditional, gtsam::DecisionTreeFactor>&’
  314 |       constexpr pair(const pair&) = default;    ///< Copy constructor
      |                      ^~~~~~~~~~~
make[2]: *** [matlab/CMakeFiles/gtsam_matlab_wrapper.dir/build.make:96: matlab/CMakeFiles/gtsam_matlab_wrapper.dir/__/wrap/gtsam/gtsam_wrapper.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:32504: matlab/CMakeFiles/gtsam_matlab_wrapper.dir/all] Error 2
make: *** [Makefile:166: all] Error 2

```

## Environment

Ubuntu 22.04.3 LTS

varunagrawal, 2023-09-04:
Hi @JTylerBoylan. I'll probably get to this tomorrow, so thank you for your patience!

varunagrawal, 2023-09-08:
@JTylerBoylan this issue has been reproduced and fixed. Thanks for reporting it! Please try the latest develop branch and let us know if it works for you?

PR #1631 (2023-09-08): Fix Matlab compilation for discrete.i
This PR fixes #1614 

Unfortunately, the Matlab wrapper has a bug where if the return type is a `std::pair` then it fails to compile, since the expected return type is `pair` (without the `std::`). I'll add a fix to the matlab wrapper, but this PR should fix the issue in the meantime.
closed 2023-09-08 by varunagrawal