borglab/gtsam #1623: release 4.2 should be tagged 4.2.0

berndpfrommer, 2023-09-04:
The tag 4.2 is inconsistent with the major.minor.patch convention that is used by most packaging frameworks.
Could somebody please change the tag from 4.2 to 4.2.0, or add an alias (that most likely will work for getting the debian packaging working):

These commands should suffice to add an alias:
```
git tag 4.2.0 4.2
git push origin --tags
```

The above commands were gleaned from the discussion on [stack overflow](https://stackoverflow.com/questions/1028649/how-do-you-rename-a-git-tag) where you can also learn how to rename a tag outright (essentially above steps, then removing the 4.2 tag).

varunagrawal, 2023-09-04:
@dellaert @ProfFan @johnwlambert

varunagrawal, 2023-09-04:
@berndpfrommer I am seeing that a tag alias for 4.2.0 has been created. Can you please check if everything looks good?

berndpfrommer, 2023-09-05:
Works, thanks much!

New Ubuntu PPA is ready with release 4.2.0:
https://launchpad.net/~borglab/+archive/ubuntu/gtsam-release-4.2

closed 2023-09-05 by berndpfrommer