borglab/gtsam #1627: Shonan rotation averaging converging but relative rotation is high!!

anilesec, 2023-09-05:
Dear @dellaert,

I tried shonan rotation averaging for a toy example.(Let say I take 3 random rotation matrices and form a full connected graph. i.e 3 nodes and 3 edges). 

I input the correct(no noise added) relative rotations between the nodes. I expect that cost should be close to zero and relative rotation errors between the estimated absolute rotations and know relative rotations should be zero? But the relative rotations error is not zero?

So, can shonan rotation averaging provide perfect rotations given correct(without any noise) relative rotations as input?

I have spent too much time to get the zero relative rotation errors but failed. Is adding noise to input relative rotations a necessary condition to get zero errors? Any help here is much appreciated

Thanks in advance!

anilesec, 2023-09-05:
Also do you have an example where shonan rotation is tried on toy example to achieve zero relative rotation errors? I would like to understand how you compute relative rotation errors? Do you align obtained absolute rotations and known rotations  before computing the error?

anilesec, 2023-09-05:
I have tried with graphs created synthetically and listed relative rotation error from shonan, below is the table:

[image]

dellaert, 2023-09-05:
I could try to take a look this weekend if you have a simple python script that shows a failure case

anilesec, 2023-09-06:
Thank you very much. I have **attached the simple Python script**:
Basically, it generates `NUM_NODES` random rotations(as ground truth rotations) -->  and creates a fully connected graph with relative rotations computed from generated random rotations to Shonan method  --> and runs Shonan method --> estimates relative rotation from obtained absolute rotations ---> and computes error with relative rotations obtained from generated rotations --> prints Error

```
import numpy as np
import gtsam
from gtsam import BinaryMeasurementRot3, Rot3, ShonanAveraging3, BinaryMeasurementsRot3
from scipy.stats import special_ortho_group
from ipdb import set_trace as bb

def mat2rotvec(rotmat, in_degrees=True, axis='xyz'):
    from scipy.spatial.transform import Rotation as ROT
    r = ROT.from_matrix(rotmat)
    if in_degrees:
        return r.as_euler(axis, in_degrees)
    else:
        return r.as_rotvec()

def compute_angular_error_batch(rotation1, rotation2):
    R_rel = np.einsum("Bij,Bjk ->Bik", rotation1.transpose(0, 2, 1), rotation2)
    t = (np.trace(R_rel, axis1=1, axis2=2) - 1) / 2
    theta = np.arccos(np.clip(t, -1, 1))
    return theta * 180 / np.pi

## creating random absolute rotations
NUM_NODES = 15
gt_rots = np.array(
    [special_ortho_group.rvs(3) for i in range(NUM_NODES)]
)
print('Random Rotations:')
for p in gt_rots:
    print(f"rot along [x, y, z]: {mat2rotvec(p)}")

## creating fully connected graph edges
from itertools import permutations
edges = list(permutations(range(NUM_NODES), 2))

## Rotation Averaging(shonan)
measurements = BinaryMeasurementsRot3()
unit3 = gtsam.noiseModel.Unit.Create(3)
for edge in edges:
    print('Edge:', edge)
    i, j = edge
    Pij = gt_rots[j] @ np.linalg.inv(gt_rots[i])
    mij = BinaryMeasurementRot3(i, j, Rot3(Pij[:3, :3]), unit3)
    measurements.append(mij)
obj = ShonanAveraging3(measurements)
assert isinstance(obj, ShonanAveraging3)
initial = obj.initializeRandomly()
result, cost = obj.run(initial, min_p=3, max_p=8)
assert cost < 1e-4
print(result)
print('Rot. Averaging cost:', cost)

## estimated rotations
est_rots = []
for i in range(result.size()):
    est_rots.append(result.atRot3(i).matrix())
est_rots = np.array(est_rots)

## Relative rotation error computation
relrot_mats_allrefs_gt = gt_rots[:, :3, :3] @ gt_rots[0, :3, :3]
relrot_mats_allrefs_est = est_rots[:, :3, :3] @ est_rots[0, :3, :3]
rel_rot_errs = compute_angular_error_batch(relrot_mats_allrefs_gt, relrot_mats_allrefs_est)
print(f"Average Relative Rotation Error:, {(np.mean(rel_rot_errs)):.5f} deg")
```

The parameter in the script is `NUM_NODES` 
1. For NUM_NODES=3, it converges but `rel_rot_errs` is high
2. For NUM_NODES=5, sometimes it converges but `rel_rot_errs` is high 
3. For NUM_NODES=10, it converges but `rel_rot_errs` is high
4. For NUM_NODES=3, **it hardly converges**

anilesec, 2023-09-10:
Dear @dellaert 

Looking forward to hearing from you. I spend a considerable amount of time debugging, perhaps I lack the relevant knowledge. Any help is highly appreciated. I am very much thankful for your precious time.
 Thank you very much in Advance!

anilesec, 2023-09-11:
Another simple example is where the shonan cost < 1e-4 but the relative rotation error is very high.

```
import gtsam
from gtsam import BinaryMeasurementRot3, Rot3, ShonanAveraging3, BinaryMeasurementsRot3
import math
import numpy as np

def compute_angular_error_batch(rotation1, rotation2):
    R_rel = np.einsum("Bij,Bjk ->Bik", rotation1.transpose(0, 2, 1), rotation2)
    t = (np.trace(R_rel, axis1=1, axis2=2) - 1) / 2
    theta = np.arccos(np.clip(t, -1, 1))
    return theta * 180 / np.pi

R0 = np.eye(3)
R1 = Rot3.Yaw(math.radians(45)).matrix()
R2 = Rot3.Pitch(math.radians(90)).matrix()
gt_rots = np.array(
    [R0, R1, R2]
)
R01 = R1 @ np.linalg.inv(R0)
R02 = R2 @ np.linalg.inv(R0)
R12 = R2 @ np.linalg.inv(R1)

measurements = BinaryMeasurementsRot3()
unit3 = gtsam.noiseModel.Unit.Create(3)

m01 = BinaryMeasurementRot3(0, 1, Rot3(R01), unit3)
m02 = BinaryMeasurementRot3(0, 2, Rot3(R02), unit3)
m12 = BinaryMeasurementRot3(1, 2, Rot3(R12), unit3)
measurements.append(m01)
measurements.append(m02)
measurements.append(m12)
obj = ShonanAveraging3(measurements)
assert isinstance(obj, ShonanAveraging3)
initial = obj.initializeRandomly()
result, cost = obj.run(initial, min_p=3, max_p=5)
assert cost < 1e-4
print(result)
print('Rot. Averaging cost:', cost)

## estimated rotations
est_rots = []
for i in range(result.size()):
    est_rots.append(result.atRot3(i).matrix())
est_rots = np.array(est_rots)

## Relative rotation error computation
relrot_mats_allrefs_gt = gt_rots[:, :3, :3] @ gt_rots[0, :3, :3]
relrot_mats_allrefs_est = est_rots[:, :3, :3] @ est_rots[0, :3, :3]
rel_rot_errs = compute_angular_error_batch(relrot_mats_allrefs_gt, relrot_mats_allrefs_est)
print(f"Average Relative Rotation Error:, {(np.mean(rel_rot_errs)):.5f} deg")
```