borglab/gtsam #1629: Cannot found tbb library when using MSVC compiler

ffgamelife, 2023-09-06:
I try to compile gtsam-4.2 using msvc 2019 under windows, Like issue #240, I also encountered a similar problem. I configured the tbb path correctly, but it still shows tbb-NOTFOUND when compiling. 
I am not familiar with cmake. After google, I found [this link](https://stackoverflow.com/questions/14850736/cmake-imported-library-behaviour) . It seems that under the windows platform, the IMPORTED_IMPLIB property needs to be used to specify the *.lib location (IMPORTED_LOCATION is used to specify the *.dll location?). 
I add set_target_properties in FindTBB.cake IMPORTED_LOCATION
I added the “IMPORTED_IMPLIB” property under “set_target_properties” in the “FindTBB.cmake” file and it works.

ProfFan, 2023-09-19:
We are using official TBB CMake now, do you have the same issue with `develop`?

varunagrawal, 2023-10-10:
Closing due to lack of a response. Please re-open if this is still an issue after trying out the latest `develop` branch.

closed 2023-10-10 by varunagrawal