borglab/gtsam #1633: error while loading shared libraries

stepeos, 2023-09-09:
## Description

I successfully compiled and linked my ./my_program using cmake. Attempting to run `./my_program` results in:
```
error while loading shared libraries: libmetis-gtsamDebug.so: cannot open shared object file: No such file or directory
```
I can find said shared library object in the local installation directory. However the output of `ldd` suggests the library file cannot be found.
```
$ ldd libgtsamDebug.so
	>linux-vdso.so.1 (0x00007fffac55f000)
	libboost_serialization.so.1.71.0 => /lib/x86_64-linux-gnu/libboost_serialization.so.1.71.0 (0x00007f037346d000)
	libboost_filesystem.so.1.71.0 => /lib/x86_64-linux-gnu/libboost_filesystem.so.1.71.0 (0x00007f037344f000)
	libboost_timer.so.1.71.0 => /lib/x86_64-linux-gnu/libboost_timer.so.1.71.0 (0x00007f0373445000)
	libtbb.so.2 => /lib/x86_64-linux-gnu/libtbb.so.2 (0x00007f03733ff000)
	libtbbmalloc.so.2 => /lib/x86_64-linux-gnu/libtbbmalloc.so.2 (0x00007f03733bd000)
	libmetis-gtsamDebug.so => not found
	libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f03731d9000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f037308a000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f037306f000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f0372e7d000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f0372e5a000)
	libboost_chrono.so.1.71.0 => /lib/x86_64-linux-gnu/libboost_chrono.so.1.71.0 (0x00007f0372e4c000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f0372e44000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f0376b1c000)
```

## Steps to reproduce
I have encountered an error when upgrading from 4.1.1 to 4.2, using the release zip from [here](https://github.com/borglab/gtsam/releases/tag/4.2). After following the installation instructions
```
  $ mkdir build
  $ cd build
  $ cmake \
  -D GTSAM_BUILD_UNSTABLE=OFF \
  -D CMAKE_INSTALL_PREFIX=../../../../../install_isolated \
  -D GTSAM_USE_SYSTEM_EIGEN=ON \ # found 4.3
  -D CMAKE_BUILD_TYPE=Debug \
  ..
  $ make check # this command finished successfully, all tests have passed
  $ make install
```
Then in my CMakeLists.txt I create a shared library object:
```
find_package(GTSAM REQUIRED)
...
target_link_libraries(${PROJECT_NAME}_backend
    ${catkin_LIBRARIES}
    #many more libraries...
    gtsam
)
```

## Expected behavior

## Environment

I am using Ubuntu focal, latest with the following configuration:
```
Setting up gtsam...
-- The CXX compiler identification is GNU 10.5.0
-- The C compiler identification is GNU 9.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- GTSAM Version: 4.2a9
-- Performing Test COMPILER_HAS_WSUGGEST_OVERRIDE
-- Performing Test COMPILER_HAS_WSUGGEST_OVERRIDE - Success
-- Performing Test COMPILER_HAS_WMISSING_OVERRIDE
-- Performing Test COMPILER_HAS_WMISSING_OVERRIDE - Failed
-- GTSAM_POSE3_EXPMAP=ON, enabling GTSAM_ROT3_EXPMAP as well
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.65") found components: serialization system filesystem thread program_options date_time timer chrono regex 
-- Found Eigen version: 3.4.0
-- Looking for execinfo.h
-- Looking for execinfo.h - found
-- Looking for getline
-- Looking for getline - found
-- checking for thread-local storage - found
-- Could NOT find MKL (missing: MKL_INCLUDE_DIR MKL_LIBRARIES) 
-- Found OpenMP_C: -fopenmp (found version "4.5") 
-- Found OpenMP_CXX: -fopenmp (found version "4.5") 
-- Found OpenMP: TRUE (found version "4.5")  
-- Found TBB: /usr/include (found suitable version "2020.1", minimum required is "4.4") found components: tbb tbbmalloc 
-- Building 3rdparty
-- Could NOT find GeographicLib (missing: GeographicLib_LIBRARY_DIRS GeographicLib_LIBRARIES GeographicLib_INCLUDE_DIRS) 
-- Building base
-- Building basis
-- Building geometry
-- Building inference
-- Building symbolic
-- Building discrete
-- Building hybrid
-- Building linear
-- Building nonlinear
-- Building sam
-- Building sfm
-- Building slam
-- Building navigation
-- GTSAM Version: 4.2.0
-- Install prefix: /path/to/some/workspace/install_isolated
-- Building GTSAM - shared: ON
-- Wrote /somewhere/inside/my_project/gtsam-4.2/build/GTSAMConfig.cmake
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
-- ===============================================================
-- ================  Configuration Options  ======================
--  CMAKE_CXX_COMPILER_ID type                       : GNU
--  CMAKE_CXX_COMPILER_VERSION                       : 10.5.0
--  CMake version                                    : 3.20.0
--  CMake generator                                  : Unix Makefiles
--  CMake build tool                                 : /usr/bin/make
-- Build flags                                               
--  Build Tests                                      : Enabled
--  Build examples with 'make all'                   : Enabled
--  Build timing scripts with 'make all'             : Disabled
--  Build shared GTSAM libraries                     : Enabled
--  Put build type in library name                   : Enabled
--  Build libgtsam_unstable                          : Disabled
--  Build GTSAM unstable Python                      : Enabled
--  Build MATLAB Toolbox for unstable                : Disabled
--  Build for native architecture                    : Disabled
--  Build type                                       : Debug
--  C compilation flags                              :  -g
--  C++ compilation flags                            :  -g
--  GTSAM_COMPILE_FEATURES_PUBLIC                    : cxx_std_11
--  GTSAM_COMPILE_OPTIONS_PUBLIC                     : 
--  GTSAM_COMPILE_DEFINITIONS_PUBLIC                 : 
--  GTSAM_COMPILE_OPTIONS_PUBLIC_DEBUG               : 
--  GTSAM_COMPILE_DEFINITIONS_PUBLIC_DEBUG           : 
--  Use System Eigen                                 : ON (Using version: 3.4.0)
--  Use System Metis                                 : OFF
--  Using Boost version                              : 1.71.0
--  Use Intel TBB                                    : Yes (Version: 2020.1)
--  Eigen will use MKL                               : MKL not found
--  Eigen will use MKL and OpenMP                    : OpenMP found but GTSAM_WITH_EIGEN_MKL is disabled
--  Default allocator                                : TBB
--  Cheirality exceptions enabled                    : YES
--  Build with ccache                                : No
-- Packaging flags
--  CPack Source Generator                           : TGZ
--  CPack Generator                                  : TGZ
-- GTSAM flags                                               
--  Quaternions as default Rot3                      : Disabled
--  Runtime consistency checking                     : Disabled
--  Rot3 retract is full ExpMap                      : Enabled
--  Pose3 retract is full ExpMap                     : Enabled
--  Allow features deprecated in GTSAM 4.2           : Enabled
--  Metis-based Nested Dissection                    : Enabled
--  Use tangent-space preintegration                 : Enabled
-- MATLAB toolbox flags
--  Install MATLAB toolbox                           : Disabled
-- Python toolbox flags                                      
--  Build Python module with pybind                  : Disabled
-- ===============================================================
-- Configuring done
```

## Additional information

A temporary fix is to add the entry to the `.bashrc`:
```
export LD_PRELOAD="$LD_PRELOAD:/path/to/some/workspace/install_isolated/lib/libmetis-gtsamDebug.so"
```
This fixes the problem temporarily, so does [this](https://github.com/borglab/gtsam/issues/380#issuecomment-1053260888) commentator's suggestion.

I am surprised, that the `make check` command finished successfully, so I wonder whether the problem stems from my CMakeLists.txt?

varunagrawal, 2023-09-10:
Since you're installing from source and have a custom install prefix, it may be possible that CMake is being picky about this. I am also surprised it worked for you before without the `LD_PRELOAD` or the `CMAKE_PREFIX_PATH` variables including the custom install directory.

Can you try building from scratch again in a new directory (e.g. build2) and report back on whether you still have this issue? You may want to delete the current install by running `xargs rm -rf < install_manifest.txt` in your current build folder.

stepeos, 2023-09-18:
> Since you're installing from source and have a custom install prefix, it may be possible that CMake is being picky about this. I am also surprised it worked for you before without the `LD_PRELOAD` [...]

This fixed the issue? That is very strange. Thanks for the help.

closed 2023-09-18 by stepeos