borglab/gtsam #1642: Building fails in VS22 due to ambiguous call

knutur94, 2023-09-20:
## Description

When building i get the following error message: 
```

Error C2668:	'gtsam::CameraSet<CAMERA>::SchurComplement': ambiguous call to overloaded function	ISAM2Example_SmartFactor	C:\cpp\libs\gtsam\gtsam\geometry	C:\cpp\libs\gtsam\gtsam\geometry\CameraSet.h	379	
```
## Steps to reproduce

1. clone repository
2. Choose Generator: Visual Studio 17 2022 with x64
3. Press configure and generate
4. Press Open Project
5. build ALL_BUILD

Then I get 6 error messages all related to ``` 'gtsam::CameraSet<CAMERA>::SchurComplement': ambiguous call to overloaded function```

These error messages comes from:
1. ISAM2Example_SmartFactor
2. SFMExample_SmartFactor
3. SFMExample_SmartFactorPCG
## Expected behavior

I expected gtsam to build.
I got it to build when unchecking GTSAM_BUILD_EXAMPLES_ALWAYS and GTSAM_BUILD_UNSTABLE, but it doesn't fix the underlying issue.

## Environment

Windows 10 x64
Visual Studio 17 2022
Boost 1.82
Cmake 3.26

Tobias-Fischer, 2023-10-29:
We observed the same issue in conda-forge, and went back to vs2019 in the meantime.

talregev, 2024-01-15:
I have the same issue. 
The problem is new CL in msvc.
I have a workaround for that.
You can lower the cl version with environment variable: 
I think it how you do it. (I manage to config it on the ci, and I think this is what control the cl version). 
`set VSCMD_ARG_VCVARS_VER=29`
Also please write down your cl version.
You can type by on the cmd:
`cl`

talregev, 2024-01-17:
I found a guide how to change cl version.
You should change to  -vcvars_ver=14.29
https://learn.microsoft.com/en-us/cpp/build/building-on-the-command-line?view=msvc-160

talregev, 2024-01-18:
@knutur94 
I create a fix that should compile with msvc 2022 without change the cl version.
Please compile my branch on this PR: #1715

PR #1715 (2024-03-18): Support msvc 2022 along 2019
Fix #1642 
Support msvc 2022 along 2019:
- Change to toolset: 14.38
- Fix ambiguous call to overloaded function CameraSet.h(331)
closed 2024-03-18 by varunagrawal