borglab/gtsam #1644: Get key index of the latest robot pose in the optimized result

sandilyasg, 2023-09-28:
I have the following code: `Marginals marginals(graph_, result_);` where graph_ is `gtsam::NonlinearFactorGraph graph_;` and result_ is `gtsam::Values result_ ;` . I want to get the key index/number of the latest robot pose in this graph after I optimize, how can I do this?

varunagrawal, 2023-09-29:
We assume you have some mechanism of keeping track of the keys you're inserting into the factor graph (such as a counter) which you can use to keep track of the latest index. Since `gtsam::Values` can have more than just robot poses, we can't assume the index number based on the size of the Values container and hence leave it to the user.

I highly recommend asking this question on the Google Group as mentioned in the issue template. You should be able to get much faster and better help there. :slightly_smiling_face:

Closing this as it is not a code issue. Hopefully you are able to resolve your issue.

closed 2023-09-29 by varunagrawal

sandilyasg, 2023-09-29:
Thank you for your help, will ask in the Google group.