borglab/gtsam #1647: Exporting my function to Matlab - explanations are outdated

etamarlu, 2023-10-05:
I am following [this repository](https://github.com/borglab/gtsam-project-matlab/tree/992cb78b1f003fdd7fc255d687f9b15cacf46eb7)

## Description

Instructions result in an error, after the below fix Matlab's toolbox files wouldn't generate.

I figured I have to replace "boost" with "std" and I added to the .h file:  
"#include <memory>" 
"include <gtsam/global_includes.h>" (it wouldn't work alone withouth including memory)

and 
"find_package(GTSAM REQUIRED)
include_directories(${GTSAM_INCLUDE_DIR})
set(GTSAM_LIBRARIES gtsam)"

to the CMakeList file

Cmake and build are successful. but no toolbox is generated.

## Steps to reproduce

## Expected behavior

## Environment

## Additional information

varunagrawal, 2023-10-05:
@etamarlu I made some [updates](https://github.com/borglab/gtsam-project-matlab/pull/2) to the project template file. Can you please clone the latest `main` branch and try again?

etamarlu, 2023-10-05:
Thanks, I tried it now, std namespace is not recognized, it still fails with the following error:

`In file included from /home/et/gtsam-project-matlab-main/example/PrintExamples.cpp:18:
/home/et/gtsam-project-matlab-main/example/PrintExamples.h:57:6: error: variable or field ‘consumePrinter’ declared void
   57 | void consumePrinter(std::shared_ptr<PrintExample>& pe) {}
      |      ^~~~~~
/home/et/gtsam-project-matlab-main/example/PrintExamples.h:57:26: error: ‘shared_ptr’ is not a member of ‘std’
   57 | void consumePrinter(std::shared_ptr<PrintExample>& pe) {}`

also, it doesn't appear in the readme (and I still haven't confirmed), but I changed the "WRAP_TOOLBOX_INSTALL_PATH" to control where the matlab 'toolbox' is generated, can you please confirm it is ok?

varunagrawal, 2023-10-05:
@etamarlu updated to handle the error message. It's a C++11 standards issue.

Yes, `WRAP_TOOLBOX_INSTALL_PATH` is the correct variable to set.

etamarlu, 2023-10-05:
same still

re-downloaded everything and started from scratch, yet..

varunagrawal, 2023-10-06:
Same compilation error? Can you please paste the output you're seeing over here?

varunagrawal, 2023-10-06:
I'll also need information about your OS, compiler, Matlab version etc. I can help you but I need additional information.

etamarlu, 2023-10-06:
It looks like the same compilation error (below)

I am using vm with 64 bits ubuntu-22.04.
matlab 23a
I guess the compiler is set by the make files, do you know how I can find which one is being used?

`In file included from /home/et/gtsam-project-matlab-main/example/PrintExamples.cpp:18:
/home/et/gtsam-project-matlab-main/example/PrintExamples.h:57:6: error: variable or field ‘consumePrinter’ declared void
   57 | void consumePrinter(std::shared_ptr<PrintExample>& pe) {}
      |      ^~~~~~~~~~~~~~
/home/et/gtsam-project-matlab-main/example/PrintExamples.h:57:26: error: ‘shared_ptr’ is not a member of ‘std’
   57 | void consumePrinter(std::shared_ptr<PrintExample>& pe) {}
      |                          ^~~~~~~~~~
/home/et/gtsam-project-matlab-main/example/PrintExamples.h:21:1: note: ‘std::shared_ptr’ is defined in header ‘<memory>’; did you forget to ‘#include <memory>’?
   20 | #include <iostream>
  +++ |+#include <memory>
   21 | #include <string>
/home/et/gtsam-project-matlab-main/example/PrintExamples.h:57:49: error: expected primary-expression before ‘>’ token
   57 | void consumePrinter(std::shared_ptr<PrintExample>& pe) {}
      |                                                 ^
/home/et/gtsam-project-matlab-main/example/PrintExamples.h:57:52: error: ‘pe’ was not declared in this scope
   57 | void consumePrinter(std::shared_ptr<PrintExample>& pe) {}
      |                                                    ^~
make[2]: *** [CMakeFiles/example_matlab.dir/build.make:76: CMakeFiles/example_matlab.dir/example/PrintExamples.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:249: CMakeFiles/example_matlab.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
`

varunagrawal, 2023-10-06:
Thanks! I was able to reproduce the issue on my Ubuntu machine and have made all the fixes. Please re-download the latest version of the project template and you should be good to go.

[image]

etamarlu, 2023-10-07:
It works perfectly!

Thank you :)

closed 2023-10-07 by etamarlu