borglab/gtsam #1648: SmartProjectionPose3Factor in Python API

Dead-Hand, 2023-10-06:
## Description
The GTSAM Python API has the SmartProjectionPose3Factor with several of its C++ classes available for use. However, it does not include some critical functions (specifically the point() function) which without render this factor unusable in Python. 

The point() function is used to extract out a 3d point triangulated estimate to feed into the initial measurements (see [here](https://github.com/borglab/gtsam/blob/73d716365578b8db5e61b2b919bd3427f89a65d5/examples/SFMExample_SmartFactor.cpp#L119C22-L119C22) for a relevant C++ example demonstrating the need for access to the point() function). Without it, it requires the initial measurement be derived by something else, which would defeat some of the purpose of using the smart factor.

## Steps to reproduce
`import gtsam`
`help(gtsam.SmartProjectionPose3Factor)`

point() is not listed as a member function.

## Expected behavior

point() to be listed as a member function.

## Environment
OS: Ubuntu 20.04 LTS
Python version 3.8.10
GTSAM version 4.3a0

ProfFan, 2023-10-07:
Hi, thank you for the bug report! This can be added fairly easily, by adding one line in the `.i` file

Dead-Hand, 2023-10-09:
> Hi, thank you for the bug report! This can be added fairly easily, by adding one line in the `.i` file

Thanks for the response! I'll preface the following by saying I do not have much experience with pybind11...

**TL;DR**: It looks like I would add something like so (at line 156): `TriangulationResult point() const;` Does this look right or were you thinking of something else? I plan on testing this out once I'm back at my computer later today.

Longer version is (writing this out for my own understanding and maybe help some others - please correct if I misunderstand though):

It appears the [slam.i](https://github.com/borglab/gtsam/blob/develop/gtsam/slam/slam.i) file is the one I'd be interested in, and I would want to add a member function to the `virtual class SmartProjectionPoseFactor` class.

However, reviewing the API documentation again, I noticed the exact return of [point()](https://gtsam.org/doxygen/a04872.html) is actually a more complex class [TriangulationResult](https://gtsam.org/doxygen/a03456.html). TriangulationResult is already defined in geometry.i (and I can confirm I am able to see it available in the Python API), I can see that `#include <gtsam/geometry/triangulation.h>` is present in SmartProjectionFactor.h, which is already imported into slam.i. 

Will give it a try and report back.

Dead-Hand, 2023-10-10:
Following up to say this worked! I have a slightly older version of the develop branch so I hesitate to link it here, but following the method I wrote up in my previous message worked to give me access to the point() method. Will try to work this into a PR when I can.

varunagrawal, 2023-10-10:
@Dead-Hand glad to see you figured it out. I made a PR to fix this issue (sorry for the delay).

varunagrawal, 2023-10-10:
Also wanted to mention that your analysis and understanding was exactly on point! If you see my PR, you'll notice I made the exact same change. If you have other methods you'd like to wrap, we welcome a PR for that.

Dead-Hand, 2023-11-06:
Confirmed that #1654 fixed this. Closing my issue. Thanks!

closed 2023-11-06 by Dead-Hand