borglab/gtsam #1662: In Python, I've defined a custom factor but I'm unsure how to compute the Jacobian matrix. Can GTSAM automatically handle differentiation? Are there relevant interfaces for this?

goodgoodliving, 2023-10-27:
> You create a custom function that computes `z-f(x)`. Then, in the `evaluateError` function of your customized factor, you can call the `numericalDerivative` function to compute the Jacobian numerically. Examples in `testNumericalDerivative.cpp` will help you get familiar with the function. 

@yetongumich 
Hello, may I ask if it's possible to automatically solve for the Jacobian using GTSAM if I'm implementing a custom factor in Python, where I only know the error function and am not familiar with how to perform the Jacobian calculation? Are there relevant interfaces for this in GTSAM?

_Originally posted by @goodgoodliving in https://github.com/borglab/gtsam/issues/1547#issuecomment-1782173621_

chrisflesher, 2024-11-18:
There are some auto-diff libraries that can do this for you (e.g. JAX, SymPy, etc.).

I'm currently using JAX like this:
```
def _my_residual(
    measurements: onp.ndarray,
    quaternions: onp.ndarray,
    relative_times: onp.ndarray,
    this: gtsam.CustomFactor,
    values: gtsam.Values,
    jacobians: typing.Optional[typing.List[onp.ndarray]],
    ) -> onp.ndarray:
    keys = this.keys()
    xi = values.atPose3(keys[0])
    vi = values.atVector(keys[1])
    vf = values.atVector(keys[2])
    xe_bi = onp.array([0., 0., 0., 0., 0., 0.])
    q_n_bi = xi.rotation().toQuaternion().coeffs()
    q_bi_b = quaternions
    args = (q_n_bi, q_bi_b, relative_times)
    in_axes = (None, None, None, None, 0, 0)
    predicted_measurements = jax.vmap(jax.jit(_predict_measurement), in_axes=in_axes)(xe_bi, vi, vf, *args)
    if jacobians is not None:
        jax_jacobians = jax.vmap(jax.jit(_predict_dvl_jacobian), in_axes=in_axes)(xe_bi, vi, vf, *args)
        jacobians[0] = jax_jacobians[0]
        jacobians[1] = jax_jacobians[1]
        jacobians[2] = jax_jacobians[2]
    errors = (onp.array(predicted_measurements) - measurements)[:, onp.newaxis]
    return errors

def _predict_measurement(xe_bi, vi, vf, q_n_bi, q_bi_b, relative_times):
    re_bi = jax_scipy_spatial.transform.Rotation.from_rotvec(xe_bi[0:3])
    r_n_bi = jax_scipy_spatial.transform.Rotation.from_quat(q_n_bi) * re_bi
    r_bi_b = jax_scipy_spatial.transform.Rotation.from_quat(q_bi_b)
    r_n_b = r_n_bi * r_bi_b
    v_n = vi + weight * (vf - vi)
    v_b = r_n_b.inv().apply(v_n)
    return v_b

_predict_dvl_jacobian = jax.jacrev(_predict_dvl_velocity, argnums=[0, 1, 2])
```

varunagrawal, 2024-12-24:
You need the Jacobians for linearizing the factor, but if `numericalDerivative` (aka finite differences) is not sufficient for your needs, @chrisflesher's comment about using Jax is really helpful.

Also since this is not quite a code issue, I highly recommend asking this question on the Google Group.

closed 2024-12-24 by varunagrawal