borglab/gtsam #1665: Significant Slowdown in Python

Dead-Hand, 2023-11-02:
## Description

I've noticed a significant slowdown (~6667% longer to run) in the Python API when using the GTSAM LM optimizer between versions 4.2 and the latest develop branch (4.3a0).

## Steps to reproduce

I modified one of the existing Python example scripts to demonstrate the problem. Please download the attached script for an example. 

1. Install GTSAM 4.2 (I've been grabbing this from pypi i.e. `pip install gtsam`)
2. Download the attached ZIP file and unzip. It contains the test script: [DogLegOptimizerExample_timing_issues.zip](https://github.com/borglab/gtsam/files/13241623/DogLegOptimizerExample_timing_issues.zip)
3. Execute the attached script which will wait until the 250th loop iteration before reporting the and note the optimizer runtime: `python3 DogLegOptimizerExample_timing_issues.py`
4. Uninstall gtsam: `pip uninstall gtsam`
5. Build the latest GTSAM develop branch (with Python support, so -DGTSAM_BUILD_PYTHON=1). Go into build/ (or wherever you have it set to build), then python/ and then install the python library using pip: `pip install .`
6. Execute the attached script and note the optimizer runtime: `python3 DogLegOptimizerExample_timing_issues.py`

When I run the above, step (2) reports a runtime of 0.00009s, whereas the output of step (5) reports 0.006s, representing a ~6667% increase in runtime. Your timing might vary slightly, but should still be significant.

The optimizer runtime reported in step (5) is significantly longer than in step (2), with the only thing being changed is the GTSAM version.

## Expected behavior

I would expect to see approximately identical optimizer runtimes between versions 4.2 and the latest develop branch (4.3a0).

## Environment
OS: Ubuntu 20.04 LTS
API: Python (version 3.8.10)

## Additional information

I have tried compiling the latest develop branch with and without TBB support and I have not noticed any difference in the above behavior.

ProfFan, 2023-11-02:
Did you compile GTSAM in `Release`?

Dead-Hand, 2023-11-02:
> Did you compile GTSAM in `Release`?

I just now recompiled in Release mode and ran the same test a few times. It is much faster, but it's still about 50% the speed of running the same code in 4.2 release. LM optimize() takes about 0.0001s - 0.0002s with all else being equal.