borglab/gtsam #1668: Possibly incorrect discretization of IMU noise

mnissov, 2023-11-10:
## Description
Sorry if this is a false alarm, but I believe the noise discretization and parameter naming are slightly misleading and/or incorrect in the `CombinedImuFactor.cpp`. Below I detail why I believe this to be the case.

In line 141 of `CombinedImuFactor.cpp`:
```For small values of | ˆω|, we can again take the limit and apply L’Hˆopital’s rule and obtain
lim
| ˆω|→0 Ψ = −I3×3∆t + lim
| ˆω|→0
sin (| ˆω|∆t) ∆t
2| ˆω| b ˆω ×c − lim
| ˆω|→0
(1 − cos (| ˆω|)) ∆t
3| ˆω|2 b ˆω ×c2 (177)
= −I3×3∆t + lim
| ˆω|→0
cos (| ˆω|∆t) ∆t2
2 b ˆω ×c − lim
| ˆω|→0
sin (| ˆω|) ∆t2
6| ˆω| b ˆω ×c2 (178)
= −I3×3∆t + ∆t2
2 b ˆω ×c − lim
| ˆω|→0
cos (| ˆω|) ∆t3
6 b ˆω ×c2 (179)
= −I3×3∆t + ∆t2
2 b ˆω ×c − ∆t3
6 b ˆω ×c2 (180)
A note regarding the small ω approximation The error committed by using the small angle approximation is
roughly in the order of ∆tn+2| ˆω|2, if the non-approximated term is divided by | ˆω|n.
Therefore, at a sampling rate of 100 Hz, the error for Θ above would be in the order of 10−6 · | ˆωthresh|2, where
| ˆωthresh| denotes the threshold for the small ω approximation.
2.5.2 The Noise Covariance Matrix Qd
The covariance of the noise in the discrete time system can be computed according to [4, p. 171]
Qd =
∫ tk+1
tk
Φ(tk+1, τ )Gc(τ )QcGT
c (τ )ΦT(tk+1, τ ) dτ (181)
=
∫ tk+1
tk
[ Θ Ψ
03×3 I3×3
] [−I3×3 03×3
03×3 I3×3
]
Qc
[−I3×3 03×3
03×3 I3×3
] [ΘT 03×3
ΨT I3×3
]
dτ (182)
=
∫ tk+1
tk
[ −Θ Ψ
03×3 I3×3
]
Qc
[−ΘT 03×3
ΨT I3×3
]
dτ (183)
(184)
Assuming the noise to be white and independent (cf. eq. (162)), we can furt
  // first order uncertainty propagation
  // Optimized matrix mult: (1/dt) * G * measurementCovariance * G.transpose()
```
Already here the comment is not correct, the first order approximation of discrete noise propagation should be

$$Q_d = G Q G^\top  \Delta T$$

where $Q$ and $Q_d$ are the diagonal matrices of the spectral density (continuous noise power) and covariance (discrete), according to [1, sec 2.5.2], [3, eq. 4.111], [4, eq. 3.45], [5, eq. 25]. I tried to reference the same sources as `doc/ImuFactor.pdf`, but Simon in [2] doesn't seem to provide system definitions which include a noise input matrix (G), therefore I defaulted to what I believe are standard aided inertial navigation texts.

I'm assuming `measurementCovariance` is referring to the PSD, given that `PreintegrationParams.h` and `PreintegratedRotation.h` contain the following comments
```
  Matrix3 accelerometerCovariance; ///< continuous-time "Covariance" of accelerometer
  Matrix3 integrationCovariance; ///< continuous-time "Covariance" describing integration uncertainty
...
  Matrix3 gyroscopeCovariance;  ///< continuous-time "Covariance" of gyroscope measurements
```
If these values do actually correspond to PSD then I think the following operations don't seem to make sense
```
  const Matrix3& aCov = p().accelerometerCovariance;
  const Matrix3& wCov = p().gyroscopeCovariance;
...
D_R_R(&G_measCov_Gt) =
      (theta_H_biasOmega * (wCov / dt) * theta_H_biasOmega.transpose()) ...
```
as dividing `wCov` with `dt` essentially performs the "conversion" to the equivalent covariance from PSD, following [1, eq. 129] and [4, eq. 14.83]. I'm not aware of how this first order discretization should look if the uncertainty is already discretized, so perhaps this is my misunderstanding.

In short, it looks like there is some discrepancy between what is implemented and a first order approximation for the discretization of covariance propagation. Looking more like individual PSDs are converted to covariances and then pre/post multiplied by the noise input matrix. I'm curious to hear your thoughts on this, if I am indeed correct the suggested fix should be simple assuming all noise values are in fact PSDs and not covariances.

## References
[1] Nikolas Trawny and Stergios I. Roumeliotis. Indirect kalman filter for 3 d attitude estimation. 2005.
[2] Dan Simon. Optimal state estimation: Kalman, H infinity, and nonlinear approaches. John Wiley & Sons, 2006.
[3] Jay Farrell. Aided Navigation: GPS with High Rate Sensors (1st. ed.). McGraw-Hill, Inc., 2008.
[4] Paul Groves, Principles of GNSS, Inertial, and Multisensor Integrated Navigation Systems, Second Edition, Artech, 2013.
[5] Janosch Nikolic. Characterisation, calibration, and design of visual-inertial sensor systems for robot navigation. PhD thesis, ETH Zurich, 2016.

varunagrawal, 2023-11-15:
I believe the confusion is arising from the fact that you are using $Q$ and $Q_d$ to represent the process noise covariance (not the PSD which is related to the autocorrelation), while our math is using $Q$ to represent the measurement noise.

In `ImuFactor.pdf` if you look at the definition of $F$, you will see that there is appropriate scaling by $\Delta t$ where applicable. Hope that makes sense?

varunagrawal, 2023-12-29:
Closing this issue for now. If this issue is still relevant after the above discussion, please feel free to reopen.

closed 2023-12-29 by varunagrawal

mnissov, 2024-01-03:
Sorry for the delay. I think you're right, I see the misunderstanding.