borglab/gtsam #1672: `Rot2::fromCosSin` normalizes when header indicates it should not

asasine, 2023-11-17:
## Description
In the declaration of `Rot2::fromCosSin`, the header indicates normalization does *not* happen:

https://github.com/borglab/gtsam/blob/1121ece0eb16324c8e1da5247f5abc62a611252e/gtsam/geometry/Rot2.h#L71-L72

But the implementation normalizes the return value:

https://github.com/borglab/gtsam/blob/1121ece0eb16324c8e1da5247f5abc62a611252e/gtsam/geometry/Rot2.cpp#L27-L30

Is this intended? It's possible I'm misunderstanding the use case of `Rot2::fromCosSin` or what "normalization" means here.

## Expected behavior
The returned object from `Rot2::fromCosSin` does not normalize the given `c` and `s`, even if they are denormalized.

## Environment
C++

## Additional information
I'm not familiar enough with the library to understand the full impact of changing this function to strictly `return Rot2(c, s)`.

arutkowski, 2023-11-21:
I also think the comment in the header is incorrect, but I don't think it leads to any real problems. At worst, the function is a little inefficient, since it would be unnecessary to normalize if it can be guaranteed that the arguments passed to this function satisfy `c*c + s*s == 1`. 

I believe a `Rot2` is stored as separate sine and cosine components (i.e. member variables `s_` and `c_`) rather than an angle. Thus, a rotation matrix can readily be formed as needed. 

It appears this function is essentially the same as `Rot2::atan2` but with the order of the arguments reversed. Indeed, `std::atan2(2,1)` and `std::atan2(4,2)` should return the same angle. However, if it weren't for the normalization, then `Rot2::fromCosSin(1,2)` would be different than `Rot2::fromCosSin(2,4)`, and neither would represent a proper rotation matrix with unit norm.

I think it could be handy if this function didn't normalize, as the comment suggests, if it is only used when `c*c + s*s == 1`. It would avoid a division and associated checks for division by a small number (which is currently 1e-10).

varunagrawal, 2024-01-05:
I agree with the comments so far. @asasine can you please make a PR? It should be a simple 1 line change.

PR #1707 (2024-08-25): ISAM2: Add variables before computing delta
Fixes #301 

A `gtsam::ValuesKeyDoesNotExist` error is thrown from the function `nonlinearFactors_.error(theta_)`.
The issue is that `nonlinearFactors_.error(theta_)` is computed as an argument to `DoglegOptimizerImpl::Iterate`, where a factor with the new key exists but `theta_` has not been updated with the new key-value pairs yet.

It is a simple quirk that Dogleg requires the updated Values since it needs them to compute the delta update via the dogleg point.

The issue in `ISAM2::update` is that the delta update is computed first to check for re [... cut]
closed 2024-08-25 by varunagrawal